karloygard/xcomfortd-go #4: Support for more xcomfort devices ?

nhmariend, 2020-05-01:
Hi !
Do you plan to plan to add support for xcomfor devices ?
I tried the xcomfortd and tested the shutters/blinds, called cover in ha.
That works fine there, I have shutters on 3 sides on the house and want them to just be down when the sun is on that side. That worked fine. After that I tried your new integration with ha but you have just support for light. I hope to get support for device like motion sensors and maybe the one that registrate use of power and device for heating.

My shutterss/blinds goes down and then jog one up so the room doesnt get all dark.

Here is how it looks like in ha. I have disabeled it now and use Sensio controller.
Hope to get more to work here so I can shut down the Sensio controller.

[image]

karloygard, 2020-05-01:
The current code supports temperature sensors, pushbuttons and switches/dimmers (including power usage) for the time being.

Adding support for blinds and motion sensors should not be too hard, but I can't test those myself, so I need assistance from you. For the blinds, I need the configuration entry for a typical blind (along the lines of what's specified in https://www.home-assistant.io/integrations/cover.mqtt/), and if would be useful to see your datapoints file.

Similar for the motion sensor, a suggestion for a configuration entry would save me a lot of time.

nhmariend, 2020-05-01:
I have  a file under folder cover that is named persienner.yaml
- platform: mqtt
  name: "Kontor"
  state_topic: "xcomfort/3/get/shutter"
  command_topic: "xcomfort/3/set/shutter"
  payload_open: "up"
  payload_close: "down"
  payload_stop: "stop"
#  position_open: 100
#  position_closed: 0
  state_open: "up"
  state_closed: "down"
- platform: mqtt
  name: "Tv"
  state_topic: "xcomfort/4/get/shutter"
  command_topic: "xcomfort/4/set/shutter"
  payload_open: "up"
  payload_close: "down"
  payload_stop: "stop"

Is this what you need ?

I have some actuators in spare that I can add to test. I have one motion sensor and a few other devices.  
#  position_open: 100
#  position_closed: 0
  state_open: "up"
  state_closed: "down"
- platform: mqtt
  name: "Verandadør"
  state_topic: "xcomfort/5/get/shutter"
  command_topic: "xcomfort/5/set/shutter"
  payload_open: "up"
  payload_close: "down"
  payload_stop: "stop"
#  position_open: 100
#  position_closed: 0
  state_open: "up"
  state_closed: "down"
- platform: mqtt
  name: "Stuebord"
  state_topic: "xcomfort/6/get/shutter"
  command_topic: "xcomfort/6/set/shutter"
  payload_open: "up"
  payload_close: "down"
  payload_stop: "stop"
#  position_open: 100
#  position_closed: 0
  state_open: "up"
  state_closed: "down"
- platform: mqtt
  name: "Sofa"
  state_topic: "xcomfort/7/get/shutter"
  command_topic: "xcomfort/7/set/shutter"
  payload_open: "up"
  payload_close: "down"
  payload_stop: "stop"
#  position_open: 100
#  position_closed: 0
  state_open: "up"
  state_closed: "down"
- platform: mqtt
  name: "Spisestue"
  state_topic: "xcomfort/8/get/shutter"
  command_topic: "xcomfort/8/set/shutter"
  payload_open: "up"
  payload_close: "down"
  payload_stop: "stop"
#  position_open: 100
#  position_closed: 0
  state_open: "up"
[... 268 lines cut ...]
        entities:
          - entity: cover.persienner_ost
          - entity: cover.verandador
          - entity: cover.stuebord
      - type: entities
        title: Persienner Sør
        entities:
          - entity: cover.persienner_sor
          - entity: cover.sofa
          - entity: cover.spisestue
          - entity: cover.kjokken

Is this what you need ?

I have some xcomfort actuators and one motion sensors etc i spare that we can use to test.
I can add them i mrf so we have a light with one motion detector in the test.

I have a friend that has helped me with this too. 
I dont remember where we fixed to jog the shutters. (that is a standard function in mrf)

I think I edited in an .h file in your xcomfortd. (I can find out that with my friend.

We can have a call if you want tomorrow? 

Niels

karloygard, 2020-05-01:
I made a quick and dirty attempt at supporting shutters.  I had issues getting Home Assistant to play along, and the UI absolutely refused to enable the open and close arrows on the shutters, so I couldn't test very much. Could definitely be that I messed something up, but I pushed a new image with some support for shutters now, please test it and see if it does anything useful for you.

nhmariend, 2020-05-02:
Hi !
We have compiled it and run it on ubuntu on the nuc.
I can run the shutters now but the status if the shutters are opened or closed is missing.
Last message recieved is opening/closing after command open/close. Would like to have the status open/closed when command is finished.
This is in the yaml file
- platform: mqtt
  name: "Kontor"
  state_topic: "xcomfort/3/get/shutter"
  command_topic: "xcomfort/3/set/shutter"
  payload_open: "open"
  payload_close: "close"
  payload_stop: "stop"
  state_opening: "opening"  
  state_open: "opening"
  state_closing: "closing"  
  state_closed: "closing"

karloygard, 2020-05-02:
The xComfort shutter actuators do not report whether the blinds have been closed or opened, they report status messages `opening`, `closing` or `stopped`. There are two issues here; first, the message `stopped` has no counterpart in Home Assistant, and second, the message can be lost in transit.

It's probably possible to emulate some sort of open/closed status message, but this will require some thinking, not sure how to make this foolproof. Suggestions are appreciated!

karloygard, 2020-05-02:
It would be useful if you can send me a log of the daemon running (with the verbose flag enabled), while you're operating the blinds a few times (e.g. all the way up/down, partially up/down etc.) This could at least let me verify my assumptions about how the actuator actually works, I may have misunderstood something.

nhmariend, 2020-05-02:
Tasted some wine, so that will be tomorrow ! I will do that.

Not shure how to turn on verbose log, but think we find out that.

Is it possible to read the status after restart HA ?   It didnt work at first in UI-Lovelace, but after some clicking on the icons it start to work. just stop was ready to use not open/close.

Thanks for good work.

nhmariend, 2020-05-03:
Hi Karl !

I run this commands (had two mistypes

hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m close
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stepopen
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stepopen
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stepcøose
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stepcclose
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stepcclose
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stepopen
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stepclose
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stepclose
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m open
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m stop
hostname$ mosquitto_pub -h 192.168.2.210 -u mqttuser/pw -P mqttuser/pw -t xcomfort/3/set/shutter -m open

Got this in the log 
]0;root@nuc1: ~/work/xcomfortdroot@nuc1:~/work/xcomfortd# ./xcomfortd-go usb -v -f datapunkte.txt -i xcomfortd 
 -s tcp://mqtt:mqtt@192.168.2.210:1883
2020/05/03 10:47:53 Dp 1: device Dimming Actuator New Generation (CDAx-01/xx), serial 6673796, channel 0, 'Gang (Dimmeaktuator)'
2020/05/03 10:47:53 Dp 2: device Dimming Actuator (CDAx-01/xx), serial 4581672, channel 0, 'Kjøkkentak'
2020/05/03 10:47:53 Dp 3: device Shutter Actuator with Security (CJAU-01/02), serial 6078292, channel 0, 'Kontor'
2020/05/03 10:47:53 Dp 4: device Shutter Actuator with Security (CJAU-01/02), serial 6078315, channel 0, 'TV'
2020/05/03 10:47:53 Dp 5: device Shutter Actuator with Security (CJAU-01/02), serial 6078304, channel 0, 'Verandadør'
2020/05/03 10:47:53 Dp 6: device Shutter Actuator with Security (CJAU-01/02), serial 6055286, channel 0, 'Stuebord'
2020/05/03 10:47:53 Dp 7: device Shutter Actuator with Security (CJAU-01/02), serial 6078313, channel 0, 'Sofa'
2020/05/03 10:47:53 Dp 8: device Shutter Actuator with Security (CJAU-01/02), serial 6078286, channel 0, 'Spisestue'
2020/05/03 10:47:53 Dp 9: device Shutter Actuator with Security (CJAU-01/02), serial 6078302, channel 0, 'Kjøkken'
2020/05/03 10:47:53 Dp 10: device Switching Actuator (CSAx-01/xx), serial 4653603, channel 0, 'Spisebord'
2020/05/03 10:47:53 Dp 11: device Dimming Actuator (CDAx-01/xx), serial 5741263, channel 0, 'Stuetak'
2020/05/03 10:47:53 Dp 12: device Dimming Actuator (CDAx-01/xx), serial 5895171, channel 0, 'Trapptak'
2020/05/03 10:47:53 Dp 13: device Dimming Actuator (CDAx-01/xx), serial 5729803, channel 0, 'Trappned'
2020/05/03 10:47:53 Dp 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 6673796, channel 0, 'Gang (Statusforbindelse)'
2020/05/03 10:47:53 Connecting to MQTT broker 'tcp://192.168.2.210:1883' with id 'xcomfortd'
2020/05/03 10:47:53 Connected to broker
2020/05/03 10:47:53 Opened USB device 0
2020/05/03 10:47:53 CI HW/RF/FW revision: 0, 13.0, 0
2020/05/03 10:47:53 CI RF/Firmware release: 2.11, 2.05
2020/05/03 10:47:53 CI serial number: 5288736
2020/05/03 10:48:55 topic: xcomfort/3/set/shutter, message: close
2020/05/03 10:48:56 rx: [c103700002000000003c100a]
2020/05/03 10:48:56 Device 6078292 (channel 0-'Kontor') sent message (battery mains-powered, signal good, cyclic false) status shutter closing
2020/05/03 10:49:45 topic: xcomfort/3/set/shutter, message: stepopen
2020/05/03 10:49:45 rx: [c103700000000000003d100c]
2020/05/03 10:49:45 Device 6078292 (channel 0-'Kontor') sent message (battery mains-powered, signal good, cyclic false) status shutter stopped
2020/05/03 10:49:47 topic: xcomfort/3/set/shutter, message: stepopen
2020/05/03 10:49:56 topic: xcomfort/3/set/shutter, message: stepcÃ¸ose
2020/05/03 10:49:56 unknown shutter command stepcÃ¸ose
2020/05/03 10:50:03 topic: xcomfort/3/set/shutter, message: stepcclose
2020/05/03 10:50:03 unknown shutter command stepcclose
2020/05/03 10:50:06 topic: xcomfort/3/set/shutter, message: stepcclose
2020/05/03 10:50:06 unknown shutter command stepcclose
2020/05/03 10:50:13 topic: xcomfort/3/set/shutter, message: stepopen
2020/05/03 10:50:19 topic: xcomfort/3/set/shutter, message: stepclose
2020/05/03 10:50:22 topic: xcomfort/3/set/shutter, message: stepclose
2020/05/03 10:50:31 topic: xcomfort/3/set/shutter, message: open
2020/05/03 10:50:31 rx: [c103700001000000003b1002]
2020/05/03 10:50:31 Device 6078292 (channel 0-'Kontor') sent message (battery mains-powered, signal good, cyclic false) status shutter opening
2020/05/03 10:50:43 topic: xcomfort/3/set/shutter, message: stop
2020/05/03 10:50:43 rx: [c103700000000000003a1004]
2020/05/03 10:50:43 Device 6078292 (channel 0-'Kontor') sent message (battery mains-powered, signal good, cyclic false) status shutter stopped
2020/05/03 10:50:45 topic: xcomfort/3/set/shutter, message: open
2020/05/03 10:50:46 rx: [c103700001000000003a1006]
2020/05/03 10:50:46 Device 6078292 (channel 0-'Kontor') sent message (battery mains-powered, signal good, cyclic false) status shutter opening

I can do the same in sensio too, maybe send that as a mail to you ?

karloygard, 2020-05-03:
Thanks, will inspect the logs once in front of a computer. Logs from Sensio could certainly be useful as well. If mail is most convenient for you, you can use karloygard@gmail.com.

closed 2021-03-28 by karloygard