karloygard/xcomfortd-go #5: Beginner friendly instructions?

RonnyAL, 2020-05-03:
Hi!

I'll preface by saying that I live in Norway, since I get the feeling you do too. Some of my questions are probably easier to answer knowing that.

I've been living for a while in an apartment where an xComfort system was installed by the previous owner. As a long-time user of Home Assistant, I would love to be able to integrate our lights and switches. Your project seems like a good place to start.

As I understand it, I need to get ahold of an Eaton CKOZ-00/14 first and foremost? After that, my understanding is that I should reach out to the electrical company who installed the system, and have them export the required datapoints file.

I would consider buying a used CKOZ-00/13 if it isn't awfully complicated, as that would very likely be cheaper than hiring professionals. However - I'd have to check with the electrical company if they'd be willing to share the password required. Not sure if that's something they're typically willing to do?

Lastly - do you have any suggestions when it comes to getting ahold of a CKOZ-00/14? Would I have to order from Germany or is it possible to purchase in Norway for a reasonable price?

Reading this I realize that the title probably isn't very fitting. I initially phrased it that way to justify submitting an issue, I suppose! Let me know if there's some other way I can reach you and we could talk elsewhere.

Hopefully you find time to answer some of my questions!

karloygard, 2020-05-03:
You will need both the 00/14 and 00/13 sticks in order to control the devices properly.  The 00/13 stick is needed to add the 00/14 stick to the network, add connections, etc., and to export the datapoints file. You will need if you want to add more devices in the future, and it's not horribly expensive, so it's good to have in any case.

The 00/13 is likely able to see your current network. If it's password protected, only the installer will know the password, the electrical company will know nothing about it. There is a good chance it's not password protected, but if it is, and you can't get hold of the password, you can factory reset all devices and start afresh. In itself that's not a terrible idea, since you gain a better understanding of how the network works, but YMMV.

I purchased the sticks from https://xcomfort-sklep.pl, completely unproblematic and cheapest at that time. That was a while ago, so I can't speak for the current situation. The 00/13 stick is easily available at finn.no, and I do see the 00/14 sticks there too from time to time (if you have the patience).

The code is pretty fresh at the moment, so it may have room for improvement. Contributions are welcome!

RonnyAL, 2020-05-03:
Thanks for the swift response!

Maybe I'll have to get both sticks, then! We're planning on moving within the next couple of years so I didn't want to invest unnecessarily much here. As such, I probably won't be expanding the system. However, it does sound like I may be able to do the work myself, which would probably be cheapest anyway. Regarding the electrical company - they are indeed the installer. I should probably have made that clearer before. They actually helped me out once before because a relay was unresponsive after a power outage.

Aside from the monetary savings, what other benefits would I have from getting the 00/13? Just so you get an idea of what hardware I have, the current system consists of:

- Single ceiling lamp controlled by wireless switch.
- 27 dimmable downlights in 3 groups, controlled by remote (with LED screen) or switch. If I dim down past a point a subgroup(?) of one group turns off.

Just by adding the 00/14 to the network, what kind of "granularity" would I be able to achieve in Home Assistant? Would each relay become an entity? Or each group as defined in the xComfort system? Or something else?

I hope you don't mind my many questions. Trying to learn as much as possible so I can make some educated decisions going forward.

karloygard, 2020-05-04:
The granularity depends on how many actuators are present in your system. Since the downlights are in three groups, there are likely three dimmers. The subgroup shutting off, is likely LED lights being incompatible with the dimming profile, but this is largely guesswork on my part.

The only way to find out what you have is to detect the network with the 00/13 stick, or read the datapoints file, if you have it. In either case, this can all be done without assistance from the electrical company.

The xComfort system is pretty solid (has been very stable for me), and is the only "smart home" system that electrical companies will install for you (to my knowledge). With Home Assistant integration, there's nothing wrong with it. Without HA integration, I'd consider it kind of a dead end.

MortenLoberg, 2020-10-04:
Hi @karloygard 

Hope you don't mind me continuing this thread with some additional questions...?

Is it possible to integrate my xComfort system with Home Assistant by ONLY using the CKOZ-00/13 and CKOZ-00/14 USB sticks, or do I need the Eaton Smart Home Controller (SHC) as well? Today I have only downlights with actuators and some dimmers (all xComfort), but not the SHC.

Is it likely that the electrician (i.e. the installer) will sell me the two USB sticks? Or do I have to buy the two USB sticks from some internet-shop (for instance from https://xcomfort-sklep.pl as mentioned)?

If I buy the USB sticks, is it likely that the electrician / installer would help me to program / set-up the 00/14 stick? Or I am on my own doing that?

If I buy the USB sticks from an internet-shop, is it likely that the electrician / installer will give me the password (if a password is set on my system)?

I would appreciate some minutes of your time to answer.

karloygard, 2020-10-04:
> Is it possible to integrate my xComfort system with Home Assistant by ONLY using the CKOZ-00/13 and CKOZ-00/14 USB sticks, or do I need the Eaton Smart Home Controller (SHC) as well? Today I have only downlights with actuators and some dimmers (all xComfort), but not the SHC.

You do not need the SHC, CKOZ 00/13 and 00/14 is sufficient.

> Is it likely that the electrician (i.e. the installer) will sell me the two USB sticks? Or do I have to buy the two USB sticks from some internet-shop (for instance from https://xcomfort-sklep.pl as mentioned)?

I see no reason why they wouldn't sell you the sticks, but I wouldn't be surprised if they charge a fair bit for them. If you are in Norway, I see the USB sticks sold on finn fairly regularly (in particular the 00/14).  Elektroimportøren sells only the 00/13, apparently.

> If I buy the USB sticks, is it likely that the electrician / installer would help me to program / set-up the 00/14 stick? Or I am on my own doing that?

Maybe, but If you can set up Home Assistant, you should have no problems setting up the 00/14 stick. The MRF application caters to end users, and is easy enough to work with. In any case, getting familiar with your network and the MRF application is a good idea if you want to get the most out of your system.

> If I buy the USB sticks from an internet-shop, is it likely that the electrician / installer will give me the password (if a password is set on my system)?

I don't know why they would withhold the password from you, it's your property. Only an electrician can install the actuators connected to mains power, but you should absolutely be free to mess around with the rest of the system.

MortenLoberg, 2020-10-04:
Thanks a lot for swift response!
I'll buy the USB-sticks and try on my own first, in the worst case I'll ask the electrician to come and help me.

tom-roger, 2020-12-06:
Hi,

I am very pleased to see someone finally have developed a integration for Xcomfort. Good work!

Is it possile to have the 00/14 in parallel with the Sensio X-1? I would like to leave the X-1 as is.

karloygard, 2020-12-06:
> Is it possile to have the 00/14 in parallel with the Sensio X-1? I would like to leave the X-1 as is. 

That should work just fine, they will both act as normal services in the network. You probably can't send to or receive messages from the X-1, because it's fully programmable and don't have predefined channels etc.

tom-roger, 2020-12-19:
Hi Karl,

I have installed Home Assistant on a separate Pi 3B+ with the 00/14. I have MQTT Broker on my main Home Assistant (Pi 4).

I receive the following error in the Xcomfort log:
2020/12/19 17:21:27 Opened USB device vid=188a,pid=1101,bus=1,addr=6
2020/12/19 17:21:27 Connecting to MQTT broker 'tcp://:' with id 'xcomfort'
2020/12/19 17:21:27 Network Error : dial tcp :0: connect: connection refused

What do I need to do to get the Xcomfort Gateway to communicate with the MQTT broker on the main Home Assistant? I would really appreciate if you could help me understand how to set this up.

karloygard, 2020-12-19:
The add-on expects an MQTT broker to be running locally, that's probably what it's complaining about.

Getting it to connect to a remote broker will require some tweaking of the add-on itself. Is there a compelling reason why you are not running the add-on on your main HA?

tom-roger, 2020-12-20:
Hi Karl,

I installed MQTT broker locally. It is working excellent.

The reason for having the Xcomfort Gateway on a separate Pi is to isolate
the Zwave and Xcomfort from the main HASS. This will lead to a more stable
system as reboots are affecting stability of device gateways. I installed a
add-on called Home Assistant Remote. It uses websockets for controlling
entities on a remote device. It works perfectly.

One question: It sees that actuators are populated with the "light"
platform. Is it possible to have actuators populated as switches? I.e. my
floor heating actuators are not lights, and I would like them to populate
as switches. This makes it possible to use sensors and switches in a
thermostat.

Br,
Tom-Roger

karloygard, 2020-12-20:
I can't see from the device configuration what the actuators actually control, so at the moment they default to being light switches. The current solution is a compromise.

I think it should be possible to use the actuators in a thermostat today as well, even though they pretend to be light switches? I don't have this configuration myself, so feel free to inform me otherwise.

tom-roger, 2020-12-20:
Hi Karl,

It's working! Just did a test. This is great! Now, I finally can get rid of
Sensio!!

Great work.

T-R

tom-roger, 2020-12-20:
Hi Karl,

I keep getting these in the log:

2020/12/20 23:01:54 handle_events: error: libusb: interrupted [code -10]
2020/12/20 23:01:57 handle_events: error: libusb: interrupted [code -10]
2020/12/20 23:02:24 handle_events: error: libusb: interrupted [code -10]
2020/12/20 23:02:31 handle_events: error: libusb: interrupted [code -10]
2020/12/20 23:02:34 handle_events: error: libusb: interrupted [code -10]
2020/12/20 23:02:34 handle_events: error: libusb: interrupted [code -10]
2020/12/20 23:02:35 handle_events: error: libusb: interrupted [code -10]

Anything to worry about?

Br,
T-R

karloygard, 2020-12-21:
Happens all the time, but appears to be completely harmless, so I haven't really bothered to try to find out what causes it.

tom-roger, 2020-12-21:
Hi Karl,

Sorry for bothering you. It seems that the Xcomfort application stops
responding after approximately 1 day. A restart of the service fixes the
issue. Have you experienced this before? It's running the latest HASS on a
Rpi 3B+ with the 00/14 stick.

I'm curious if the application loses communication with the stick?

Br,
T_R

karloygard, 2020-12-21:
That shouldn't be a problem. Can you see if anything pops up in the logs?

tom-roger, 2020-12-21:
I'll check the log next time it happens and give you a notice.

Thank you for assisting!

Br,
T-R

tom-roger, 2020-12-21:
There's nothing in the Xcomfort log at the moment since I did a
reboot. However, In the supervisor log I can see that the Xcomfort
service was not responding.

20-12-21 07:06:33 INFO (MainThread) [supervisor.resolution.check]
Starting system checks with state CoreState.RUNNING20-12-21 07:06:33
INFO (MainThread) [supervisor.resolution.check] System checks
complete20-12-21 07:07:54 INFO (MainThread) [supervisor.store.git]
Update add-on https://github.com/hassio-addons/repository
repository20-12-21 07:07:54 INFO (MainThread) [supervisor.store.git]
Update add-on https://github.com/home-assistant/addons
repository20-12-21 07:07:54 INFO (MainThread) [supervisor.store.git]
Update add-on https://github.com/karloygard/hassio-addons
repository20-12-21 07:07:57 INFO (MainThread) [supervisor.store]
Loading add-ons from store: 67 all - 0 new - 0 remove20-12-21 07:07:57
INFO (MainThread) [supervisor.store] Loading add-ons from store: 67
all - 0 new - 0 remove20-12-21 07:57:06 WARNING (MainThread)
[supervisor.misc.tasks] Watchdog found a problem with
4117215b_xcomfort!

man. 21. des. 2020 kl. 10:05 skrev Tom-Roger Stensberg <tom.rste@gmail.com>:

> I'll check the log next time it happens and give you a notice.
>
> Thank you for assisting!
>
> Br,
> T-R
>
> man. 21. des. 2020 kl. 09:58 skrev Karl Anders Øygard <
>
>> That shouldn't be a problem. Can you see if anything pops up in the logs?
>>
>>
>
>
> *Mvh.*
> *Tom-Roger Stensberg*
> Mob.: +47 95817391
> E-post: tom.rste@gmail.com

karloygard, 2020-12-21:
My only idea is that the add-on is crashing. I should be able to see what's up if you catch the issue in the logs.

tom-roger, 2020-12-30:
Hi Karl,

I was able to get the application stable after some clean-up of other applications on the Rpi. I have now ported my environments over to a Ubuntu server with VirtualBox. I now run two virtual servers. One for the Main HASS, and the other for Xcomfort. 

Xcomfort application is for some reason not able to read the EPROM configuration from the stick. I am able to make it work with datafile. Am I missing something in the configuration? Any suggestions?

The application runs for only some minutes on the VM. It freezes after running some commands.

karloygard, 2020-12-31:
Can you post some logs?

closed 2021-01-10 by karloygard