karloygard/xcomfortd-go #9: Eprom issue

geeewizzz, 2020-09-23:
Tried the new version. Couldnt get eprom to work.
I blanked out the datapoints_file configuration setting, and set eprom to true.

First time i tried it said unsupported format of DPL (or something similar, I sadly didnt save the logs, i panicked :)), I then switched to using the datapoints_file with the new version, got everything cleaned up since all the entities now had changed names(!).

Tried the switch again today, changed my config to this:

> mqtt_client_id: xcomfort
> datapoints_file: ''
> eprom: true
> device_number: 0
> ha_discovery: true
> ha_discovery_prefix: homeassistant
> verbose: true

Logs says this:

> [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
> [s6-init] ensuring user provided files have correct perms...exited 0.
> [fix-attrs.d] applying ownership & permissions fixes...
> [fix-attrs.d] done.
> [cont-init.d] executing container initialization scripts...
> [cont-init.d] done.
> [services.d] starting services
> [services.d] done.
> [12:31:30] INFO: Starting xcomfortd version 0.3
> 2020/09/23 12:31:30 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
> 2020/09/23 12:31:30 Connected to broker
> 2020/09/23 12:31:30 Opened HID device 0
> 2020/09/23 12:32:11 RX: [c11762030000f600004d250f]
> 2020/09/23 12:32:11 Received message from unknown datapoint 23
> 2020/09/23 12:32:11 Message not handled [c11762030000f600004d250f]
> 2020/09/23 12:35:32 RX: [c13c62170000ec00a241240a]
> 2020/09/23 12:35:32 Received message from unknown datapoint 60
> 2020/09/23 12:35:32 Message not handled [c13c62170000ec00a241240a]

What to do?

karloygard, 2020-09-23:
I had a lot of similar issues to with the eprom processing at first, so I was pushing a lot of changes. If you updated in the middle of that, you may be running a broken version. I bumped version to 0.4 now, try reinstalling the whole thing in order to make sure that you are running the latest version.

geeewizzz, 2020-09-23:
I have uninstalled, reinstalled, "rebuild" etc several times, no matter what I do it says its running version 0.3 in the logs, while the status page of the add-on in HA says version 0.4. Not sure what to do, I'm assuming the source code is deleted when I uninstall the add-on?

> [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
> [s6-init] ensuring user provided files have correct perms...exited 0.
> [fix-attrs.d] applying ownership & permissions fixes...
> [fix-attrs.d] done.
> [cont-init.d] executing container initialization scripts...
> [cont-init.d] done.
> [services.d] starting services
> [services.d] done.
> [17:03:05] INFO: Starting xcomfortd version 0.3
> 2020/09/23 17:03:05 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
> 2020/09/23 17:03:05 Connected to broker
> 2020/09/23 17:03:05 Opened HID device 0
> 2020/09/23 17:03:05 CI HW/RF/FW revision: 0, 13.0, 0
> 2020/09/23 17:03:05 CI RF/Firmware release: 2.11, 2.05
> 2020/09/23 17:03:05 CI serial number: 7606894
> 2020/09/23 17:03:05 EPROM: [d1210000018008000000000000000000]
> 2020/09/23 17:03:06 EPROM: [d111080001800a0000000000000000000000]
> 2020/09/23 17:03:06 Unknown DPL format
> [cmd] /run.sh exited 1
> [cont-finish.d] executing container finish scripts...
> [cont-finish.d] done.
> [s6-finish] waiting for services.
> [s6-finish] sending all processes the TERM signal.

karloygard, 2020-09-23:
It is at least trying to read the eprom. I'll decode that when I get home.

karloygard, 2020-09-23:
Have you ever uploaded the DPL to the stick?

geeewizzz, 2020-09-23:
Where is the embarrassed emoji? :) I didn't even look into this, and assumed the DPL file was uploaded to the stick when I "commit changes" in the Eaton MRF tool. 

Did it now after reading a bit in the docs, restarted the add-on, first time it crashed HA, but I got it rebooted and now it seems to be working perfectly! Nice.

geeewizzz, 2020-09-25:
Sometimes after i restart HA, xcomfort fails to load, any ideas why?

> [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
> [s6-init] ensuring user provided files have correct perms...exited 0.
> [fix-attrs.d] applying ownership & permissions fixes...
> [fix-attrs.d] done.
> [cont-init.d] executing container initialization scripts...
> [cont-init.d] done.
> [services.d] starting services
> [services.d] done.
> [17:38:21] INFO: Starting xcomfortd version 0.3
> 2020/09/25 17:38:21 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
> 2020/09/25 17:38:22 Connected to broker
> 2020/09/25 17:38:22 Opened HID device 0
> 2020/09/25 17:40:44 RX: [c1007339001f0469004d0001000027000000561007]
> 2020/09/25 17:40:44 Received extended status message from unknown device 6882335
> 2020/09/25 17:40:44 Message not handled [c1007339001f0469004d0001000027000000561007]
> 2020/09/25 17:42:02 RX: [c13e62030001020000412200]
> 2020/09/25 17:42:02 Received message from unknown datapoint 62
> 2020/09/25 17:42:02 Message not handled [c13e62030001020000412200]
> 2020/09/25 17:44:06 RX: [c10c6203000111000031240e]
> 2020/09/25 17:44:06 Received message from unknown datapoint 12
> 2020/09/25 17:44:06 Message not handled [c10c6203000111000031240e]
> 2020/09/25 17:47:10 RX: [c115620300010300004a250e]
> 2020/09/25 17:47:10 Received message from unknown datapoint 21
> 2020/09/25 17:47:10 Message not handled [c115620300010300004a250e]
> 2020/09/25 17:47:12 RX: [c13562040042a866663f2401]
> 2020/09/25 17:47:12 Received message from unknown datapoint 53
> 2020/09/25 17:47:12 Message not handled [c13562040042a866663f2401]
> 2020/09/25 17:47:22 RX: [c13662170000d100f631240a]
> 2020/09/25 17:47:22 Received message from unknown datapoint 54
> 2020/09/25 17:47:22 Message not handled [c13662170000d100f631240a]
> 2020/09/25 17:47:26 RX: [c109620400426acccd312409]
> 2020/09/25 17:47:26 Received message from unknown datapoint 9
> 2020/09/25 17:47:26 Message not handled [c109620400426acccd312409]
> 2020/09/25 17:48:02 RX: [c13e62030001070000422201]
> 2020/09/25 17:48:02 Received message from unknown datapoint 62
> 2020/09/25 17:48:02 Message not handled [c13e62030001070000422201]
> 2020/09/25 17:49:22 RX: [c13b620400429b3333422206]
> 2020/09/25 17:49:22 Received message from unknown datapoint 59
> 2020/09/25 17:49:22 Message not handled [c13b620400429b3333422206]
> 2020/09/25 17:49:22 RX: [c13f62030000f8000031240c]
> 2020/09/25 17:49:22 Received message from unknown datapoint 63
> 2020/09/25 17:49:22 Message not handled [c13f62030000f8000031240c]
> 2020/09/25 17:50:46 RX: [c10162170000f400de312400]
> 2020/09/25 17:50:46 Received message from unknown datapoint 1
> 2020/09/25 17:50:46 Message not handled [c10162170000f400de312400]
> 2020/09/25 17:52:34 RX: [c11762030000e1000049250e]
> 2020/09/25 17:52:34 Received message from unknown datapoint 23
> 2020/09/25 17:52:34 Message not handled [c11762030000e1000049250e]
> 2020/09/25 17:53:21 RX: [c13d62040042786666422403]
> 2020/09/25 17:53:21 Received message from unknown datapoint 61
> 2020/09/25 17:53:21 Message not handled [c13d62040042786666422403]
> 2020/09/25 17:54:02 RX: [c13a62170000e500c1432202]
> 2020/09/25 17:54:02 Received message from unknown datapoint 58
> 2020/09/25 17:54:02 Message not handled [c13a62170000e500c1432202]
> 2020/09/25 17:54:32 RX: [c13462170000d200003e2400]
> 2020/09/25 17:54:32 Received message from unknown datapoint 52
> 2020/09/25 17:54:32 Message not handled [c13462170000d200003e2400]
> 2020/09/25 17:54:48 RX: [c108700000000000004a1003]
> 2020/09/25 17:54:48 Received message from unknown datapoint 8
> 2020/09/25 17:54:48 Message not handled [c108700000000000004a1003]
> 2020/09/25 17:54:50 RX: [c10870001700000000491006]
> 2020/09/25 17:54:50 Received message from unknown datapoint 8
> 2020/09/25 17:54:50 Message not handled [c10870001700000000491006]

karloygard, 2020-09-25:
I'll add some more debug logging, it's not obvious what's going wrong.

karloygard, 2020-09-25:
Pushed a new image with more logging now (0.5), give it a shot when you have time. I expect this to be a sporadic issue, so you may have to run it several times in order to provoke the issue. I'll do the same on my part.

geeewizzz, 2020-09-29:
Upgraded to 0.5 today, first start provoked the issue. Worked fine after restart.

> [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
> [s6-init] ensuring user provided files have correct perms...exited 0.
> [fix-attrs.d] applying ownership & permissions fixes...
> [fix-attrs.d] done.
> [cont-init.d] executing container initialization scripts...
> [cont-init.d] done.
> [services.d] starting services
> [services.d] done.
> [21:09:45] INFO: Starting xcomfortd version 0.4
> [21:09:45] INFO: xcomfortd usb --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --device-number 0 --hadiscoveryprefix homeassistant --eprom --hadiscovery --verbose
> 2020/09/29 21:09:45 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
> 2020/09/29 21:09:45 Connected to broker
> 2020/09/29 21:09:45 Opened HID device 0
> 2020/09/29 21:09:45 CONFIG: [04b21b10]

karloygard, 2020-09-30:
Kind of as suspected. Configuration of the stick stalls for some reason, causing the datapoint list not to be loaded. I have to inspect why the configuration stalls. Looks like the stick is not responding to a request, but it could be my code that's broken as well.

geeewizzz, 2020-10-21:
I'm still seeing this issue half of the time i restart. Do you have a good suggestion on how to find a work around? So that I can reboot it by some automation or something.

I was thinking "if entityid = unavailable for 30mins, restart add-on, wait 5mins, repeat until available (max 5 retries or something)" 

Suggestion for the code is that it terminates itself if eprom has been set to true in config and it actually doesnt read it in the startup, or something similar.

karloygard, 2020-10-21:
I'll put some extra effort into resolving this.

geeewizzz, 2020-10-21:
Thanks a lot! I'd buy you a beer if i could, and if you are looking for a job at some point and want to work in media industry in norway we are looking for good go developers. I'm sure you'll find a way to get in touch with me in that case. :)

karloygard, 2020-10-21:
Still not sure why this is happening, to be honest, but I believe 931119c8bebb67a27754c19efa0770fb2dcff2d0 resolves your issue. Code is present in version 0.11.

closed 2020-10-22 by geeewizzz