karloygard/xcomfortd-go #10: Terminal error

geeewizzz, 2020-10-07:
Had a crash today when I ran an automation to adjust temp on all my floor heaters. Log below.

> 2020/10/07 14:23:25 Device 6784078 (channel 6-'1.etg dusj/bad (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 24.7
> 2020/10/07 14:24:59 RX: [c13462170000d200003c2408]
> 2020/10/07 14:24:59 dropping wheel position on the ground: 0.0
> 2020/10/07 14:24:59 Device 6809944 (channel 0-'kjeller bad temp (Temperatur)') sent message (battery good, signal good, cyclic true) event 'value' with value 21
> 2020/10/07 14:25:55 RX: [c10162170000ee00de30240e]
> 2020/10/07 14:25:55 dropping wheel position on the ground: 22.2
> 2020/10/07 14:25:55 Device 6784078 (channel 0-'1.etg dusj/bad (Temperatur)') sent message (battery good, signal good, cyclic true) event 'value' with value 23.8
> 2020/10/07 14:33:04 RX: [c13a62170000ce00c1422204]
> 2020/10/07 14:33:04 dropping wheel position on the ground: 19.3
> 2020/10/07 14:33:04 Device 6807667 (channel 0-'kjeller vaskerom temp (Temperatur)') sent message (battery very weak, signal good, cyclic true) event 'value' with value 20.6
> 2020/10/07 14:33:15 RX: [c13f62030000e40000302406]
> 2020/10/07 14:33:15 Device 6783968 (channel 6-'kjeller inngang temp (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 22.8
> 2020/10/07 14:33:47 RX: [c14062030000ee0000432409]
> 2020/10/07 14:33:47 Device 7417943 (channel 6-'hybel bad temp (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 23.8
> 2020/10/07 14:34:59 RX: [c133620300ffff00003d240b]
> 2020/10/07 14:34:59 Device 6809944 (channel 6-'kjeller bad temp (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value -0.1
> 2020/10/07 14:36:22 RX: [c10f620300010e0000502508]
> 2020/10/07 14:36:22 Device 5881341 (channel 0-'1.etg stor gang temp (Kanal A)') sent message (battery new, signal weak, cyclic true) event 'value' with value 27
> 2020/10/07 14:37:04 RX: [c13e62030000f0000042220d]
> 2020/10/07 14:37:04 Device 6807667 (channel 6-'kjeller vaskerom temp (ekstern Sensor (PT1000))') sent message (battery very weak, signal good, cyclic true) event 'value' with value 24
> 2020/10/07 14:37:15 RX: [c1376204004280cccd30240f]
> 2020/10/07 14:37:15 Device 6783968 (channel 1-'kjeller inngang temp (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 64.4
> 2020/10/07 14:37:47 RX: [c13c62170000f500a241240f]
> 2020/10/07 14:37:47 dropping wheel position on the ground: 16.2
> 2020/10/07 14:37:47 Device 7417943 (channel 0-'hybel bad temp (Temperatur)') sent message (battery good, signal good, cyclic true) event 'value' with value 24.5
> 2020/10/07 14:40:40 RX: [c11762030000e800004d2508]
> 2020/10/07 14:40:40 Device 5881353 (channel 1-'1.etg liten gang temp (Kanal B)') sent message (battery new, signal weak, cyclic true) event 'value' with value 23.2
> 2020/10/07 14:45:05 RX: [c1096204004284666630240f]
> 2020/10/07 14:45:05 Device 6784078 (channel 1-'1.etg dusj/bad (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 66.2
> 2020/10/07 14:46:19 RX: [c13562040042bbcccd3c2408]
> 2020/10/07 14:46:19 Device 6809944 (channel 1-'kjeller bad temp (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 93.9
> 2020/10/07 14:47:04 RX: [c13e62030000ea000041220e]
> 2020/10/07 14:47:04 Device 6807667 (channel 6-'kjeller vaskerom temp (ekstern Sensor (PT1000))') sent message (battery very weak, signal good, cyclic true) event 'value' with value 23.4
> 2020/10/07 14:47:15 RX: [c13662170000c900f6302401]
> 2020/10/07 14:47:15 dropping wheel position on the ground: 24.6
> 2020/10/07 14:47:15 Device 6783968 (channel 0-'kjeller inngang temp (Temperatur)') sent message (battery good, signal good, cyclic true) event 'value' with value 20.1
> 2020/10/07 14:47:43 RX: [c11562030000ff00004c2505]
> 2020/10/07 14:47:43 Device 5881271 (channel 0-'1.etg bad temp (Kanal A)') sent message (battery new, signal weak, cyclic true) event 'value' with value 25.5
> 2020/10/07 14:51:55 RX: [c13f62030000e60000302407]
> 2020/10/07 14:51:55 Device 6783968 (channel 6-'kjeller inngang temp (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 23
> 2020/10/07 14:52:24 RX: [c13b62040042946666422203]
> 2020/10/07 14:52:24 Device 6807667 (channel 1-'kjeller vaskerom temp (Fuktighet)') sent message (battery very weak, signal good, cyclic true) event 'value' with value 74.2
> 2020/10/07 14:52:27 RX: [c13d620400427d999a422409]
> 2020/10/07 14:52:27 Device 7417943 (channel 1-'hybel bad temp (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 63.4
> 2020/10/07 14:53:25 RX: [c10c62030000f80000302402]
> 2020/10/07 14:53:25 Device 6784078 (channel 6-'1.etg dusj/bad (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 24.8
> 2020/10/07 14:54:24 RX: [c1197000620000000059100b]
> 2020/10/07 14:54:24 Device 5740172 (channel 0-'1.etg dusj/bad') sent message (battery mains-powered, signal weak, cyclic false) value 98
> 2020/10/07 14:54:59 RX: [c13462170000d200003c2409]
> 2020/10/07 14:54:59 dropping wheel position on the ground: 0.0
> 2020/10/07 14:54:59 Device 6809944 (channel 0-'kjeller bad temp (Temperatur)') sent message (battery good, signal good, cyclic true) event 'value' with value 21
> 2020/10/07 14:55:55 MQTT message; topic: 'xcomfort/20/set/switch', message: 'true'
> 2020/10/07 14:55:55 STATUS: [c31c0480100000]
> 2020/10/07 14:55:55 RX: [1c0480100000]
> 2020/10/07 14:55:55 RX: [c10162170000ed00de30240f]
> 2020/10/07 14:55:55 dropping wheel position on the ground: 22.2
> 2020/10/07 14:55:55 Device 6784078 (channel 0-'1.etg dusj/bad (Temperatur)') sent message (battery good, signal good, cyclic true) event 'value' with value 23.7
> 2020/10/07 14:55:55 RX: [c11470000100000000481002]
> 2020/10/07 14:55:55 Device 5727267 (channel 0-'1.etg dusj/bad vk') sent message (battery mains-powered, signal normal, cyclic false) status switched on
> 2020/10/07 14:55:56 MQTT message; topic: 'xcomfort/24/set/switch', message: 'true'
> 2020/10/07 14:55:57 STATUS: [c31c0490100000]
> 2020/10/07 14:55:57 RX: [1c0490100000]
> 2020/10/07 14:55:57 RX: [c118700001000000004b1005]
> 2020/10/07 14:55:57 Device 5733583 (channel 0-'1.etg stor gang vk') sent message (battery mains-powered, signal normal, cyclic false) status switched on
> 2020/10/07 14:55:58 MQTT message; topic: 'xcomfort/16/set/switch', message: 'true'
> 2020/10/07 14:55:59 RX: []
> 2020/10/07 14:55:59 WARNING: command for datapoint 16 failed, state now unknown: Terminal error
> 2020/10/07 14:55:59 read failed: hid: device closed
> github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).Run.func1
>         /xcomfortd-go/pkg/xc/loop.go:25
> runtime.goexit
>         /usr/local/go/src/runtime/asm_arm.s:857
> 2020/10/07 14:55:59 MQTT message; topic: 'xcomfort/39/set/switch', message: 'true'
> 2020/10/07 14:56:06 Sent MQTT autodiscover remove for 47 devices and 68 datapoints
> 2020/10/07 14:56:06 hidapi: unknown failure
> github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).Run
>         /xcomfortd-go/pkg/xc/loop.go:58
> main.Hid
>         /xcomfortd-go/hid.go:28
> main.start
>         /xcomfortd-go/main.go:208
> main.hidCommand
>         /xcomfortd-go/main.go:131
> github.com/urfave/cli.HandleAction
>         /go/pkg/mod/github.com/urfave/cli@v1.22.4/app.go:526
> github.com/urfave/cli.Command.Run
>         /go/pkg/mod/github.com/urfave/cli@v1.22.4/command.go:174
> github.com/urfave/cli.(*App).Run
>         /go/pkg/mod/github.com/urfave/cli@v1.22.4/app.go:279
> main.main
>         /xcomfortd-go/main.go:120
> runtime.main
>         /usr/local/go/src/runtime/proc.go:203
> runtime.goexit
>         /usr/local/go/src/runtime/asm_arm.s:857
> [cmd] /run.sh exited 1
> [cont-finish.d] executing container finish scripts...
> [cont-finish.d] done.
> [s6-finish] waiting for services.
> [s6-finish] sending all processes the TERM signal.

karloygard, 2020-10-07:
I have observed similar issues with hidapi, but I cannot see what the issue is, the actual error message is `hidapi: unknown failure`. I suspect there is an issue with hidapi itself.

I have played with using libusb directly instead, but had difficulties building hassio images with it. The code is in usb.go and works, but I had to comment it out for now.  I'll mess around with it a bit more and see if I can get it to build properly. I at least believe libusb to be more resilient.

karloygard, 2020-10-10:
0.6 uses libusb instead of hidapi. Please give it a shot and see if it performs better.

The start up issue is not yet resolved, by the way.

closed 2020-10-10 by karloygard