karloygard/xcomfortd-go #12: Feature request: Humidity sensors

geeewizzz, 2020-11-27:
Humidity data is coming to the usb stick from the RCT devices. Could you please create sensors for them?

>    2020/11/27 08:03:17 Device 6784078 (channel 6-'1.etg dusj/bad (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 28.3
2020/11/27 08:05:16 RX: [c1376204004252cccd412408]
2020/11/27 08:05:16 Device 6783968 (channel 1-'kjeller inngang temp (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 52.7
2020/11/27 08:05:47 RX: [c10c620300011b00003c2404]
2020/11/27 08:05:47 Device 6784078 (channel 6-'1.etg dusj/bad (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 28.3
2020/11/27 08:13:22 Device 5881271 (channel 0-'1.etg bad temp (Kanal A)') sent message (battery new, signal normal, cyclic true) event 'value' with value 27
2020/11/27 08:13:37 RX: [c1356204004264cccd3a2403]
2020/11/27 08:13:37 Device 6809944 (channel 1-'kjeller bad temp (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 57.2
2020/11/27 08:14:01 RX: [c13b62040042803333402202]
2020/11/27 08:14:01 Device 6807667 (channel 1-'kjeller vaskerom temp (Fuktighet)') sent message (battery very weak, signal good, cyclic true) event 'value' with value 64.1
2020/11/27 08:14:07 RX: [c10c620300011300003b2405]
2020/11/27 08:14:07 Device 6784078 (channel 6-'1.etg dusj/bad (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 27.5
2020/11/27 08:15:50 Device 5881353 (channel 1-'1.etg liten gang temp (Kanal B)') sent message (battery new, signal good, cyclic true) event 'value' with value 19.6
2020/11/27 08:17:27 RX: [c109620400425ccccd39240c]
2020/11/27 08:17:27 Device 6784078 (channel 1-'1.etg dusj/bad (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 55.2
2020/11/27 08:18:26 RX: [c13d620400423a00004d240e]
2020/11/27 08:18:26 Device 7417943 (channel 1-'hybel bad temp (Fuktighet)') sent message (battery good, signal weak, cyclic true) event 'value' with value 46.5
2020/11/27 08:20:41 RX: [c13e6203000107000042220e]
2020/11/27 08:20:41 Device 6807667 (channel 6-'kjeller vaskerom temp (ekstern Sensor (PT1000))') sent message (battery very weak, signal good, cyclic true) event 'value' with value 26.3
2020/11/27 08:33:45 Device 6881943, type CDAU 01/04 sent extended status message: value 0, temp 32C (battery mains-powered, signal weak)
2020/11/27 08:35:16 RX: [c1376204004251999a422409]
2020/11/27 08:35:16 Device 6783968 (channel 1-'kjeller inngang temp (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 52.4
2020/11/27 08:35:56 RX: [c13f62030000e600003b2404]
2020/11/27 08:35:56 Device 6783968 (channel 6-'kjeller inngang temp (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 23

karloygard, 2020-11-28:
👍🏻 Will take a look.

karloygard, 2020-11-28:
I added support for the humidity channel of the RCT, give it a shot and see if it works.

closed 2020-11-28 by karloygard

geeewizzz, 2020-11-28:
Wow, that was quick. Works like a charm, thanks!

[image]