karloygard/xcomfortd-go #13: MQTT Autodiscovery issues

Jangomnz, 2021-01-03:
Hi
Not sure if this belongs here or in the hassio-addon part, or if this is really an issue with the xcomfort MQTT gateway at all.

After a month of downtime of my home assistant server (disk failure), I have retsored and updated to the latest version of all addons (including this).  To my surprise the entire setup of xcomfort devices in Home Assistant was also deleted with accosiated added information with area links etc, and replaced with new dicovery data.  If you update the datapoint list to eprom the module calls:
2021/01/03 23:06:46 Read datapoint list from eprom in 1m19.614060518s
2021/01/03 23:06:46 DPL Changed
2021/01/03 23:06:52 Sent MQTT autodiscover remove for 53 devices and 62 datapoints
2021/01/03 23:06:52 Sent MQTT autodiscover add for 53 devices and 62 datapoints

This seems to scratch the entire device configuration in HA, and with 51 devices and 115 entities with updated information, this is a showstopper (I can't have to update the entire configuration every time the datapoint list is updated).  This did not seem to be an issue in previous versions, but I'm no 100% sure.  I don't see any updates to the MQTT integration, so I assumed this is something that is changed in this addon.  

Have I overlooked an obvious setting?

Jan

karloygard, 2021-01-03:
I wonder if this is caused by the `MQTT autodiscover remove` part. Does this happen on a full restart of hassio as well?

I can make autodiscover remove optional, but I believe this would require you to manually remove old devices. I'm not 100% informed when it comes to the mechanics of MQTT autodiscovery.

Jangomnz, 2021-01-04:
Yeah, I think this is the issue.  The problem is that this seems to brute force remove all devices added, even if they have been updated with added information in Home Assistant.   I would be fine with removing old entries on my own.  How would MQTT autodiscover add behave if the devices already have been added?

Jangomnz, 2021-01-04:
Just tested with a HA restart:
This only calls:
2021/01/04 09:17:18 Read datapoint list from eprom in 1m19.550166461s
2021/01/04 09:17:18 Sent MQTT autodiscover add for 53 devices and 62 datapoints

This does not delete all the devices, nor make changes to updated devices.

I've not updated to you lates rev. 0.24.  This was tested on 0.23.  I'll update to the latest rev and  test this again.

Regards

Jan

karloygard, 2021-01-04:
Okay, I'll make autodiscover remove optional, and we'll see how that fares. I'll see if I can come up with a better solution.

hans99, 2021-01-04:
I would like to see autodiscover remove optional in any case, please 🥇

Jangomnz, 2021-01-04:
Thanks for the quick implementation, but I can't get the new version (0.25) to accept the config (even when reset to defaults):

Configuration:
mqtt_client_id: xcomfort
datapoints_file: ''
eprom: True
ha_discovery: True
ha_discovery_prefix: homeassistant
ha_discovery_remove: False
verbose: False
use_hidapi: False
eci_hosts: []
​
Failed to save addon configuration, not a valid value for dictionary value @ data['options']. Got {'mqtt_client_id': 'xcomfort', 'datapoints_file': '', 'eprom': True, 'ha_discovery': True, 'ha_discovery_prefix': 'homeassistant', 'ha_discovery_remove': False, 'verbose': False, 'use_hidapi': False, 'eci_hosts': []}

Regards

karloygard, 2021-01-04:
I screwed up the configuration for a short while, install 0.26.

karloygard, 2021-01-04:
Let me know how if 0.26 fixes the issue, so I can close this.

Jangomnz, 2021-01-05:
0.26 seems to works as intentended.  Autodicover remove is not called when set to false in config.

Thanks

karloygard, 2021-01-05:
Cool, let me know if you see any other issues!

closed 2021-01-05 by karloygard