karloygard/xcomfortd-go #14: Xcomfort add-on is not responding to commands after several days

tom-roger, 2021-01-08:
Hi,

It is now several days since I restarted the Xcomfort add-on, and last night it stopped responding. After taking a brief look at the logs, I am not able to see anything suspicious. Everything looks normal. I have enabled verbose logging. The logs shows normal activity and communications (same as before the application stopped responding). Nothing suspicious in the Home Assistant logs as well.

A simple restart of Home Assistant application fixes the issue.

- Rpi 4
- Xcomfort ver. 0.18
- Home Assistant OS 5.8
- Home Assistant 2020.12.7

tom-roger, 2021-01-08:
[image]

karloygard, 2021-01-08:
Please upgrade to 0.26. There were a few issues with exactly this that were fixed in the last weeks, hopefully your issue is already resolved.

tom-roger, 2021-01-08:
Hi Karl,

Thank you  very much. I will upgrade. I have noticed that new versions have been release, but I was not sure what was fixed. Will you write some lines in the future about what is fixed in future releases? I really enjoy this add-on. Now I am finally able to develop the smart home I was dreaming of. 

PS! I have some Eaton thermostats with humidity sensors. I can see that the stick is reading the value of these sensors, but the add-on does not read the values. Are you planning for supporting this in the future? Also, the wheel (+/- 3 gr. C) is not read.

Keep up the good work!

T-R

karloygard, 2021-01-08:
I'll consider adding a basic changelog in the future. Until now, there hasn't really been enough users to warrant the effort.

RCT with humidity sensor *should* be supported, since 0.17. Can you check the logs and see if it's complaining about something? It's best to turn off verbose, since it just logs way too much that way. The wheel setting is currently dropped on the ground, since I don't know what to use it for. I don't have one of these devices myself, so I haven't investigated them too closely.

dreeas, 2021-02-27:
Hi Karl,

First, thank you so much for creating this addon. I have been looking for a way out to other controllers for my xcomfort equipment for a long time. Really a cause worth supporting as well.

As for the issue, I am experiencing the same as Tom-Roger above here. 
Just did a fresh install on a Rpi3b+ only for testing out this addon. Got everything up and running on latest versions as of today and all was working good for about an hour. Then I noticed that no commands can be sent, no xcomfort actuators can be switched. Still receiving switch status signals and temperature readings from thermostats.
And also no apparent errors in any logs, either xcomfort addon or mosquitto broker addon. 

After restarting Home Assistant everything is seemingly OK receiving data but still not able to send commands. 
It is only after I restart the xcomfort addon that I can send commands. This lasts for a while then commands stop working again.
Find this strange and talked with others using this addon and no one experienced this before.

[image]

dreeas, 2021-02-27:
I see there is an ongoing case on the hassio addon page - https://github.com/karloygard/hassio-addons/issues/12
Sorry for doubleposting. I can follow and comment on that one.

br
Andreas

karloygard, 2021-03-01:
Closing this, followups in https://github.com/karloygard/hassio-addons/issues/12

closed 2021-03-01 by karloygard