karloygard/xcomfortd-go #16: "Stick didn't respond after five seconds, retrying command" error

havardthom, 2021-02-06:
Hi, I'm trying to start the Hassio addon but getting a "Stick didn't respond after five seconds, retrying command". I tried connecting the CKOZ-00/14 stick to multiple USB ports and it is discovered by the Eaton RF system, so the stick should be fine. I also loaded a datapoint file to the CKOZ-00/14 eprom as described in the RF manual docs. Any idea what could be the problem?

```
[17:39:41] INFO: Starting xcomfortd version 0.27
[17:39:41] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --eprom --hadiscovery --verbose
2021/02/06 17:39:41 Opened USB device vid=188a,pid=1101,bus=1,addr=11
2021/02/06 17:39:41 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
2021/02/06 17:39:41 Connected to broker
2021/02/06 17:39:41 CONFIG: [b21b10]
2021/02/06 17:39:46 Stick didn't respond after five seconds, retrying command
2021/02/06 17:39:46 CONFIG: [b21b10]
2021/02/06 17:39:46 read failed: device sent more data than requested
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).Run.func1
	/xcomfortd-go/pkg/xc/loop.go:28
runtime.goexit
	/usr/local/go/src/runtime/asm_arm64.s:1148
2021/02/06 17:39:46 Exiting
2021/02/06 17:39:46 Terminal error
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).sendConfigCommand
	/xcomfortd-go/pkg/xc/loop.go:227
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).Revision
	/xcomfortd-go/pkg/xc/config_messages.go:29
main.run.func1
	/xcomfortd-go/main.go:163
runtime.goexit
	/usr/local/go/src/runtime/asm_arm64.s:1148
[cmd] /run.sh exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

[image]

karloygard, 2021-02-06:
It looks like you are running a obsolete version of xcomfort-go, the error message `device sent more data than requested` was retired a while ago. Not sure how that happened, can you uninstall and reinstall the add-on and see if that resolves your issue? I've pushed 0.28, which also logs the version of the binary, just to be sure.

havardthom, 2021-02-06:
Thank you for the quick response, here are the logs after reinstalling. It unfortunately did not resolve the issue.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[19:47:16] INFO: Starting xcomfortd version 0.28
[19:47:16] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --eprom --hadiscovery --verbose
2021/02/06 19:47:16 Starting xcomfortd, version 0.28
2021/02/06 19:47:16 Opened USB device vid=188a,pid=1101,bus=1,addr=12
2021/02/06 19:47:16 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
2021/02/06 19:47:16 handle_events: error: libusb: interrupted [code -10]
2021/02/06 19:47:18 Connected to broker
2021/02/06 19:47:18 CONFIG: [b21b10]
2021/02/06 19:47:20 handle_events: error: libusb: interrupted [code -10]
2021/02/06 19:47:22 handle_events: error: libusb: interrupted [code -10]
2021/02/06 19:47:22 handle_events: error: libusb: interrupted [code -10]
2021/02/06 19:47:23 Stick didn't respond after five seconds, retrying command
2021/02/06 19:47:23 CONFIG: [b21b10]
2021/02/06 19:47:28 Stick didn't respond after five seconds, retrying command
2021/02/06 19:47:28 CONFIG: [b21b10]
2021/02/06 19:47:28 read failed: device sent more data than requested
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).Run.func1
	/xcomfortd-go/pkg/xc/loop.go:28
runtime.goexit
	/usr/local/go/src/runtime/asm_arm64.s:1148
2021/02/06 19:47:28 Exiting
2021/02/06 19:47:28 Terminal error
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).sendConfigCommand
	/xcomfortd-go/pkg/xc/loop.go:227
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).Revision
	/xcomfortd-go/pkg/xc/config_messages.go:29
main.run.func1
	/xcomfortd-go/main.go:169
runtime.goexit
	/usr/local/go/src/runtime/asm_arm64.s:1148
[cmd] /run.sh exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

karloygard, 2021-02-06:
Okay, my initial theory was incorrect, your install was valid after all.

I suspect the firmware on your USB stick needs to be updated. But first, try setting `use_hidapi: true` in the configuration and see if the stick will respond better that way. Grab the log in any case. Revert to `use_hidapi: false` afterwards, hidapi has stability issues and is not a viable option in the long term.

havardthom, 2021-02-06:
That is very possible, I bought the stick second hand and have not updated the firmware on it yet. I will try that.

Here are the logs with hidapi on

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[20:20:17] INFO: Starting xcomfortd version 0.28
[20:20:17] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --eprom --hadiscovery --verbose --hidapi
2021/02/06 20:20:17 Starting xcomfortd, version 0.28
2021/02/06 20:20:17 Opened HID device 0
2021/02/06 20:20:17 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
2021/02/06 20:20:19 Connected to broker
2021/02/06 20:20:19 CONFIG: [b21b10]
2021/02/06 20:20:24 Stick didn't respond after five seconds, retrying command
2021/02/06 20:20:24 CONFIG: [b21b10]
2021/02/06 20:20:24 STATUS: [c31b1000820000]
2021/02/06 20:20:24 CI HW/RF/FW revision: 0, 13.0, 0
2021/02/06 20:20:24 CONFIG: [b21b00]
2021/02/06 20:20:24 STATUS: [c31b0001080104]
2021/02/06 20:20:24 CI RF/Firmware release: 1.08, 1.04
2021/02/06 20:20:24 CONFIG: [b20e00]
2021/02/06 20:20:24 STATUS: [c30e000054b44e]
2021/02/06 20:20:24 CI serial number: 5551182
2021/02/06 20:20:24 Reading datapoints list from eprom
2021/02/06 20:20:25 EPROM: [d121000001800800020817d21e600100]
2021/02/06 20:20:25 EPROM: [d111080001800a0000030078400000000000]
2021/02/06 20:20:25 EPROM: [d111100001800a0001009065750000000000]
2021/02/06 20:20:25 EPROM: [d1111a0001800a000040780000008b000000]
2021/02/06 20:20:25 EPROM: [d111240001800a0077004d00000000000000]
```

karloygard, 2021-02-06:
As suspected, 1.08, 1.04 is pretty old, not sure exactly how old, but I'm guessing dating back to 2015 or so. 2.11, 2.05 is the latest that I am aware of. Upgrade and you are golden.

havardthom, 2021-02-06:
Upgraded to 2.11/2.05 and it fixed the issue. Thank you very much for helping and thank you for the opportunity to use xComfort with HA!

closed 2021-02-06 by havardthom