karloygard/xcomfortd-go #17: USB stick not found

geeewizzz, 2021-02-09:
Hi, thank you for your efforts in maintaining this, its greatly appreciated. After upgrading to the newest version of HA, Supervisor, and upgrading the xcomfort plugin to the latest version I'm struggling to open the USB device and the xcomfort plugin is exiting on init. Here are some logs. I've tried with both hidapi and libusb

> pi@raspberrypi:~ $ dmesg
[  331.633295] hassio: port 8(vethf693360) entered blocking state
[  331.633312] hassio: port 8(vethf693360) entered disabled state
[  331.633913] device vethf693360 entered promiscuous mode
[  331.849349] IPv6: ADDRCONF(NETDEV_CHANGE): veth14f09eb: link becomes ready
[  331.849732] IPv6: ADDRCONF(NETDEV_CHANGE): vethf693360: link becomes ready
[  331.849843] hassio: port 8(vethf693360) entered blocking state
[  331.849853] hassio: port 8(vethf693360) entered forwarding state
[  333.286531] hassio: port 8(vethf693360) entered disabled state
[  333.300277] eth0: renamed from veth14f09eb
[  333.337662] hassio: port 8(vethf693360) entered blocking state
[  333.337678] hassio: port 8(vethf693360) entered forwarding state
[  339.529023] hassio: port 8(vethf693360) entered disabled state
[  339.536672] veth14f09eb: renamed from eth0
[  339.711622] hassio: port 8(vethf693360) entered disabled state
[  339.730829] device vethf693360 left promiscuous mode
[  339.730857] hassio: port 8(vethf693360) entered disabled state
[  364.191696] hassio: port 8(vethc50eee1) entered blocking state
[  364.191716] hassio: port 8(vethc50eee1) entered disabled state
[  364.192284] device vethc50eee1 entered promiscuous mode
[  364.413713] IPv6: ADDRCONF(NETDEV_CHANGE): veth9efb412: link becomes ready
[  364.413987] IPv6: ADDRCONF(NETDEV_CHANGE): vethc50eee1: link becomes ready
[  364.414068] hassio: port 8(vethc50eee1) entered blocking state
[  364.414077] hassio: port 8(vethc50eee1) entered forwarding state
[  365.696819] hassio: port 8(vethc50eee1) entered disabled state
[  365.699421] eth0: renamed from veth9efb412
[  365.768789] hassio: port 8(vethc50eee1) entered blocking state
[  365.768805] hassio: port 8(vethc50eee1) entered forwarding state
[  371.938847] hassio: port 8(vethc50eee1) entered disabled state
[  371.939997] veth9efb412: renamed from eth0
[  372.088627] hassio: port 8(vethc50eee1) entered disabled state
[  372.111617] device vethc50eee1 left promiscuous mode
[  372.111647] hassio: port 8(vethc50eee1) entered disabled state
[  374.988116] usb 1-1.2: USB disconnect, device number 4
[  377.595925] usb 1-1.4: new full-speed USB device number 5 using dwc_otg
[  377.740280] usb 1-1.4: New USB device found, idVendor=188a, idProduct=1101, bcdDevice= 2.05
[  377.740296] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  377.740305] usb 1-1.4: Product: EATON Usb CI Stick    
[  377.740312] usb 1-1.4: Manufacturer: EATON
[  377.740320] usb 1-1.4: SerialNumber: 0007606894
[  377.748343] hid-generic 0003:188A:1101.0002: hiddev96,hidraw0: USB HID v1.11 Device [EATON EATON Usb CI Stick    ] on usb-3f980000.usb-1.4/input0
pi@raspberrypi:~ $ lsusb
Bus 001 Device 005: ID 188a:1101  
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Configuration:
>mqtt_client_id: xcomfort
datapoints_file: ''
eprom: true
ha_discovery: true
ha_discovery_prefix: homeassistant
ha_discovery_remove: false
verbose: true
use_hidapi: false
eci_hosts: []

Log
>[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[08:57:50] INFO: Starting xcomfortd version 0.28
[08:57:50] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --eprom --hadiscovery --verbose
2021/02/09 08:57:50 Starting xcomfortd, version 0.28
2021/02/09 08:57:50 libusb: i/o error [code -1]
[cmd] /run.sh exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

Changed to use hidapi
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[08:59:34] INFO: Starting xcomfortd version 0.28
[08:59:34] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --eprom --hadiscovery --verbose --hidapi
2021/02/09 08:59:34 Starting xcomfortd, version 0.28
2021/02/09 08:59:34 hidapi: failed to open device
main.openHidDevices
	/xcomfortd-go/hid.go:18
main.openDevices
	/xcomfortd-go/main.go:103
github.com/urfave/cli/v2.(*App).RunContext
	/go/pkg/mod/github.com/urfave/cli/v2@v2.3.0/app.go:322
github.com/urfave/cli/v2.(*App).Run
	/go/pkg/mod/github.com/urfave/cli/v2@v2.3.0/app.go:224
main.main
	/xcomfortd-go/main.go:77
runtime.main
	/usr/local/go/src/runtime/proc.go:203
runtime.goexit
	/usr/local/go/src/runtime/asm_arm.s:857
[cmd] /run.sh exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

Any suggestions?

geeewizzz, 2021-02-09:
[image]

geeewizzz, 2021-02-09:
Reinstalled the whole thing on hassOS, so that I'm now on a supported OS. Getting the same issue.

Here is the hardware output from the "Host" if that is interesting. I can see that it sees the USB down there at least. The stick itself lights up green and flickers when I hit a light switch for example.

devices:
  - name: gpiochip1
    sysfs: /sys/devices/gpiochip1
    dev_path: /dev/gpiochip1
    subsystem: gpio
    by_id: null
    attributes:
      DEVNAME: /dev/gpiochip1
      DEVPATH: /devices/gpiochip1
      MAJOR: '254'
      MINOR: '1'
      OF_COMPATIBLE_0: 'brcm,bcm2835-virtgpio'
      OF_COMPATIBLE_N: '1'
      OF_FULLNAME: /soc/virtgpio
      OF_NAME: virtgpio
      SUBSYSTEM: gpio
  - name: media0
    sysfs: /sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/media0
    dev_path: /dev/media0
    subsystem: media
    by_id: null
    attributes:
      DEVNAME: /dev/media0
      DEVPATH: /devices/platform/soc/3f00b840.mailbox/bcm2835-codec/media0
      MAJOR: '251'
      MINOR: '0'
      SUBSYSTEM: media
  - name: video10
    sysfs: >-
      /sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/video4linux/video10
    dev_path: /dev/video10
    subsystem: video4linux
    by_id: null
    attributes:
      DEVLINKS: /dev/v4l/by-path/platform-bcm2835-codec-video-index0
      DEVNAME: /dev/video10
      DEVPATH: /devices/platform/soc/3f00b840.mailbox/bcm2835-codec/video4linux/video10
      ID_PATH: platform-bcm2835-codec
      ID_PATH_TAG: platform-bcm2835-codec
      ID_V4L_CAPABILITIES: ':'
      ID_V4L_PRODUCT: bcm2835-codec-decode
      ID_V4L_VERSION: '2'
      MAJOR: '81'
      MINOR: '1'
      SUBSYSTEM: video4linux
      USEC_INITIALIZED: '174651926'
  - name: video11
    sysfs: >-
      /sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/video4linux/video11
    dev_path: /dev/video11
    subsystem: video4linux
    by_id: null
    attributes:
      DEVLINKS: /dev/v4l/by-path/platform-bcm2835-codec-video-index0
      DEVNAME: /dev/video11
      DEVPATH: /devices/platform/soc/3f00b840.mailbox/bcm2835-codec/video4linux/video11
      ID_PATH: platform-bcm2835-codec
      ID_PATH_TAG: platform-bcm2835-codec
      ID_V4L_CAPABILITIES: ':'
      ID_V4L_PRODUCT: bcm2835-codec-encode
      ID_V4L_VERSION: '2'
      MAJOR: '81'
      MINOR: '4'
      SUBSYSTEM: video4linux
      USEC_INITIALIZED: '174675850'
  - name: video12
[... 969 lines cut ...]
      MAJOR: '116'
      MINOR: '33'
      SUBSYSTEM: sound
  - name: vc-mem
    sysfs: /sys/devices/virtual/vc-mem/vc-mem
    dev_path: /dev/vc-mem
    subsystem: vc-mem
    by_id: null
    attributes:
      DEVNAME: /dev/vc-mem
      DEVPATH: /devices/virtual/vc-mem/vc-mem
      MAJOR: '247'
      MINOR: '0'
      SUBSYSTEM: vc-mem
  - name: vcsm
    sysfs: /sys/devices/virtual/vc-sm/vcsm
    dev_path: /dev/vcsm
    subsystem: vc-sm
    by_id: null
    attributes:
      DEVNAME: /dev/vcsm
      DEVPATH: /devices/virtual/vc-sm/vcsm
      MAJOR: '241'
      MINOR: '0'
      SUBSYSTEM: vc-sm

karloygard, 2021-02-09:
Checking. What was the last xcomfort version you were running?

geeewizzz, 2021-02-09:
26

But i restored an old snapshot from a few days ago where it downgraded HA and the xcomfort plugin, got the same issue.

karloygard, 2021-02-09:
I haven't changed anything related to connecting to the stick in a while, so I suspect that something else has changed in the supervisor/hassio/etc. Upgrading now to see if I see the same issues are you.

karloygard, 2021-02-09:
Confirm that I now see the same issue as you.

karloygard, 2021-02-09:
Should now be resolved with 0.28.1. A change in the supervisor required changes to be make in the addon config. Supervisor is apparently not part of the snapshot, so restoring couldn't fix the issue.

Apologies for the inconvenience!

geeewizzz, 2021-02-09:
Very nice, its working again. Thanks a lot.

closed 2021-02-09 by karloygard