karloygard/xcomfortd-go #18: panic: runtime error: slice bounds out of range [:53]

Jan-Arild-Blekken, 2021-03-29:
Hi. Trying to set this up useing ckoz 00/14. 

I'm getting this error in the log file. Any suggestion what this could be?

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[21:14:57] INFO: Starting xcomfortd version 0.34
[21:14:57] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --eprom --file /config/Dataspunkt_homeassistant.dpl --hadiscovery
2021/03/29 21:14:57 Opened USB device 'vid=188a,pid=1101,bus=1,addr=6', packet size 32/32
2021/03/29 21:14:57 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
2021/03/29 21:14:57 CI HW/RF/FW revision: 0, 13.0, 0
2021/03/29 21:14:57 CI RF/Firmware release: 2.11, 2.05
[WARN]  [store]    memorystore wiped
2021/03/29 21:14:57 Connected to broker
2021/03/29 21:14:57 CI serial number: 7666233
panic: runtime error: slice bounds out of range [:53] with capacity 0
goroutine 13 [running]:
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).dplReader(0xc000089cc0, 0x89bf20, 0xc00011e400, 0x8216f8, 0x5, 0xc00011e301, 0xbf0000c00007b950)
	/xcomfortd-go/pkg/xc/readers.go:172 +0x17c9
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).RequestDPL(0xc000089cc0, 0x89f058, 0xc000054a40, 0x0, 0x0)
	/xcomfortd-go/pkg/xc/extended.go:63 +0x113
main.run.func1(0xc000089cc0, 0xc000054a00, 0x89f058, 0xc000054a40)
	/xcomfortd-go/main.go:205 +0x770
created by main.run
	/xcomfortd-go/main.go:174 +0x428
[cmd] /run.sh exited 2
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

karloygard, 2021-03-29:
Interesting, will investigate.

Jan-Arild-Blekken, 2021-03-29:
It could be caused by the RF usb stick 00/14 did not have the dp list written to it in MRF. Changed the settings to eprom: false then i got this log:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[00:02:34] INFO: Starting xcomfortd version 0.34
[00:02:34] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --file /config/Dataspunkt_homeassistant.txt --hadiscovery
2021/03/30 00:02:34 Opened USB device 'vid=188a,pid=1101,bus=1,addr=6', packet size 32/32
2021/03/30 00:02:34 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
2021/03/30 00:02:34 CI HW/RF/FW revision: 0, 13.0, 0
[WARN]  [store]    memorystore wiped
2021/03/30 00:02:34 Connected to broker
2021/03/30 00:02:35 CI RF/Firmware release: 2.11, 2.05
2021/03/30 00:02:35 CI serial number: 7666233
2021/03/30 00:02:35 Sent MQTT autodiscover add for 3 devices and 1 datapoints

karloygard, 2021-03-29:
Yeah, there's something strange with the DPL file. Can you run again with verbose turned on?

Jan-Arild-Blekken, 2021-03-29:
eprom: true
verbose: true

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[00:18:50] INFO: Starting xcomfortd version 0.34
[00:18:50] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --file /config/Dataspunkt_homeassistant.txt --hadiscovery --verbose
2021/03/30 00:18:50 Starting xcomfortd, version 0.34
2021/03/30 00:18:50 Opened USB device 'vid=188a,pid=1101,bus=1,addr=6', packet size 32/32
2021/03/30 00:18:50 Datapoint 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 7538453, channel 0, 'Kjellerstue midten (Dimmeaktuator)'
2021/03/30 00:18:50 Datapoint 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 7541213, channel 0, 'Kjellerstue foran (Dimmeaktuator)'
2021/03/30 00:18:50 Datapoint 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 7541181, channel 0, 'Kjellerstue bak (Dimmeaktuator)'
2021/03/30 00:18:50 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
2021/03/30 00:18:50 CONFIG: [b21b10]
2021/03/30 00:18:51 STATUS: [c31b1000820000]
2021/03/30 00:18:51 CI HW/RF/FW revision: 0, 13.0, 0
2021/03/30 00:18:51 CONFIG: [b21b00]
2021/03/30 00:18:51 STATUS: [c31b00020b0205]
2021/03/30 00:18:51 CI RF/Firmware release: 2.11, 2.05
2021/03/30 00:18:51 CONFIG: [b20e00]
[WARN]  [store]    memorystore wiped
2021/03/30 00:18:51 Connected to broker
2021/03/30 00:18:51 STATUS: [c30e000074fa39]
2021/03/30 00:18:51 CI serial number: 7666233
2021/03/30 00:18:51 CONFIG: [b20401]
2021/03/30 00:18:52 STATUS: [c31c0500000000]
2021/03/30 00:18:52 CONFIG: [b21e01]
2021/03/30 00:18:52 STATUS: [c31c0500000000]
2021/03/30 00:18:52 Sent MQTT autodiscover add for 3 devices and 1 datapoints

karloygard, 2021-03-29:
Heh, I meant also with eprom enabled. :)

Jan-Arild-Blekken, 2021-03-29:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[00:22:02] INFO: Starting xcomfortd version 0.34
[00:22:02] INFO: xcomfortd --client-id xcomfort --server tcp://*****@core-mosquitto:1883 --hadiscoveryprefix homeassistant --eprom --file /config/Dataspunkt_homeassistant.txt --hadiscovery --verbose
2021/03/30 00:22:02 Starting xcomfortd, version 0.34
2021/03/30 00:22:02 Opened USB device 'vid=188a,pid=1101,bus=1,addr=6', packet size 32/32
2021/03/30 00:22:02 Datapoint 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 7538453, channel 0, 'Kjellerstue midten (Dimmeaktuator)'
2021/03/30 00:22:02 Datapoint 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 7541213, channel 0, 'Kjellerstue foran (Dimmeaktuator)'
2021/03/30 00:22:02 Datapoint 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 7541181, channel 0, 'Kjellerstue bak (Dimmeaktuator)'
2021/03/30 00:22:02 Connecting to MQTT broker 'tcp://core-mosquitto:1883' with id 'xcomfort'
2021/03/30 00:22:02 CONFIG: [b21b10]
[WARN]  [store]    memorystore wiped
2021/03/30 00:22:02 Connected to broker
2021/03/30 00:22:02 STATUS: [c31b1000820000]
2021/03/30 00:22:02 CI HW/RF/FW revision: 0, 13.0, 0
2021/03/30 00:22:02 CONFIG: [b21b00]
2021/03/30 00:22:02 STATUS: [c31b00020b0205]
2021/03/30 00:22:02 CI RF/Firmware release: 2.11, 2.05
2021/03/30 00:22:02 CONFIG: [b20e00]
2021/03/30 00:22:03 STATUS: [c30e000074fa39]
2021/03/30 00:22:03 CI serial number: 7666233
2021/03/30 00:22:03 CONFIG: [b20401]
2021/03/30 00:22:03 STATUS: [c31c0500000000]
2021/03/30 00:22:03 CONFIG: [b21e01]
2021/03/30 00:22:03 STATUS: [c31c0500000000]
2021/03/30 00:22:03 Reading datapoints list from eprom
2021/03/30 00:22:03 EPROM: [d1210000018008000200000000000000]
2021/03/30 00:22:04 EPROM: [d111080001800a00004b0078c00401800100]
2021/03/30 00:22:04 EPROM: [d111100001800a000100cd2b71004d000000]
2021/03/30 00:22:04 EPROM: [d1111a0001800a0000000000000000000000]
2021/03/30 00:22:04 EPROM: [d111240001800a0000000000000000000000]
2021/03/30 00:22:05 EPROM: [d1112e0001800a0000000000000000000000]
2021/03/30 00:22:05 EPROM: [d111380001800a0000000000000000000000]
2021/03/30 00:22:05 EPROM: [d111420001800a0000000000000000000000]
2021/03/30 00:22:05 EPROM: [d1114c0001800a0000000000000000000000]
2021/03/30 00:22:05 EPROM: [d111560001800a0000000000000000000000]
2021/03/30 00:22:06 EPROM: [d111600001800a0000000000000000000000]
2021/03/30 00:22:06 EPROM: [d1116a0001800a0000000000000000000000]
2021/03/30 00:22:06 EPROM: [d111740001800a0000000000000000000000]
2021/03/30 00:22:06 EPROM: [d1117e0001800a0000000000000000000000]
2021/03/30 00:22:07 EPROM: [d111880001800a0000000000000000000000]
2021/03/30 00:22:07 EPROM: [d111920001800a0000000000000000000000]
2021/03/30 00:22:07 EPROM: [d1119c0001800a0000000000000000000000]
2021/03/30 00:22:07 EPROM: [d111a60001800a0000000000000000000000]
2021/03/30 00:22:08 EPROM: [d111b00001800a0000000000000000000000]
2021/03/30 00:22:08 EPROM: [d111ba0001800a0000000000000000000000]
2021/03/30 00:22:08 EPROM: [d111c40001800a0000000000000000000000]
2021/03/30 00:22:08 EPROM: [d111ce0001800a0000000000000000000000]
2021/03/30 00:22:08 EPROM: [d111d80001800a0000000000000000000000]
2021/03/30 00:22:09 EPROM: [d111e20001800a0000000000000000000000]
2021/03/30 00:22:09 EPROM: [d111ec0001800a0000000000000000000000]
2021/03/30 00:22:09 EPROM: [d111f60001800a0000000000000000000000]
2021/03/30 00:22:09 EPROM: [d111000101800a0000000000000000000000]
2021/03/30 00:22:10 EPROM: [d1110a0101800a0000000000000000000000]
2021/03/30 00:22:10 EPROM: [d111140101800a0000000000000000000000]
2021/03/30 00:22:10 EPROM: [d1111e0101800a0000000000000000000000]
2021/03/30 00:22:10 EPROM: [d111280101800a0000000000000000000000]
2021/03/30 00:22:11 EPROM: [d111320101800a0000000000000000000000]
2021/03/30 00:22:11 EPROM: [d1113c0101800a0000000000000000000000]
2021/03/30 00:22:11 EPROM: [d111460101800a0000000000000000000000]
2021/03/30 00:22:11 EPROM: [d111500101800a0000000000000000000000]
2021/03/30 00:22:12 EPROM: [d1115a0101800a0000000000000000000000]
2021/03/30 00:22:12 EPROM: [d111640101800a0000000000000000000000]
2021/03/30 00:22:12 EPROM: [d1116e0101800a0000000000000000000000]
[... 86 lines cut ...]
2021/03/30 00:22:32 EPROM: [d111d40401800a0000000000000000000000]
2021/03/30 00:22:32 EPROM: [d111de0401800a0000000000000000000000]
2021/03/30 00:22:33 EPROM: [d111e80401800a0000000000000000000000]
2021/03/30 00:22:33 EPROM: [d111f20401800a0000000000000000000000]
2021/03/30 00:22:33 EPROM: [d111fc0401800a0000000000000000000000]
2021/03/30 00:22:33 EPROM: [d111060501800a0000000000000000000000]
2021/03/30 00:22:34 EPROM: [d111100501800a0000000000000000000000]
2021/03/30 00:22:34 EPROM: [d1111a0501800a0000000000000000000000]
2021/03/30 00:22:34 EPROM: [d111240501800a0000000000000000000000]
2021/03/30 00:22:34 EPROM: [d1112e0501800a0000000000000000000000]
panic: runtime error: slice bounds out of range [:53] with capacity 0
goroutine 11 [running]:
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).dplReader(0xc000089cc0, 0x89bf20, 0xc0001402e0, 0x8216f8, 0x5, 0xc000140201, 0xbf0000c00007b880)
	/xcomfortd-go/pkg/xc/readers.go:172 +0x17c9
github.com/karloygard/xcomfortd-go/pkg/xc.(*Interface).RequestDPL(0xc000089cc0, 0x89f058, 0xc000054a80, 0x0, 0x0)
	/xcomfortd-go/pkg/xc/extended.go:63 +0x113
main.run.func1(0xc000089cc0, 0xc000054a40, 0x89f058, 0xc000054a80)
	/xcomfortd-go/main.go:205 +0x770
created by main.run
	/xcomfortd-go/main.go:174 +0x428
[cmd] /run.sh exited 2
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

karloygard, 2021-03-29:
I agree that that does look somewhat empty. I'll make a fix for that, but in the meanwhile, if you upload the DPL to the stick with MRF, this should resolve itself.

Jan-Arild-Blekken, 2021-03-29:
When I try to upload the dpl to the stick it only gives me an error message about the upload process failed.

karloygard, 2021-03-29:
Strange. You can try saving the DPL to disk, and it'll be the same as reading from eprom. The DPL file format contains more info than then TXT files, so is preferable (but TXT will still work).

Jan-Arild-Blekken, 2021-04-02:
The problem with saving the DPL list to stick was that I had both the 00/13 and the 00/14 in the same pc at once when programming. Moving the 00/14 to another pc during MRF programming solved the issue.

karloygard, 2021-04-02:
You can leave the 00/14 connected to Home Assistant while uploading the DPL, it will autodetect the change and refresh HA.

Great to hear that you figured this out. I'll push a change that prevents the crash in any case.

closed 2021-09-18 by Jan-Arild-Blekken