karloygard/xcomfortd-go #19: Feature request - Multi Channel Heating Actuator (CHAZ-01/12)

Jan-Arild-Blekken, 2021-04-02:
Hi. 

Is the multi channel heating actuator CHAZ-01/12 supported? If not is it possible to add support for it?

I run this system via sensio today. 

Regards Jan Arild

karloygard, 2021-04-02:
100% possible. I don't have such a device, so it may take a bit longer to get it implemented, though.

Jan-Arild-Blekken, 2021-04-02:
I could provide you logs and do testing if that helps you. :-)

Jan-Arild-Blekken, 2021-04-04:
This line is from the log:

2021/04/04 13:05:37 Unknown channel 13 for device Multi Channel Heating Actuator (CHAZ-01/12)

Does this mean the system is reading the 12 channels og gives an error message when it does not fint channel 13? Guess each channel has to be read and listed as a switch. Then it can be controlled by adding a generic thermostat in HA.

karloygard, 2021-04-04:
The heating actuators are a substantially more complicated than the regular actuators, with a number of different operation modes. Can you figure out it there is a mode where the actuator will receive just on/off messages? It's not immediately obvious to me from the documentation, and I don't have a devices to play with.

Jan-Arild-Blekken, 2021-04-04:
Not sure but in my currenct config I did'nt have to change anything from default. I'm guessing when sending the desired temperature and actual temperature the actuator either turns on or off the channel. (true or false)

[image]
[image]
[image]
[image]
[image]

A solution from Heatit:

https://www.facebook.com/groups/680252689011262/permalink/1365731213796736

Jan-Arild-Blekken, 2021-04-10:
Hi. Not sure if you have had the time to look at this yet. Do you need me to try to dig up some more information?

karloygard, 2021-04-11:
I'm still thinking about it. It's not an insubstantial effort.

karloygard, 2021-04-21:
I am struggling to find an elegant way of integrating the actuator with Home Assistant. The actuator requires both the desired and current temperatures, which is different from how HA expects to control HVAC; HA just wants to switch the actuator on and off. The xComfort way makes sense, but I can't find a simple way of shoe horning it into HA without introducing lots of complicated automations.

Jan-Arild-Blekken, 2021-04-22:
That makes sense. How complicated automations are we talking about? The Actuator as I understand it just needs input for temperature and desired temperature sent in intervals. I guess no need for an database but you have to define the attribute for each channel and then map them to the desired RCT.

Ost99, 2021-06-19:
If you have Xcomfort room controllers (RCT) and the Xcomfort actuators it will probably be best to let the RCTs handle the actual control of the actuators (that's what is normally done with Sensio or SHC). If you want to control the temperature from HA, you find a way to read and write the desired value in the RCT ().

The only instance you would need to control the actuators from HA is if you want it use a non-xcomfort thermostat.

Jan-Arild-Blekken, 2021-06-19:
I do have RCT in each room.

karloygard, 2021-06-21:
It may be possible to solve it this way, except that the desired temperature is sent to the heating actuator. The RCT is sleeping most of the time, and does not listen for messages.

I don't have SHC/Sensio or any heating actuators, so I not experience with them, but is this how SHC/Sensio does it?

Jan-Arild-Blekken, 2021-06-21:
This is from the MRF 2.64 manual. Hope that helps.

[image]

karloygard, 2021-06-21:
That appears to be a specific mode when connecting to SHC. Is it possible to set this mode when connecting to the USB communication interface as well?

Jan-Arild-Blekken, 2021-06-21:
You are correct. Deleted the old connection and drew a new connection from RCT to the USB stick related to Temperature. 

[image]

[image]

karloygard, 2021-06-21:
Do we know how Sensio handles this?

Jan-Arild-Blekken, 2021-06-21:
I have a sensio controller and the mode is set to ekstern controller. Change the temperature via app. The connections is drawn from RCT to controller and from heating actuator to controller.

Jan-Arild-Blekken, 2021-06-21:
From the manual: Temperaturinformasjonen for hver av de 12 inngangene kan enten sendes av en Romtermostat
(Type CRCA-00/06 eller CRCA-00/07), en Room-Manager, en Home-Manager eller en annen
intelligent komponent som Ethernet CI, USB-RF-kommunikasjonsstick etc.

karloygard, 2021-06-21:
Judging by your observations, MRF has hard coded functionality for Sensio and SHC, which is unavailable to us.

I can take a shot at this, but without the extra functionality, the implementation will be relatively basic and will only talk to the heating actuator.

Jan-Arild-Blekken, 2021-06-22:
Yes it seem so if anybody else has more information regarding the RCT.
Guess Therese will be problems controlling RCT with P1000 also from HA.

karloygard, 2021-06-24:
I thought the PT1000 should just work, what's the issue that you are seeing?

Jan-Arild-Blekken, 2021-06-27:
I guess as HA is not able to communicate directly to the RCT like the SHC does via the closed channel we can't control the temperature via HA. Correct?

oyvindth, 2021-07-20:
As far as I understand, in the Normal mode the RCT is just a sensor, sending temperature, moisture, and battery data to the USB stick. The display on the RCT is showing sensor data from the internal sensors in the RCT.

If you have a SHC or a Sensio controller connected to the RCT, the RCT is still sending sensordata to the controller, but the display on the RCT is showing data sent from the controller. Hence you can have the RCT displaying outdoor temp, you don't have to set time and date in MRF, it is done by the controler, and a few other features.

HA could very well use the RCT to control the temperature, but the RCT will just operate as a temperature/moisture sensor. The desired temperature will just be available through the HA interface. 

The problem is with the heating actuators. they require as I understand more info than regular switching commands to work as intended. They require both actual temp, and desired temp to be sent to the actuator itself. I have tried to connect a plugged heating actuator, and a normal single channel actuator to the USB stick. They both show up as lights in HA, and when trying to switch them, nothing happens. Most likely because the actuators have not received the necessary info. the Xcomfort log shows the mqtt command being sent, but no action on the actuator side is happening. If I however use MFR to switch the actuators on, the interface in HA change to on. 
MRF is somehow managing to switch the heating actuators as normal switches. Maybe simulating MRF is a possible walkaround.
I do not know it the switching done by MRF require the Konfiguration stick, and that the komunication stik is unable to switch the actuator manualy.  but if you could find a way to use the actuator as a normal switching actuator, that would be a step on the way atleast.

Jan-Arild-Blekken, 2021-09-16:
Fixed. See #25.

closed 2021-09-16 by Jan-Arild-Blekken