karloygard/xcomfortd-go #20: Feature request - PT1000 sensor CSEZ-01/36

Jan-Arild-Blekken, 2021-04-04:
Hi

Have looked a bit more in the code and could not find the sensor PT1000 listed. 

Support for this device would be highly appreciated. Because the RCT and light switches is mountet on the wall outside of the bathroom I use this sensor for temperature measurement of the concrete floor. The sensor is connected to the RCT with two wires. Is it possible to add support for this device also? 

Regards Jan Arild

karloygard, 2021-04-04:
Can you send me the DPL file with this device configured? I'm not sure PT1000 mode is specified in the DPL file, so a different scheme would have to be used to detect it.

Jan-Arild-Blekken, 2021-04-04:
Yes. Have attached the DPL file. Included both the USB stick 00/14 and the sensio controller.

Attached to the termostat:

Termostat Bad
Varmekabel bad 1.etg

[Homeassistant.zip](https://github.com/karloygard/xcomfortd-go/files/6254803/Homeassistant.zip)

[image]
[image]

karloygard, 2021-04-06:
The PT1000 sensor may be supported already, but I can't really tell. Can you send me the log from the add-on so I can check which message the RCT is sending?

oyvindth, 2021-08-29:
I have as a test added a RTC with a PT1000 to Home assistant. 
What I see is that when the PT1000 in MRF is set as "Internal floor sensor", is does not show up in Home assistant, in Sensio and SHC, is shows, but it looks like the data is sent on the same DP as temperature, humidity, floor sensor and light intensity. When adding the RTC to Sensio or SHC in MFT only the Info/Control channel is available to add unless you sett the PT1000 as outdoor then it wil have a sepparate DP.

If setting the PT1000 to any of the two choises, (normal or outdoor), it shows up when trying to add the RCT to Home Assistant, and can be added as a separate DP to homeassistant. It will show up as a datapoint in HA.

karloygard, 2021-08-30:
I am going to need some logs in order to see what gets sent.

oyvindth, 2021-08-30:
If adding the PT1000 as internal sensor, this is what it sends from the Xcomfort Logs. This is the same as MRF sees. No mention of the PT1000.

`2021/08/30 23:58:04 wheel position: -5.0
2021/08/30 23:58:04 Device 6923557 (channel 0-'Termostat Kjeller (Temperatur)') sent message (battery good, signal good, cyclic true) event 'value' with value 26.7
2021/08/30 23:58:05 Device 6923557 (channel 1-'Termostat Kjeller (Fuktighet)') sent message (battery good, signal good, cyclic true) event 'value' with value 46.1`

Is adding the PT1000 as normal temperature sennsor it gets it's own DP. This is picked up by home assistant, and works fine. Also shows up in Lovelace.

This is what the logs show.

`2021/08/31 00:06:06 Device 6923557 (channel 6-'Termostat Kjeller (ekstern Sensor (PT1000))') sent message (battery good, signal good, cyclic true) event 'value' with value 29.3`

oyvindth, 2021-08-30:
Here is what the RTC sends to Sensio, is seems only SHC or Sensio X-1 gets the extended info messages.
`31.8.2021	00:05:54	+9 Sek.	Termostat Garderobe (Info/Styring) sender <RCT Out> til Sensio X1-Controller (DP:46) Temp: 25.2°C,   Fuktighet: 49.5% rH,  Ratt: 0.0°, <Natt> Gulv: 23.8°C, VOC: 655.3%, 0 Lux. Tidsur inaktivt, Oppvarming, Sensor ( Termostat Garderobe => Downlights Hall 1.Etg )     (Batteri: nesten fullt)	28 %	`

karloygard, 2021-08-31:
Apparently, the `RCT_Out` (and `RCT_Req`) messages are apparently only sent to SHC and Sensio X-1. MRF simply won't let you configure RCT to send that to the CI devices. It would have been trivial to support it, but I presently don't see a way of tricking MRF.

oyvindth, 2021-09-08:
The PT 1000 shows up in HA if you select it as normal temperature sensor. Then you may use it in an automation to limit floor temperature. I have checked what Sensio does when floor temp is out of limits, and it does not send the floor temp to the heating actuator, Sensio just changes the temp it reports to the actuator. So if floor temp is to high,  sensio sends current_temp 20 set temperature 10 is to the actuator regardless of what is set. Should be possible to do in HA aswell.

karloygard, 2021-09-09:
> So if floor temp is to high, sensio sends current_temp 20 set temperature 10 is to the actuator regardless of what is set.

This didn't make sense to me, can you reformulate?

oyvindth, 2021-09-09:
When Sensio senses a floor temperature that is higher than the programmed max floor temperature, the controller disregards the actual temperature in the room/current_temp and reports a actual temp of 20 degrees to the actuator. At the same time it stops sending the actual required temperature, and reports the required temperature as 10 degrees. So actual temp 20 req temp 10 the actuator switches off. When floor temp is within limits again, the correct temperatures are transmitted to the heating actuator.

karloygard, 2021-09-11:
Do you understand why it has this extra logic? It eludes me.

oyvindth, 2021-09-11:
The purpose of the floor sensor, is to avoid the floor becoming too hot. Wooden flooring in particular handles heat poorly. Because of this, when the floor sensor hits the programmed max temperature, it has to turn the heat of, regardless of the room temperature. With a switch controlling the heat, it would be a simple matter of turning the switch off, but since the heating actuators do their own switching based on received temperatures, the only way of making sure the heating actuators turn off, and cool the floors, is to feed it with a set of temperatures that ensures the heating actuator switches off. Sensio is sending actual 20 req 10 to force the heating actuator to switch.

closed 2022-05-06 by Jan-Arild-Blekken