karloygard/xcomfortd-go #21: Responsiveness / timeout issues

havardthom, 2021-05-13:
Hi,
I have been using this xComfort gateway for the past 6 months and it's been working great, so thanks for that!

I recently purchased a Philips Hue Dimmer switch to remote control my lights. 
In home assistant I've set up a light group with 3 xComfort Actuators (from this gateway) and 2 Ikea light drivers (from zigbee2mqtt). 

[image]

I've also made a Node-Red flow to control the lights with the Hue Dimmer Switch.

[image]

The flow uses the Home Assistant Light integration (https://www.home-assistant.io/integrations/light) to toggle lights or change brightness by step values (e.g. increase/decrease 20 percent brightness). 

The Hue Dimmer switch works fine if I'm very patient and wait 1+ seconds between each button press. However if I start to push the buttons a bit faster (to decrease brightness for example), the responsiveness of the xComfort lights starts to deteriorate and eventually they timeout / stop responding. I have to restart the xComfort gateway to get the lights working again. This issue might be related to https://github.com/karloygard/hassio-addons/issues/12 ? Let me know if I should move my issue there.

Here are some logs from when the lights stops responding. It also seems like the Hassio Light service publishes commands twice which is quite strange.
```
2021/05/13 13:17:58 STATUS: [c31c04d0100000]
2021/05/13 13:17:58 TX (seq e, 1 parallel): [b1020d4052e0] 
2021/05/13 13:17:59 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:00 RX: [c100733900906575004d0002d2002200000049100d]
2021/05/13 13:18:00 Device 7693712, type CDAU 01/04 sent extended status message: status ON, value 210, temp 34C (battery mains-powered, signal normal)
2021/05/13 13:18:00 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:01 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:10 STATUS: [c30a0100000000]
2021/05/13 13:18:10 Timeaccount zero, no more transmission possible
2021/05/13 13:18:10 STATUS: [c31c04e0100000]
2021/05/13 13:18:10 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:10 TX (seq f, 1 parallel): [b1030d4052f0] 
2021/05/13 13:18:11 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:12 RX: [c100733900cc8b77004d0002d20020000000391002]
2021/05/13 13:18:12 Device 7834572, type CDAU 01/04 sent extended status message: status ON, value 210, temp 32C (battery mains-powered, signal good)
2021/05/13 13:18:22 STATUS: [c30a0100000000]
2021/05/13 13:18:22 Timeaccount zero, no more transmission possible
2021/05/13 13:18:22 STATUS: [c31c04f0100000]
2021/05/13 13:18:22 TX (seq 0, 1 parallel): [b1010d405200] 
2021/05/13 13:18:24 RX: [c100733900b92577004d0002d200240000003a100d]
2021/05/13 13:18:24 Device 7808441, type CDAU 01/04 sent extended status message: status ON, value 210, temp 36C (battery mains-powered, signal good)
2021/05/13 13:18:24 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:31 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:34 STATUS: [c30a0100000000]
2021/05/13 13:18:34 Timeaccount zero, no more transmission possible
2021/05/13 13:18:35 STATUS: [c31c0400100000]
2021/05/13 13:18:35 TX (seq 1, 1 parallel): [b1020d405210] 
2021/05/13 13:18:46 STATUS: [c30a0100000000]
2021/05/13 13:18:46 Timeaccount zero, no more transmission possible
2021/05/13 13:18:47 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:47 STATUS: [c31c0410100000]
2021/05/13 13:18:47 TX (seq 2, 1 parallel): [b1030d405220] 
2021/05/13 13:18:52 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:55 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:58 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:18:59 STATUS: [c30a0100000000]
2021/05/13 13:18:59 Timeaccount zero, no more transmission possible
2021/05/13 13:18:59 STATUS: [c31c0420100000]
2021/05/13 13:18:59 TX (seq 3, 1 parallel): [b1010d405930] 
2021/05/13 13:19:01 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:06 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:07 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:11 STATUS: [c30a0100000000]
2021/05/13 13:19:11 Timeaccount zero, no more transmission possible
2021/05/13 13:19:11 STATUS: [c31c0430100000]
2021/05/13 13:19:11 TX (seq 4, 1 parallel): [b1020d405940] 
2021/05/13 13:19:13 RX: [c100733900906575004d0002e300220000004a100e]
2021/05/13 13:19:13 Device 7693712, type CDAU 01/04 sent extended status message: status ON, value 227, temp 34C (battery mains-powered, signal normal)
2021/05/13 13:19:14 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:16 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:17 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:21 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:22 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:23 STATUS: [c30a0100000000]
2021/05/13 13:19:23 Timeaccount zero, no more transmission possible
2021/05/13 13:19:23 STATUS: [c31c0440100000]
2021/05/13 13:19:23 TX (seq 5, 1 parallel): [b1030d405950] 
2021/05/13 13:19:25 RX: [c100733900cc8b77004d0002e30020000000381003]
2021/05/13 13:19:25 Device 7834572, type CDAU 01/04 sent extended status message: status ON, value 227, temp 32C (battery mains-powered, signal good)
2021/05/13 13:19:25 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:35 STATUS: [c30a0100000000]
2021/05/13 13:19:35 Timeaccount zero, no more transmission possible
2021/05/13 13:19:35 STATUS: [c31c0450100000]
2021/05/13 13:19:35 TX (seq 6, 1 parallel): [b1010d405960] 
2021/05/13 13:19:37 RX: [c100733900b92577004d0002e3002400000039100e]
2021/05/13 13:19:37 Device 7808441, type CDAU 01/04 sent extended status message: status ON, value 227, temp 36C (battery mains-powered, signal good)
2021/05/13 13:19:37 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:38 MQTT message; topic: 'xcomfort/1/set/dimmer', message: '56'
2021/05/13 13:19:38 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:39 MQTT message; topic: 'xcomfort/2/set/dimmer', message: '56'
2021/05/13 13:19:39 MQTT message; topic: 'xcomfort/3/set/dimmer', message: '56'
2021/05/13 13:19:40 MQTT message; topic: 'xcomfort/1/set/dimmer', message: '57'
2021/05/13 13:19:41 MQTT message; topic: 'xcomfort/2/set/dimmer', message: '57'
2021/05/13 13:19:41 MQTT message; topic: 'xcomfort/3/set/dimmer', message: '57'
2021/05/13 13:19:41 MQTT message; topic: 'xcomfort/1/set/dimmer', message: '56'
2021/05/13 13:19:41 MQTT message; topic: 'xcomfort/2/set/dimmer', message: '56'
2021/05/13 13:19:42 MQTT message; topic: 'xcomfort/3/set/dimmer', message: '56'
2021/05/13 13:19:42 MQTT message; topic: 'xcomfort/1/set/dimmer', message: '56'
2021/05/13 13:19:42 MQTT message; topic: 'xcomfort/2/set/dimmer', message: '56'
2021/05/13 13:19:42 MQTT message; topic: 'xcomfort/3/set/dimmer', message: '56'
2021/05/13 13:19:43 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:47 STATUS: [c30a0100000000]
2021/05/13 13:19:47 Timeaccount zero, no more transmission possible
2021/05/13 13:19:47 STATUS: [c31c0460100000]
2021/05/13 13:19:47 TX (seq 7, 1 parallel): [b1020d405970] 
2021/05/13 13:19:48 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:51 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:52 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:56 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:57 handle_events: error: libusb: interrupted [code -10]
2021/05/13 13:19:59 MQTT message; topic: 'xcomfort/1/set/switch', message: 'false'
2021/05/13 13:19:59 MQTT message; topic: 'xcomfort/1/set/switch', message: 'false'
2021/05/13 13:19:59 MQTT message; topic: 'xcomfort/2/set/switch', message: 'false'
2021/05/13 13:19:59 MQTT message; topic: 'xcomfort/3/set/switch', message: 'false'
2021/05/13 13:19:59 MQTT message; topic: 'xcomfort/2/set/switch', message: 'false'
2021/05/13 13:19:59 MQTT message; topic: 'xcomfort/3/set/switch', message: 'false'
2021/05/13 13:19:59 STATUS: [c30a0100000000]
2021/05/13 13:19:59 Timeaccount zero, no more transmission possible
2021/05/13 13:19:59 STATUS: [c31c0470100000]
2021/05/13 13:19:59 TX (seq 8, 1 parallel): [b1030d405980] 
```

karloygard, 2021-05-15:
There's more than one issue at work here; `Timeaccount zero, no more transmission possible` indicates that the USB stick has used up its transmission quota and needs to wait to send more messages. I assume that's because you have been testing extensively?

Can you test again now and attach logs from that run? Sending lots of messages to a single device in a short amount of time is likely to bog down the xComfort code, it doesn't discard any messages and will patiently wait for the dimmer to go through *all* the commanded steps, that's possible to tweak.

karloygard, 2021-05-19:
I've issued a fix i 0.35 which will rate limit updates to a device; if multiple commands are issues in quick succession, only the last will be applied. Give it a shot and see if it fixes your problem.

closed 2021-05-19 by karloygard

havardthom, 2021-05-20:
Yes it feels more responsive now and I'm not getting `Timeaccount zero` errors anymore. Great work!