karloygard/xcomfortd-go #22: Feature request - Move/step light brightness

havardthom, 2021-05-13:
Hi,
It would be great to have the option to send move/step brightness commands to xComfort lights, similar to https://www.zigbee2mqtt.io/devices/ICPSHC24-30EU-IL-1.html#movingstepping

```
move: this will automatically move the brightness value over time, to stop send value stop or 0.
step: this will increment/decrement the current brightness value by the given one.
```

In my case this would be very useful for decreasing/increasing light brightness continuously while holding down buttons on a Hue Dimmer Switch.

karloygard, 2021-05-13:
New dimmers support this, so I may be able to add that, as long as it's supported by MQTT discovery.

karloygard, 2021-05-16:
This approach doesn't look like it's part of MQTT discovery? Having some trouble seeing how this functionality would be used.

havardthom, 2021-05-20:
Unfortunately I'm not very familiar with MQTT discovery. I had a look at the docs and i could not find info to help achieve this functionality, except maybe the `transition` value? e.g. going from 100% brightness to 1% brightness with a 5 second transition, I'm not sure if that's how it works though.

As mentioned, I would find this functionality useful for my Hue Dimmer Switch where there are buttons to increase/decrease brightness. When i hold the decrease brightness button i want the light brightness to decrease continuously, exactly how the xComfort dimmer switches work. With the current `xcomfort/+/set/dimmer` topic i imagine i would have to send alot of messages to e.g. get a smooth transition from 100% - 1% brightness over 5 seconds. And if i want to decrease brightness on 3 lights simultaneously, it will be even more messages. But maybe it isn't a problem, have you tried anything similar before?