karloygard/xcomfortd-go #23: 2021/07/18 01:02:51 handle_events: error: libusb: interrupted [code -10]

oyvindth, 2021-07-17:
I am seing a lot of these errors **(2021/07/18 01:02:51 handle_events: error: libusb: interrupted [code -10])** in my xcomfort logg. What are they, and will it affect the comfort addon.

kind regards.

karloygard, 2021-07-17:
As far as I can remember, it's gousb logging a return code from libusb and then proceeding as normal. It's nothing to worry about, but it is a bit annoying.

I tried silencing the log message, but that involved forking various libraries, which I'd rather not do.

closed 2021-07-17 by karloygard

FIS-X, 2022-02-03:
Could you please have another go at removing this error? My CDO is not liking it :S

karloygard, 2022-02-03:
It this issue: https://github.com/google/gousb/issues/87. It may be useful to voice your opinion there, maybe they'll get around to fixing it.

FIS-X, 2022-04-18:
Sorry for bumping this old thread again, but could the answer given here be of any help? https://github.com/google/gousb/issues/87#issuecomment-1100956460

karloygard, 2022-04-19:
Okay, since you're so persistent, it's been fixed in 0.54. 😄

FIS-X, 2022-04-19:
> Okay, since you're so persistent, it's been fixed in 0.54. 😄

Thank you so much!! :D