karloygard/xcomfortd-go #24: Feature request - Add temp displacement as a entity for RCT

oyvindth, 2021-07-19:
I have several RCT around the house, and when connecting them to HA they have 4 entities Temp/Moist/Battery/Signal.

When looking at the logs it also reports the temp displacement sett on the RTC (The adjustment wheel on the older thermostats) as "dropping wheel position on the ground: -3.0" is it possible to have this as a separate entity, so that this can be used to change the set temperature in HA for the specific zone?

karloygard, 2021-07-19:
Adding the wheel position is trivial, it mostly depends how you want to expose it. I don't have a RCT, so I'm not sure what it reports, so for the time being, I just logged the wheel position and ignored it.

The wheel position could be exposed as a separate value (`offset`, maybe) or added to the reported temperature for simplicity. I don't know which is easier for you to deal with?

oyvindth, 2021-07-19:
Reported as a separate value (offset is good) would be great in my use case. It would give me all the flexibility I need.

Kind Regards

Øyvind Thoresen

oyvindth, 2021-07-20:
Updated to 0.36. the wheel position is no longer dropped on the floor in the xcomfort logs, but I see no entity displaying the value in the mqtt integration.

karloygard, 2021-07-21:
I'm just experimenting at the moment.

karloygard, 2021-07-21:
I just released 0.36.1, the temperature adjustment should hopefully show up now.

oyvindth, 2021-07-21:
Working great.

closed 2021-07-21 by karloygard