karloygard/xcomfortd-go #25: CHAP-01/F5-12E pluggable heating actuator not switching

oyvindth, 2021-07-19:
I have just bought a CHAP-01/F5-12E puggable heating actuator. When added to HA, it shows up as a heating actuator, and if controlled by MRF, the status updates in HA, but no ems shows. If trying to controll from HA, nothing happens with the actuator have tried bout lovelace and MQTT.publish. 
Is this because it is not Supported? or is it another snag. 
All other connected devises works fine.

karloygard, 2021-07-21:
That device is barely supported by the code. I don't have a heating actuator myself, so it's not completely trivial to fix that.

oyvindth, 2021-07-21:
By your name, I guess you are located in Norway, if it would help, I am more than willing to lend you the devise if it would help with integrating heating actuators. 
As I understand the unit, it requires actual temp, and desired temp to work. If you find a way to send those messages to the unit, maby an automation could handle the cyclic sending of the data?

karloygard, 2021-07-21:
That would likely speed up development substantially. I live in Oslo, but as most Norwegians, I'm on vacation at the moment. We can touch base in the beginning of August and take it from there?

oyvindth, 2021-07-21:
Sounds like a great plan. I drive through Oslo on a regular basis, or i could send it to you by Mail. Enjoy the rest of fellesferien.

oyvindth, 2021-08-14:
Hi again. I am back from vacation. Easiest for me would be to send you the device pr mail, but I understand if you are not to keen on posting personal data here. Any chance I could PM you on Facebook Messenger.

karloygard, 2021-08-15:
👍🏻 Find me there, full name Karl Anders Øygard.

karloygard, 2021-08-28:
> I am stil not able to sent the current_temperature to the actuator however.

I'll fiddle with it a little myself and see if I get any further.

>The number.current_temperature entity is not showing anymore after the last update, might be a problem on my end thoug. I havent tried to rebuild the xcomfort addon yet. and I am not able to figure out what MQTT topic to send the current temperature to. My plan was to create an automation that simply forwards the RCT temp with a MQTT publich command to the actuator, but I am not sure to what topic to send the temperature to.

The MQTT topic for current temperature is `xcomfort/{dp}/set/current`, for example `xcomfort/50/set/current`.

> When sending a desired temperature to the actuator, I can see the MQTT transmission, but the confirmation from the actuator is only showed in MRF. Not showing up in MQTT for some reason. Don't know if it is relevant.

Currently, no confirmation is sent via MQTT, so this is correct.

karloygard, 2021-08-28:
The following automation does the trick:
```
- alias: Copy temperature every minute                                                                                                                                                                
  trigger:                                                                                                                                                                                            
  - platform: time_pattern                                                                                                                                                                            
    minutes: "/1"                                                                                                                                                                                     
  action:                                                                                                                                                                                             
  - service: number.set_value                                                                                                                                                                         
    target:                                                                                                                                                                                           
      entity_id: number.current_temperature                                                                                                                                                           
    data_template:                                                                                                                                                                                    
      value: "{{ states('sensor.romkontroller') }}"                                                                                                                                                   
```
Runs every minute, which is maybe a little aggressive.

Jan-Arild-Blekken, 2021-08-28:
I have watched the latest updates and read the code updates. It seems like it should also work with the Multi Channel Heating Actuator (CHAZ-01/12). Is this correct?

@oyvindth do you mind sharing your automation code and the thermostat setup in lovelace? It seems that we share the same problem.

oyvindth, 2021-08-28:
This is working great. Actuator switching as expected. And indications in Lovelace shows that the heating is on. This is looking Great

@Jan-Arild-Blekken I am still during the tryout phase. Karloygard have only been working on this for 2 days, and I am currently just sending messages to the actuator using direct MQTT commands. Setting the temperature with the basic temperature/Climate card works good though.

oyvindth, 2021-08-28:
Have uppdatet the automation to be somewhat less agressive. Now it only fires when the temperature changes, or every 30 min to avoid actuator going into safety mode.
`- id: '1630180139158'
  alias: Test Current Temp
  description: Oppdaterer current Temp ved oppdatering fra rtc eller hver 30 min
  trigger:
  - platform: state
    entity_id: sensor.test_home_temperatur
  - platform: time_pattern
    minutes: '30'
  condition: []
  action:
  - service: number.set_value
    target:
      entity_id: number.varmeaktuator_varmeinngang_current
    data_template:
      value: '{{ states(''sensor.test_home_temperatur'') }}'
  mode: single`

@karloygard Seems like the energy channel is a cumulativ usage number for the actuator. Would be fantastic to be able to access the average effect portion of the status message as this gives current load on the actuator.

Jan-Arild-Blekken, 2021-08-28:
I agree that every 30 min is better. I have waterborne heating at it takes some time before the temperature changes in the room.

When everything is working as it should perhaps a blueprint would makes things easier?

oyvindth, 2021-08-28:
Any idea on how to set the current_temp in the climate entity? it only showes as null, I found a PY script that might do the job, and update the atribute, but it would be fantastic to just have it read the figure that is sent to the actuator. Do you think that is a possibility?

karloygard, 2021-08-28:
> I have watched the latest updates and read the code updates. It seems like it should also work with the Multi Channel Heating Actuator (CHAZ-01/12). Is this correct?

Yes, this is close to something that should work for CHAZ-01/12. It has some extra features that is specific to that device, but that shouldn't be critical.  Feel free to test it and report back!

karloygard, 2021-08-29:
> Any idea on how to set the current_temp in the climate entity? it only showes as null, I found a PY script that might do the job, and update the atribute, but it would be fantastic to just have it read the figure that is sent to the actuator. Do you think that is a possibility?

Might be possible. Out of curiosity, can you link the script that you found?

oyvindth, 2021-08-29:
Here is the script. Haven’t tried it, so don’t know if it works, but according to description it should

karloygard, 2021-08-29:
0.40.4 now updates the current temperature in the climate entity. This is mostly for show, the value will always be whatever `number.varmeaktuator_varmeinngang_current` (in this particular case) is set to.

oyvindth, 2021-08-30:
I have the actuator running for a few days now and it is working nicely. I still have a few features that would be fantastic if could be implemented though.

Energy/Power: The energy datapoint is a cumulative figure, adding up the amount of KWH the connected heat-source have consumed since the actuator was first initiated. In the extended status message recieved from the actuator however there is a power (sometimes reported as average power if the actuator is modulating (duty cycle less than 100%)) figure in W. This is the current Electrical usage in W. Would be great if this figure could be displayed as a separate entity.

karloygard, 2021-09-05:
I added power as a separate entity of the "new generation" actuators in 0.41.4. At startup time I don't know whether the actuator actually reports power, so all NG actuators will get a power entity, but for devices that don't ever report power, the value will always remain unknown.

Jan-Arild-Blekken, 2021-09-05:
I have tested the Multi Channel Heating Actuator (CHAZ-01/12) now. Just tried channel 5 and it is working.

oyvindth, 2021-09-06:
This is getting better and better. 

Would it be possible to add:
state_class: "meassurement" to the power entity
state_class: "total_increasing" to the energy entity

I think this would allow us to use it with the buildt inn energy integration.

[https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics](url)

karloygard, 2021-09-06:
Added the above in 0.42. No idea if it works, though, I don't have the energy integration set up yet.

oyvindth, 2021-09-14:
[image]
Working great.

Following automations at work:
1: Makes sure the heating actuator do not go into safemode (No Floorsensor)
`- id: '1630180139158'
  alias: Temperaturkontroll Gjesterom
  description: Oppdaterer number.varmeaktuator_gjesterom_current hver gang RTC sender,
    eller hvert 30Min
  trigger:
  - platform: state
    entity_id: sensor.gjesterom_temperatur
  - platform: time_pattern
    minutes: /30
  condition: []
  action:
  - service: number.set_value
    target:
      entity_id: number.varmeaktuator_gjesterom_current
    data_template:
      value: '{{ states(''sensor.gjesterom_temperatur'') }}'
  - service: mqtt.publish
    data:
      topic: xcomfort/8/set/temperature
      payload: '{{ state_attr(''climate.varmeaktuator_gjesterom'', ''temperature'')
        }}'
  mode: single`

Automation 2 Makes sure the actuator do not go into safemode, and turns off heat if floorsensor senses to high temperature.
`- id: '16309140548041'
  alias: Temperaturkontroll Vindfang
  description: Oppdaterer number.varme_vindfang_varmeinngang_current hver gang RTC
    sender, eller hvert 30Min
  trigger:
  - platform: state
    entity_id:
    - sensor.termostat_vinnfang_temperatur
    - sensor.termostat_vinnfang_ekstern_sensor_pt1000
    - input_number.max_gulv_vindfang
  - platform: time_pattern
    minutes: /30
  condition: []
  action:
  - service_template: "{% if states('sensor.termostat_vinnfang_ekstern_sensor_pt1000')\
      \ > states('input_number.max_gulv_vindfang') %}\n  script.vindfang_floor_high\n\
      {% else %}\n  script.vindfang_floor_normal\n{% endif %}\n"
  mode: single`

3. Scripst
`vindfang_floor_high:
  sequence:
  - service: mqtt.publish
    data:
      topic: xcomfort/21/set/temperature
      payload: '5'
  - service: input_boolean.turn_on
    target:
      entity_id: input_boolean.gulvvarme_vindfang
  mode: single
  alias: Vindfang Floor High
vindfang_floor_normal:
  sequence:
  - service: number.set_value
    target:
      entity_id: number.varme_vindfang_varmeinngang_current
    data_template:
      value: '{{ states(''sensor.termostat_vinnfang_temperatur'') }}'
  - service: mqtt.publish
    data:
      topic: xcomfort/21/set/temperature
      payload: "{{ state_attr('climate.varme_vindfang_varmeinngang', 'temperature') }}"
  - service: input_boolean.turn_off
    target:
      entity_id: input_boolean.gulvvarme_vindfang
  mode: single
  alias: Vindfang Floor Normal
`

closed 2021-09-15 by oyvindth