karloygard/xcomfortd-go #26: Support for new window/Door sensor CDWA-01/3x

oyvindth, 2021-09-08:
Sorry for nagging you even more. But is it possible to incorporate support for the Window/Door sensors.

The Xcomfort logs look like this.

Door Open: 
2021/09/08 22:20:10 Device 7577673 (channel 0-'Terassedør Kjøkken ()') sent message (battery new, signal good, cyclic false) event 'switchOn'

Door Closed: 
2021/09/08 22:20:48 Device 7577673 (channel 0-'Terassedør Kjøkken ()') sent message (battery new, signal good, cyclic false) event 'switchOff'

From the looks of it it works as a simple binary sensor.

Regds.
Øyvind

karloygard, 2021-09-09:
I need more info about this sensor, it sure looks like a binary sensor, but I can't find it in MRF documents or anywhere else. Can you send me the startup logs so I can see what it's detected as?

oyvindth, 2021-09-09:
How do I pull the startup logs.

karloygard, 2021-09-09:
Make sure you have verbose turned on. Restart the add-on and watch the logs, i.e. refresh until you see the relevant datapoint reported. As an example, the following is for a NG Dimming actuator:
```
2021/09/09 23:00:08 EXTENDED: [d1103e0d01800a00]
2021/09/09 23:00:09 EPROM: [d1113e0d01800a0000000000000000000000]
2021/09/09 23:00:09 Datapoint 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 7133322, channel 0, mode 0, 'Hallway downstairs (Statusforbindelse)'
2021/09/09 23:00:09 Level: 1.1.5, location [House, 1st floor, Hallway]
```
I need the corresponding log lines for the CDWA-01/3x sensor.

oyvindth, 2021-09-09:
`021/09/10 00:20:50 EXTENDED: [d110de0601800a00]
2021/09/10 00:20:50 EPROM: [d111de0601800a0000000000030000000000]
2021/09/10 00:20:50 Datapoint 13: device , serial 7577673, channel 0, mode 0, 'Terassedør Kjøkken ()'
2021/09/10 00:20:50 EXTENDED: [d110e80601800a00]
2021/09/10 00:20:50 EPROM: [d111e80601800a005661726d65204b6a656c] `

oyvindth, 2021-09-09:
In the hjelp file for MRF 2.74 it is a short description. Where it says that when connected to other intelligent units (USB-Stick, ECI, Room manager etc) use is identical with a binary switch modus 2 Don’t know it it tells you anything

karloygard, 2021-09-10:
The log lines above weren't enough to figure out the device identifier. Can you send me the entire DPL file?

oyvindth, 2021-09-10:
`1	Bereder  (Aktuator)	7338730	74	0	0	0	#000#000#000#000#0#000#000#001#010#007#	
3	Lys Kontor  (Dimmeaktuator)	7638366	77	0	0	0	#000#000#000#000#0#000#000#001#005#000#	
4	Lys Bad 1.Etg  (Aktuator)	7281988	74	0	0	0	#000#000#000#000#0#000#000#001#010#004#	
5	Lys Loftstue  (Dimmeaktuator)	7723418	77	0	0	0	#000#000#000#000#0#000#000#001#006#000#	
8	Varme Gjesterom (P)  (Varmeinngang )	7349407	81	0	0	0	#000#000#000#000#0#000#000#001#032#003#	
14	Varme Kjeller (P)  (Varmeinngang )	7311986	81	0	0	0	#000#000#000#000#0#000#000#001#032#003#	
21	Varme Vindfang  (Varmeinngang )	7809782	81	0	0	0	#000#000#000#000#0#000#000#001#030#000#	
0	Lys Kontor  (Statusforbindelse)	7638366	77	0	0	1	#000#000#000#000#0#000#000#001#005#000#	
0	Lys Bad 1.Etg  (Statusforbindelse)	7281988	74	0	0	1	#000#000#000#000#0#000#000#001#010#004#	
0	Lys Loftstue  (Statusforbindelse)	7723418	77	0	0	1	#000#000#000#000#0#000#000#001#006#000#	
6	Termostat Gjesterom  (Temperatur)	7268934	78	0	1	1	#001#000#000#000#0#000#000#002#005#000#	
7	Termostat Gjesterom  (Fuktighet)	7268934	78	1	1	1	#001#000#000#000#0#000#000#002#005#000#	
0	Varme Gjesterom (P)  (Statusforbindelse)	7349407	81	0	0	1	#000#000#000#000#0#000#000#001#032#003#	
9	Varme Gjesterom (P)  (Energi)	7349407	81	2	0	1	#000#000#000#000#0#000#000#001#032#003#	
10	Varme Gjesterom (P)  (Belastningsfeil)	7349407	81	3	0	1	#000#000#000#000#0#000#000#001#032#003#	
11	Termostat Kjeller  (Temperatur)	6923557	78	0	1	1	#001#000#000#000#0#000#000#002#005#000#	
12	Termostat Kjeller  (Fuktighet)	6923557	78	1	1	1	#001#000#000#000#0#000#000#002#005#000#	
13	Terassedør Kjøkken ()	7577673	76	0	0	1	#000#000#000#000#0#000#000#001#004#003#	
0	Varme Kjeller (P)  (Statusforbindelse)	7311986	81	0	0	1	#000#000#000#000#0#000#000#001#032#003#	
15	Varme Kjeller (P)  (Energi)	7311986	81	2	0	1	#000#000#000#000#0#000#000#001#032#003#	
16	Varme Kjeller (P)  (Belastningsfeil)	7311986	81	3	0	1	#000#000#000#000#0#000#000#001#032#003#	
0	Bereder  (Statusforbindelse)	7338730	74	0	0	1	#000#000#000#000#0#000#000#001#010#007#	
2	Bereder  (Energi)	7338730	74	2	0	1	#000#000#000#000#0#000#000#001#010#007#	
17	Bereder  (Effekt)	7338730	74	3	0	1	#000#000#000#000#0#000#000#001#010#007#	
18	Termostat Vinnfang  (Temperatur)	7765525	78	0	1	1	#003#000#000#000#0#000#000#002#005#000#	
19	Termostat Vinnfang  (Fuktighet)	7765525	78	1	1	1	#003#000#000#000#0#000#000#002#005#000#	
20	Termostat Vinnfang  (ekstern Sensor (PT1000))	7765525	78	6	1	1	#003#000#000#000#0#000#000#002#005#000#	
0	Varme Vindfang  (Statusforbindelse)	7809782	81	0	0	1	#000#000#000#000#0#000#000#001#030#000#	
22	Varme Vindfang  (Energi)	7809782	81	2	0	1	#000#000#000#000#0#000#000#001#030#000#	
23	Varme Vindfang  (Belastningsfeil)	7809782	81	3	0	1	#000#000#000#000#0#000#000#001#030#000#	
`

Her er datapunktlisten.

Ser ut som enheten ikke registrerer seg med navn. men enhetsnummer 76

karloygard, 2021-09-10:
Pushed 0.43 with support for the new sensors.

oyvindth, 2021-09-10:
Working great. Thank you for the effort.

closed 2021-09-15 by oyvindth