karloygard/xcomfortd-go #27: Is it possible to send Switch commands to the actuators

oyvindth, 2021-09-15:
When programing Xcomfort in MRF, you can programe switch actions. (Short press up, long press up, short press down, long press down) in the actuators setup.
When creating grups/senarios in MRF (Several entities on the same datapoint) being able to sent these commands would be very helpfull as it makes it very easy to create senarios without a lot of automations. is it possible to send these commands to the actuators using MQTT

karloygard, 2021-09-15:
Yeah, that should be possible. I don't see how it would integrate with HA, but it could be possible to talk straight MQTT. I can take a look when I have some spare time.

oyvindth, 2021-12-03:
Sensio uses this function to program scenes. ( https://support-ha.sensio.no/support/solutions/articles/79000116798-xcomfort-hvordan-styre-flere-lys-samtidig-scenarier- )
The idea is that, for instance a press up could activate some of the lights in a datapoint, but a press down would turn them all off. A long press up would dim some lights but not others. By doing it this way, you can programe 4 scenes to one datapoint. My plan is to send the command using an automation activated by a push button in lovelace. The main benefit by doing it this way, is that all lights activate simultanusly, and no danger of timeout in the xComfort system.

closed 2022-10-25 by oyvindth