karloygard/xcomfortd-go #29: Initialization of plugin fails while reading eprom

geeewizzz, 2021-11-03:
I have an issue sometimes when the plugin is reading the eprom from the usb stick, and it just dies at some point in the intialization and the broadcasting message to mqtt is never sent. Here is a snippet of my logs. At the bottom I have also attached my config. 
Plugin version: 0.43
core version: core-2021.10.7
supervisor version: supervisor-2021.10.8

Xcomfort addon log
> 2021/11/03 10:17:24 EXTENDED: [d110881401800a00]
2021/11/03 10:17:24 EPROM: [d111881401800a006b6a656c6c6572206261]
2021/11/03 10:17:24 EXTENDED: [d110921401800a00]
2021/11/03 10:17:24 EPROM: [d111921401800a00642074656d7020202854]
2021/11/03 10:17:24 EXTENDED: [d1109c1401800a00]
2021/11/03 10:17:25 EPROM: [d1119c1401800a006f7563682d73656e736f]
2021/11/03 10:17:25 EXTENDED: [d110a61401800a00]
2021/11/03 10:17:25 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:17:25 EPROM: [d111a61401800a0072204229000000000000]
2021/11/03 10:17:25 EXTENDED: [d110b01401800a00]
2021/11/03 10:17:25 EPROM: [d111b01401800a0000000000000000000000]
2021/11/03 10:17:25 EXTENDED: [d110ba1401800a00]
2021/11/03 10:17:25 EPROM: [d111ba1401800a0000000001000204200511]
2021/11/03 10:17:25 EXTENDED: [d110c41401800a00]
2021/11/03 10:17:25 EPROM: [d111c41401800a00400000005b005e613803]
2021/11/03 10:17:25 EXTENDED: [d110ce1401800a00]
2021/11/03 10:17:25 EPROM: [d111ce1401800a0001000098000000000000]
2021/11/03 10:17:25 Datapoint 47: device Room Controller Touch (CRCA-00/xx), serial 6809944, channel 5, mode 0, 'kjeller bad temp (Touch-sensor B)'
2021/11/03 10:17:25 Level: 2.5.0, location [Kjeller, Bad, ]
2021/11/03 10:17:25 EXTENDED: [d110d81401800a00]
2021/11/03 10:17:25 EPROM: [d111d81401800a006b6a656c6c6572206761]
2021/11/03 10:17:25 EXTENDED: [d110e21401800a00]
2021/11/03 10:17:25 EPROM: [d111e21401800a006e673120202853746174]
2021/11/03 10:17:25 EXTENDED: [d110ec1401800a00]
2021/11/03 10:17:25 EPROM: [d111ec1401800a007573666f7262696e6465]
2021/11/03 10:17:25 EXTENDED: [d110f61401800a00]
2021/11/03 10:17:26 EPROM: [d111f61401800a006c736529000000000000]
2021/11/03 10:17:26 EXTENDED: [d110001501800a00]
2021/11/03 10:17:26 EPROM: [d111001501800a0000000000000000000000]
2021/11/03 10:17:26 EXTENDED: [d1100a1501800a00]
2021/11/03 10:17:26 EPROM: [d1110a1501800a000000000104020420030f]
2021/11/03 10:17:26 EXTENDED: [d110141501800a00]
2021/11/03 10:17:26 EPROM: [d111141501800a00400000005b005e613803]
2021/11/03 10:17:26 EXTENDED: [d1101e1501800a00]
2021/11/03 10:17:26 EPROM: [d1111e1501800a0001000098000000000000]
2021/11/03 10:17:26 Datapoint 0: device Dimming Actuator New Generation (CDAx-01/xx), serial 6882412, channel 0, mode 0, 'kjeller gang1 (Statusforbindelse)'
2021/11/03 10:17:26 Level: 2.3.0, location [Kjeller, Gang, ]
2021/11/03 10:17:26 EXTENDED: [d110281501800a00]
2021/11/03 10:17:26 EPROM: [d111281501800a006b6a656c6c6572207476]
2021/11/03 10:17:26 EXTENDED: [d110321501800a00]
2021/11/03 10:17:26 EPROM: [d111321501800a0073747565203220202853]
2021/11/03 10:17:26 EXTENDED: [d1103c1501800a00]
2021/11/03 10:17:26 EPROM: [d1113c1501800a007461747573666f726269]
2021/11/03 10:17:26 EXTENDED: [d110461501800a00]
2021/11/03 10:17:27 EPROM: [d111461501800a006e64656c736529000000]
2021/11/03 10:17:27 EXTENDED: [d110501501800a00]
2021/11/03 10:17:27 EPROM: [d111501501800a0000000000000000000000]
2021/11/03 10:17:27 EXTENDED: [d1105a1501800a00]
2021/11/03 10:17:27 EPROM: [d1115a1501800a0000000001040204200609]
2021/11/03 10:17:27 EXTENDED: [d110641501800a00]
2021/11/03 10:17:27 EPROM: [d111641501800a00400000005b005e613803]
2021/11/03 10:17:27 EXTENDED: [d1106e1501800a00]
2021/11/03 10:17:31 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:17:38 RX: [c14062030000f10000422200]
2021/11/03 10:17:38 Received message from unknown datapoint 64
2021/11/03 10:17:38 Message not handled [c14062030000f10000422200]
2021/11/03 10:17:41 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:18:24 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:18:58 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:19:04 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:20:58 RX: [c13c62170000f1011042220a]
2021/11/03 10:20:58 Received message from unknown datapoint 60
2021/11/03 10:20:58 Message not handled [c13c62170000f1011042220a]
2021/11/03 10:21:04 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:21:36 RX: [c133620300ffff00003f2209]
2021/11/03 10:21:36 Received message from unknown datapoint 51
2021/11/03 10:21:36 Message not handled [c133620300ffff00003f2209]
2021/11/03 10:21:38 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:22:23 RX: [c13a62170000e200c143240b]
2021/11/03 10:22:23 Received message from unknown datapoint 58
2021/11/03 10:22:23 Message not handled [c13a62170000e200c143240b]
2021/11/03 10:23:03 RX: [c13b620400423e0000482404]
2021/11/03 10:23:03 Received message from unknown datapoint 59
2021/11/03 10:23:03 Message not handled [c13b620400423e0000482404]
2021/11/03 10:23:23 RX: [c10f62030000e400003d2509]
2021/11/03 10:23:23 Received message from unknown datapoint 15
2021/11/03 10:23:23 Message not handled [c10f62030000e400003d2509]
2021/11/03 10:24:18 RX: [c14062030000ec0000422201]
2021/11/03 10:24:18 Received message from unknown datapoint 64
2021/11/03 10:24:18 Message not handled [c14062030000ec0000422201]
2021/11/03 10:24:21 RX: [c1096204004274cccd2f2209]
2021/11/03 10:24:21 Received message from unknown datapoint 9
2021/11/03 10:24:21 Message not handled [c1096204004274cccd2f2209]
2021/11/03 10:26:01 RX: [c10c62030000d300002f2208]
2021/11/03 10:26:01 Received message from unknown datapoint 12
2021/11/03 10:26:01 Message not handled [c10c62030000d300002f2208]
2021/11/03 10:26:55 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:27:02 RX: [c13f62030000c600002f2205]
2021/11/03 10:27:02 Received message from unknown datapoint 63
2021/11/03 10:27:02 Message not handled [c13f62030000c600002f2205]
2021/11/03 10:27:18 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:27:26 RX: [c11562030000d800004c250b]
2021/11/03 10:27:26 Received message from unknown datapoint 21
2021/11/03 10:27:26 Message not handled [c11562030000d800004c250b]
2021/11/03 10:29:28 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:29:42 handle_events: error: libusb: interrupted [code -10]
2021/11/03 10:29:43 RX: [c13e62030000f20000472401]
2021/11/03 10:29:43 Received message from unknown datapoint 62
2021/11/03 10:29:43 Message not handled [c13e62030000f20000472401]
2021/11/03 10:29:43 handle_events: error: libusb: interrupted [code -10]

Config
> mqtt_client_id: xcomfort
datapoints_file: ''
eprom: true
ha_discovery: true
ha_discovery_prefix: homeassistant
ha_discovery_remove: false
verbose: true
use_hidapi: false
eci_hosts: []

karloygard, 2021-11-03:
Will take a look.

karloygard, 2021-11-07:
I've never seen this issue myself, but I attempted a fix in 0.44, give it a shot and report back.

closed 2021-11-07 by karloygard