karloygard/xcomfortd-go #30: CHAZ-01/01 supported?

kstormhansen, 2021-11-09:
I have connected one CRCA-00/04 thermostat and one CHAZ-01/01 actuator (connected to a Dimplex oven) for testing. The thermostat works great, but for the CHAZ-01/01 I'm only getting signal strength and the following errors in the Xcomfort log:

At initial loading from ECI:
- unknown status 0 for unsupported device 55
(device 55 refers to CHAZ-01/xx)
- Device 5939337 (channel 0-'Varme trimrom (Settpunktsinnstilling)') sent message (battery mains-powered, signal weak, cyclic false) unknown

Repeating:
-Unknown channel 0 for device E-Radiator Actuator (CHAZ-01/xx)

Is the CHAZ-01/01 supported?

karloygard, 2021-11-09:
Currently unsupported, but I can take a look at it.

kstormhansen, 2021-11-09:
That is very much appreciated!

karloygard, 2021-11-11:
Attempted fix in 0.45, please try. There is scant information about the workings of this device and it seems to have been discontinued a while ago, so I am having a very hard time finding out how this device really works.

Turn on verbose and send me logs, so that I can verify that things are working. You will have to follow the instructions in https://github.com/karloygard/hassio-addons/blob/master/xcomfort/DOCS.md to send the current temperature to the actuator. I don't know if the actuator needs updated values every 30 minutes, the documentation I found said nothing about it. Other than that, the actuator will show up as a climate device in Home Assistant, and should be trivially controllable.

kstormhansen, 2021-11-12:
Thanks a lot, updated to 0.45 and get the log below. I understand I need to add the following code

  description: Update current temp on change or every 30 min
  trigger:
  - platform: state
    entity_id: sensor.{name of the sensor}
  - platform: time_pattern
    minutes: '/30'
  condition: []
  action:
  - service: number.set_value
    target:
      entity_id: number.{name of the heating actuator}_current
    data_template:
      value: '{{ states(''sensor.{name of the sensor}'') }}'

 Should this be in the main configuration file?

Log:
2021/11/12 08:11:11 EPROM: [d1115a0001800a002032313a31303a323920]
2021/11/12 08:11:11 EXTENDED: [d110640001800a00]
2021/11/12 08:11:11 EPROM: [d111640001800a0000000000000000000000]
2021/11/12 08:11:11 EXTENDED: [d1106e0001800a00]
2021/11/12 08:11:11 EPROM: [d1116e0001800a000000bd00486a656d6d65]
2021/11/12 08:11:11 EXTENDED: [d110780001800a00]
2021/11/12 08:11:11 EPROM: [d111780001800a00204d5246206d69677261]
2021/11/12 08:11:11 EXTENDED: [d110820001800a00]
2021/11/12 08:11:11 EPROM: [d111820001800a0074652076650044da7500]
2021/11/12 08:11:11 EXTENDED: [d1108c0001800a00]
2021/11/12 08:11:11 EPROM: [d1118c0001800a0045746865726e65742d43]
2021/11/12 08:11:11 EXTENDED: [d110960001800a00]
2021/11/12 08:11:11 EPROM: [d111960001800a0049000000000000000000]
2021/11/12 08:11:11 EXTENDED: [d110a00001800a00]
2021/11/12 08:11:11 EPROM: [d111a00001800a00000000409dba83d1fd88]
2021/11/12 08:11:11 EXTENDED: [d110aa0001800a00]
2021/11/12 08:11:11 EPROM: [d111aa0001800a008d0d632ad85500000000]
2021/11/12 08:11:11 EXTENDED: [d110b40001800a00]
2021/11/12 08:11:11 EPROM: [d111b40001800a0000000030c0a801bd0000]
2021/11/12 08:11:11 EXTENDED: [d110be0001800a00]
2021/11/12 08:11:11 EPROM: [d111be0001800a000c0000002200f8010180]
2021/11/12 08:11:11 EXTENDED: [d110f80101800a00]
2021/11/12 08:11:11 EPROM: [d111f80101800a0002200c3120657461736a]
2021/11/12 08:11:11 EXTENDED: [d110020201800a00]
2021/11/12 08:11:11 EPROM: [d111020201800a00650001200b4b6a656c6c]
2021/11/12 08:11:11 EXTENDED: [d1100c0201800a00]
2021/11/12 08:11:11 EPROM: [d1110c0201800a0065720002400b5472696d]
2021/11/12 08:11:11 EXTENDED: [d110160201800a00]
2021/11/12 08:11:11 EPROM: [d111160201800400726f6d00]
2021/11/12 08:11:11 EXTENDED: [d110c80001800a00]
2021/11/12 08:11:11 EPROM: [d111c80001800a005661726d65207472696d]
2021/11/12 08:11:11 EXTENDED: [d110d20001800a00]
2021/11/12 08:11:11 EPROM: [d111d20001800a00726f6d20285365747470]
2021/11/12 08:11:11 EXTENDED: [d110dc0001800a00]
2021/11/12 08:11:11 EPROM: [d111dc0001800a00756e6b7473696e6e7374]
2021/11/12 08:11:11 EXTENDED: [d110e60001800a00]
2021/11/12 08:11:11 EPROM: [d111e60001800a00696c6c696e6729000000]
2021/11/12 08:11:11 EXTENDED: [d110f00001800a00]
2021/11/12 08:11:11 EPROM: [d111f00001800a0000000000000000000000]
2021/11/12 08:11:11 EXTENDED: [d110fa0001800a00]
2021/11/12 08:11:11 EPROM: [d111fa0001800a0000000008000101200202]
2021/11/12 08:11:11 EXTENDED: [d110040101800a00]
2021/11/12 08:11:11 EPROM: [d111040101800a00400000004c656b6b6173]
2021/11/12 08:11:11 Datapoint 2: device E-Radiator Actuator (CHAZ-01/xx), serial 5939337, channel 0, mode 0, 'Varme trimrom (Settpunktsinnstilling)'
2021/11/12 08:11:11 Level: 1.2.0, location [Kjeller, Trimrom, ]
2021/11/12 08:11:11 EXTENDED: [d110080101800a00]
2021/11/12 08:11:11 EPROM: [d111080101800a004c656b6b61736a652062]
2021/11/12 08:11:11 EXTENDED: [d110120101800a00]
2021/11/12 08:11:11 EPROM: [d111120101800a0061642068616c6c202028]
2021/11/12 08:11:11 EXTENDED: [d1101c0101800a00]
2021/11/12 08:11:11 EPROM: [d1111c0101800a004b616e616c2041290000]
[... 24 lines cut ...]
2021/11/12 08:11:11 EXTENDED: [d110940101800a00]
2021/11/12 08:11:11 EPROM: [d111940101800a0040000000000000000000]
2021/11/12 08:11:11 EXTENDED: [d1109e0101800a00]
2021/11/12 08:11:11 EPROM: [d1119e0101800a0000000000030000000000]
2021/11/12 08:11:11 Datapoint 3: device Room Controller (with Switch) (CRCA-00/01..04), serial 6129618, channel 0, mode 1, 'Romkontroll trimrom'
2021/11/12 08:11:11 Level: 1.2.0, location [Kjeller, Trimrom, ]
2021/11/12 08:11:11 EXTENDED: [d110a80101800a00]
2021/11/12 08:11:11 EPROM: [d111a80101800a0044f8722067617261736a]
2021/11/12 08:11:11 EXTENDED: [d110b20101800a00]
2021/11/12 08:11:11 EPROM: [d111b20101800a0065202829000000000000]
2021/11/12 08:11:11 EXTENDED: [d110bc0101800a00]
2021/11/12 08:11:12 EPROM: [d111bc0101800a0000000000000000000000]
2021/11/12 08:11:12 EXTENDED: [d110c60101800a00]
2021/11/12 08:11:12 EPROM: [d111c60101800a0000000000000000000000]
2021/11/12 08:11:12 EXTENDED: [d110d00101800a00]
2021/11/12 08:11:12 EPROM: [d111d00101800a0000000000000000000000]
2021/11/12 08:11:12 EXTENDED: [d110da0101800a00]
2021/11/12 08:11:12 EPROM: [d111da0101800a0000000001040000000000]
2021/11/12 08:11:12 EXTENDED: [d110e40101800a00]
2021/11/12 08:11:12 EPROM: [d111e40101800a0000000000000000000000]
2021/11/12 08:11:12 EXTENDED: [d110ee0101800a00]
2021/11/12 08:11:12 EPROM: [d111ee0101800a0000000000030000000000]
2021/11/12 08:11:12 Datapoint 4: device Door/window sensor (CDWA-01/3x), serial 7581257, channel 0, mode 0, 'Dør garasje ()'
2021/11/12 08:11:12 Read datapoint list from eprom in 552.410683ms
2021/11/12 08:11:12 Sent MQTT autodiscover add for 4 devices and 4 datapoints

karloygard, 2021-11-14:
Did you substitute `{name of the sensor}`and `{name of the heating actuator}` in the automation?

You seem to already have a room controller, is that controlling the heating actuator?

kstormhansen, 2021-11-15:
> Did you substitute `{name of the sensor}`and `{name of the heating actuator}` in the automation?
> 
> You seem to already have a room controller, is that controlling the heating actuator?

Yes, here is the full automation in the automations.yaml:

  description: Update current temp on change or every 30 min
  trigger:
  - platform: state
    entity_id: sensor.romkontroll_trimrom
  - platform: time_pattern
    minutes: '/30'
  condition: []
  action:
  - service: number.set_value
    target:
      entity_id: number.varme_trimrom_settpunktsinnstilling_current
    data_template:
      value: '{{ states(''sensor.romkontroll_trimrom'') }}'

I have a sensor (CRCA 00/04) named romkontroll_trimrom which should switch the CHAZ on/off. These are not connected in MRF, both are only connected to the ECI.

I tried to create a thermostat with the romkontroll_trimrom sensor and the CHAZ heating actuator. This did not work. I have not figured out how to turn the CHAZ on/off yet. New to Home Assistant.

karloygard, 2021-11-15:
Don't worry about setting the operation, it should be updated by the actuator itself. Also, you shouldn't have to create a new thermostat, the actuator should already show up as a climate entity.

The MRF log is useful, it clearly states that the setpoint is sent to the actuator and acked. If you manually set `number.varme_trimrom_settpunktsinnstilling_current`, that temperature should be sent to the actuator, and it should attempt to reach that temperature. Can you use MRF to verify that the value is sent to the actuator?

kstormhansen, 2021-11-15:
> Don't worry about setting the operation, it should be updated by the actuator itself. Also, you shouldn't have to create a new thermostat, the actuator should already show up as a climate entity. [...]

Perhaps I got this backwards. I thought `number.varme_trimrom_settpunktsinnstilling_current` is the current temperature, which is sent to the actuator from the sensor through the automation script? 

I have made a thermostat in HA with the entity `climate.varme_trimrom_settpunktsinnstilling`. 

[image]

When a target temperature is set, this is sent to the actuator with the MRF log below:

[image]

karloygard, 2021-11-15:
You were correct, `number.varme_trimrom_settpunktsinnstilling_current` is the current room temperature, `climate.varme_trimrom_settpunktsinnstilling` is the desired temperature.

The MRF log looks somewhat strange, although it could be showing the actual routing of the messages. Have you double checked that all actuators are updated with correct data and that the DPL has been updated in the ECI?

If everything else is in order, you can use MRF to route the room controller directly to the CHAZ and log the messages that they are sending between them. That way we can verify the operation of the heating actuator and see that we are sending the correct messages to it. (The room controller we already know works.)

kstormhansen, 2021-11-17:
> You were correct, `number.varme_trimrom_settpunktsinnstilling_current` is the current room temperature, `climate.varme_trimrom_settpunktsinnstilling` is the desired temperature.
>
> The MRF log looks [...]

I have gone through the MRF set-up, replaced the heating actuator (in case it was fault), set up a seperate MRF project with only sensor, ECI and heating actuator (to eliminate any conflict with SHC and routing). Still same effect, see log below:

[image]

I have a number of CHAZ 01/01 connected to SHC, and looking at the log I see the following differences:
- SHC never uses "Office-Mode", mainly Comfort and Eco
- In line 3 in the log above, the sensor send current temperature to ECI, followed by "1308.8K". Not sure what this means, but in with SHC this is always "0.0K".

I tried to connect the sensor directly to the CHAZ in MRF. This is not possible with this thermostat (CRCA 00/04). I think CRCA 00/06 or 07 is needed for this operation.

Everything seems to work good except that I'm unable to turn heating on/off in the HA thermostat.

karloygard, 2021-11-17:
If it works with SHC, that helps a lot. Can you capture some more of the traffic from SHC?

karloygard, 2021-11-18:
Tweaked the messages in 0.46, test it and report back. In particular interested in whether the messages in MRF now are the same as from the SHC.

kstormhansen, 2021-11-19:
Thanks again for your help on this. I have updated to version 0.46. Still not turning on heat, here are logs from HA and MRF:

Sending a temperature in HA to CHAZ:
[image]

Sending a temperature in MRF to CHAZ:
[image]

Logs from other CHAZ-01/01 in the system connected to SHC and working properly:

Sending temperature from SHC to CHAZ (comfort mode changing temp to 22 degrees):
[image]

Sending Economy mode (19.5 degrees) from SHC to CHAZ:
[image]

Turning off heating in the zone (through SHC app):
[image]

Turning on heating in the zone /through SHC app):
[image]

The last log from turning on heating in the zone is interesting. SHC initially sends temperature 0,0 and the the set temperature (21,5). Here is the enture log turning on heat for 3 CHAZ-01/01 in kitchen (1) and living room (2):

[image]

Hope this helps :-)

karloygard, 2021-11-19:
Looks fine so far, the setpoint commands are at least the same as the SHC commands. Can you do the same test with sending the current temperature?

kstormhansen, 2021-11-19:
The HA sensor in MRF log:
[image]

The same type sensor (CRCA 00/04) sending temperature to SHC:
[image]

karloygard, 2021-11-22:
This also looks fine, "Romkontroll trimrom" is sending 16.6C + wheel is turned down to -3C. We presently just drop the wheel position on the ground, but that's under control.

I need to see a message where ECI is sending a temperature command to the CHAZ. Formerly, this had the 1305.8K wheel position, which seemed completely broken.

kstormhansen, 2021-11-22:
It looks OK now (temp sent to CHAZ when the automation is executed):
[image]

karloygard, 2021-11-22:
Yeah, you can't turn the actuator on and off with the flame button, only by setting the current temperature.

Awesome that you got it to work! I'm a little confused by the 0.4 value, I'm sending zero, but if it works...

closed 2021-11-25 by karloygard

giancum, 2023-01-24:
Dear @kstormhansen can you tell how you made it work? I am not able to get the multiheat channel status in HA, so it is always unkown. 
flame is always off. 
thank you. 
@karloygard thank you for your great work.