karloygard/xcomfortd-go #33: Future request: Push buttons

DiseaseNO, 2021-12-09:
Hi! Would it be possible for a future version to support the push buttons from xcomfort as well? The integrations is by the way working really good and stable.

karloygard, 2021-12-10:
This should already work—you can trigger automations with the push buttons.

DiseaseNO, 2021-12-19:
Thanks. I dont think I understand. 

I have multiple Xcomfort pust buttons in my house and I have connected these in MRF to connect to the 00/14 USB, and when I press, then I can see the buttons getting pressed in debug of the plugin. However I cannot see anything about the push buttons in MQTT.

karloygard, 2021-12-19:
The way it's set up in HA today, pushbuttons can only be used for triggering automations. I have only tested this briefly a long time ago, but I believe it should be working fine still. See the image below:

[image]

closed 2021-12-29 by karloygard