karloygard/xcomfortd-go #34: HA complaining about temp sensors reporting with unsupported unit

E-K-I, 2021-12-14:
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:269
Integration: Sensor (documentation, issues)
First occurred: 14:20:10 (4 occurrences)
Last logged: 15:00:10

sensor.temperature has unit C which is unsupported for device_class temperature
sensor.bad_vest_temperature has unit C which is unsupported for device_class temperature
sensor.temperature_adjustment_3 has unit C which is unsupported for device_class temperature
sensor.tekniskrom_temperatur has unit C which is unsupported for device_class temperature

karloygard, 2021-12-19:
Yeah, that should probably be "°C". Will issue a fix.

karloygard, 2021-12-20:
Resolved in 0.49.

closed 2021-12-20 by karloygard