karloygard/xcomfortd-go #35: Adding the CKOZ-14 stick to MRF

bjoornoor, 2021-12-17:
Hi! I realise this is probably a "noob" question, but i'm struggling a bit with adding the CKOZ/14 stick to the Eaton-RF program in order to transfer the datapoint file and get started with home assistant. I have both the 13 and 14 stick, but i can't figure out how i add the stick to Eaton-RF.
I have managed to add it with insert - devices by barcode, but then i get a purple square around the stick and i'm a bit unsure on what the next step is in order to get it to communicate with the rest of the units. 
Is the barcode insert the correct way, or is there some other trick?  
I can't see it specifically mentioned in Windows Device Manager with a Eaton name, but i'm unsure if that's the problem.

karloygard, 2021-12-17:
I assume you are already using MRF. The CKOZ-00/14 (aka CI stick) is included in the network in the same manner as other devices, just scan for new devices in MRF. Draw connections to and from the CI stick and upload the datapoints file to it when you are satisfied. The stick needs to be powered when you do this, but doesn't have to talk to the host computer, so device drivers are not an issue.

bjoornoor, 2021-12-18:
Thank you for the reply. I tried some more last night and eventually got it scanned after i plugged out the Logitech Unifying Receiver from another usb-port. I dont know if that was causing some interference problems in the past or if it was just a pure coincidence. Should hopefully get started with Home Assistant now, thanks again for the help :-)

closed 2021-12-20 by karloygard