karloygard/xcomfortd-go #36: no ACK received

DiseaseNO, 2021-12-28:
Hi!

I have had this working great for a couple of weeks. But suddenly it does not work anymore and the connection to the xcomfort products does not work. See the error below:

`2021/12/28 20:40:38 Level: 6.0.0, location [, , ]
2021/12/28 20:40:38 Read datapoint list from eprom in 32.384314608s
2021/12/28 20:40:38 Sent MQTT autodiscover add for 13 devices and 18 datapoints
2021/12/28 20:40:42 MQTT message; topic: 'xcomfort/3/set/dimmer', message: '100'
2021/12/28 20:40:42 TX (seq 1, 1 parallel): [b1030d406410] 
2021/12/28 20:40:52 TX message was silently lost, likely never sent
2021/12/28 20:40:52 TX command failed, retrying (1/2): timeout, no ACK received
2021/12/28 20:40:52 TX (seq 2, 1 parallel): [b1030d406420] 
2021/12/28 20:41:02 TX message was silently lost, likely never sent
2021/12/28 20:41:02 TX command failed, retrying (2/2): timeout, no ACK received
2021/12/28 20:41:02 TX (seq 3, 1 parallel): [b1030d406430] 
2021/12/28 20:41:12 TX message was silently lost, likely never sent
2021/12/28 20:41:12 WARNING: command for datapoint 3 failed, state now unknown: timeout, no ACK received`

[image]
[image]

Any tips? I have recalculated all routing and the reception is good on the tested device.

DiseaseNO, 2021-12-29:
I fixed it! 

I deleted all connecting lines i MRF and removed all devices from the USB 00/14. Then I uploaded the empty datapoints list to the stick. 

After I added them again all devices working properly again.

karloygard, 2021-12-29:
Weird. If this happens again, it's sometimes useful to use MRF to look at the messages being sent. This way, you can at least see if the messages are being sent.

closed 2021-12-29 by karloygard