karloygard/xcomfortd-go #37: Is it possible to get the name and not the serial number in discovery?

hans99, 2021-12-30:
In MQTT integration the name is set to the serial number, but the MQTT info has the real name, would it be possible to get the real name as name instead?

karloygard, 2021-12-30:
Maybe, let me dig a bit.

hans99, 2022-01-20:
Any progress on this? :)

karloygard, 2022-01-23:
I attempted a fix for this with 0.51, take a look. It's relatively simplistic, using the common part of the datapoint names for the device (the device itself is not explicitly named).

hans99, 2022-01-25:
Great, I will try it out soon!

hans99, 2022-02-03:
Devices looks good, great work! Thanks!
Would it be possible to get the names reflected in the entities too?

karloygard, 2022-02-03:
Can you give me an example?

hans99, 2022-02-11:
I think hass will add numbers for duplicates, I think that's we there is a number now. So if "we" assign name that makes duplicates its "our" fault and nothing to be handled by "you" :)