karloygard/xcomfortd-go #38: Tilfeldige enheter slutter å funkere

shoslemo, 2022-01-10:
I mitt oppsett opplever jeg at det alltid er en eller flere enheter som ikke fungerer. De svarer ikke på kommandoer sendt fra HA, og de oppdaterer ikke status. Starter jeg xcomfort-addonen på nytt, kommer det seg ofte, men da er det plutselig en eller to andre enheter som ikke fungerer igjen. Gjelder spesielt dimmere og varmeaktuatorer virker det som. Bruker forøvrig en CI fra Eaton som grensesnitt, (den med PoE).
Har prøvd de tingene jeg kan tenke meg, sjekket brannmurregler, lastet opp mrf på nytt, fremstilt ny datapunktliste, fjernet mellomrom og æøå i navn på aktuatorer osv., men jeg sliter fortsatt med at en til to enheter til enhver tid er utilgjengelige. Veldig godt mulig jeg har gjort noen nybegynnerfeil i konfigurasjonen min, men jeg klarer bare ikke å se hva det er. Noen tips..?

karloygard, 2022-01-11:
`handle_events: error: libusb: interrupted [code -10]` betyr ikke noe, du kan trygt se bort fra dem.

Dersom meldinger forsvinner og/eller mistes, skal det synes i loggene, kan du se etter hva du finner og eventuelt poste her?

shoslemo, 2022-01-11:
Når jeg går gjennom loggene finner jeg meldingene. De blir sendt fra MQTT til xcomfort-tillegget, men det kommer ingen bekreftelser eller noen ting. Har oppe overvåkning av xcomfort-traffikk via mrf, og det hverken blinker i grensesnittet eller kommer opp data når jeg betjener de enhetene som ikke fungerer. Denne gangen er det bare en enhet, men før forrige restart av xcomfort var det tre. For meg virker det som at meldingene stopper opp i xcomfort-addonen, før de blir sendt videre til det fysiske grensesnittet..?

Dette er alt jeg finner av loggoppføringer, den øverste fra din addon, den underste fra MQTT-logger.

2022/01/11 18:49:21 MQTT message; topic: 'xcomfort/8/set/dimmer', message: '100'

2022-01-11 18:49:21 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on xcomfort/8/set/dimmer: '100', mid: 102

karloygard, 2022-01-11:
MQTT-loggen er ikke så viktig, men jeg trenger logg fra når du ser at meldinger går tapt. Det er nyttig å kryssreferere dette med trafikklogg fra MRF, med klokkeslett er det ikke så vanskelig å synkronisere loggene.

Når vi har litt bedre oversikt, kan du skru på verbose, for å få enda bedre innsikt i hva det er som foregår.

shoslemo, 2022-01-11:
Det som vanligvis dukker opp i loggen er følgende: MQTT message; topic: 'xcomfort/8/set/dimmer', message: '100'
Det kommer ikke noe mer enn dette, trykker jeg på funksjonen flere ganger, kommer bare denne oppføringen om igjen. Med de enhetene som virker kommer det en del andre oppføringer i tillegg. 

Har uthevet loggoppføringen.  I motsetning til de andre som fungerer kommer det aldri noe mer enn denne ene linjen før jeg restarter addonen og det er en annen enhet som "tar over" og begynner å oppføre seg på samme måte. 

2022/01/11 20:25:08 handle_events: error: libusb: interrupted [code -10]
2022/01/11 20:25:09 Status list messages currently ignored: 62
2022/01/11 20:25:09 Device 7858117, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 10C (battery mains-powered, signal weak)
**2022/01/11 20:25:41 MQTT message; topic: 'xcomfort/19/set/dimmer', message: '100'**
2022/01/11 20:25:56 handle_events: error: libusb: interrupted [code -10]
2022/01/11 20:26:15 handle_events: error: libusb: interrupted [code -10]
2022/01/11 20:26:18 Status list messages currently ignored: 62
2022/01/11 20:26:18 Device 4974862 (channel 1-'T3 (Kanal B)') sent message (battery new, signal good, cyclic true) event 'value' with value 2.6

karloygard, 2022-01-11:
Ok, skru på verbose og kjør samme test igjen.

shoslemo, 2022-01-11:
Samme problemet, men denne gangen prøvde addonen å sende meldingen på nytt etter en stund, dette har den gjort noen få ganger før også men som oftest gjør den ikke det, og det er kun loggoppføringen som er uthevet som kommer. 

Er det noen måte å tvinge sending av ny kommando selv om det ikke er motatt bekreftelse på forrige..? Evt. tvinge sending selv om status på enheten er ukjent..? De gangene meldingene ikke blir oppfattet som ikke bekreftet får jeg ikke sende noe som helst før jeg restarter, uansett hvor lenge jeg venter.

Verbose på:

2022/01/11 22:34:10 Device 4974965 (channel 1-'T2 (Kanal B)') sent message (battery new, signal good, cyclic true) event 'value' with value 38.8
2022/01/11 22:34:35 handle_events: error: libusb: interrupted [code -10]
2022/01/11 22:34:35 EPROM: [d162000000000000]
2022/01/11 22:34:35 Status list messages currently ignored: 62
2022/01/11 22:34:35 RX: [c114620300008d0000472507]
2022/01/11 22:34:35 Device 4974862 (channel 0-'T3 (Kanal A)') sent message (battery new, signal normal, cyclic true) event 'value' with value 14.1
2022/01/11 22:34:37 handle_events: error: libusb: interrupted [code -10]
2022/01/11 22:34:39 handle_events: error: libusb: interrupted [code -10]
2022/01/11 22:34:46 handle_events: error: libusb: interrupted [code -10]
2022/01/11 22:35:06 handle_events: error: libusb: interrupted [code -10]
**_2022/01/11 22:35:17 MQTT message; topic: 'xcomfort/9/set/dimmer', message: '6'_**
2022/01/11 22:35:36 handle_events: error: libusb: interrupted [code -10]
2022/01/11 22:35:49 handle_events: error: libusb: interrupted [code -10]
2022/01/11 22:36:07 TX message was silently lost, likely never sent
2022/01/11 22:36:07 TX command failed, retrying (1/2): timeout, no ACK received
2022/01/11 22:36:07 TX (seq 3, 1 parallel): [b1090a0030] 
2022/01/11 22:36:10 EPROM: [d162000000000000]
2022/01/11 22:36:10 Status list messages currently ignored: 62
2022/01/11 22:36:10 STATUS: [c31c0430100000]
2022/01/11 22:36:10 TX (seq 4, 1 parallel): [b1090d400640] 
2022/01/11 22:36:11 EPROM: [d162000000000000]
2022/01/11 22:36:11 Status list messages currently ignored: 62
2022/01/11 22:36:11 STATUS: [c31c0440100000]
2022/01/11 22:36:13 handle_events: error: libusb: interrupted [code -10]
2022/01/11 22:36:13 EPROM: [d162000000000000]
2022/01/11 22:36:13 Status list messages currently ignored: 62
2022/01/11 22:36:13 RX: [c100733900954368004d000210001e00000035100c]
2022/01/11 22:36:13 Device 6833045, type CDAU 01/04 sent extended status message: status ON, value 16, temp 30C (battery mains-powered, signal good)

karloygard, 2022-01-11:
Koden skal i utgangspunktet takle dette–har tidligere sett problemer med at meldinger bare forsvinner sporløst, selv om man vanligvis får tilbakemelding om at meldinger ikke er sendt ikke blir acket.

`TX message was silently lost, likely never sent` er et slikt case, men her detekteres feilen og meldingen sendes på nytt.

Jeg trenger å se mer logging (aller helst hele, slik at jeg kan se hvilke devicer du har koblet opp) for å forstå hvorfor koden ikke henter seg inn igjen her, godt mulig at det er en bug et sted. Som nevnt, er det nyttig med MRF-logging også, slik at jeg kan se om meldingene faktisk blir sent på eteren eller ikke.

ECIen er oppgradert med siste firmware?

shoslemo, 2022-01-12:
Her er utdrag fra logger fra både MRF og addon. Har uthevet hendelsene i addon-loggen, disse finnes ikke i MRF-logg da de aldri blir sendt ut til CI (tror jeg).

Logg MRF:

12.1.2022	09:08:58	+?	Ethernet-CI (DP: 7) sender <AV> til Pendler_Spisestue      (Netzversorgt)	76 %	
12.1.2022	09:08:58	+0.0902	Pendler_Spisestue sender <BEKREFTELSE> til Ethernet-CI (DP:7)      (Netzversorgt)	78 %	
12.1.2022	09:08:58	+0.1253	Ethernet-CI (DP: 12) sender <AV> til Barskap_Vegg      (Netzversorgt)	78 %	
12.1.2022	09:08:58	+0.0899	Barskap_Vegg sender <BEKREFTELSE> til Ethernet-CI (DP:12)      (Netzversorgt)	92 %	
12.1.2022	09:08:58	+0.1277	Ethernet-CI (DP: 16) sender <AV> til Stalampe      (Netzversorgt)	82 %	
12.1.2022	09:08:58	+0.0894	Stalampe sender <BEKREFTELSE> til Ethernet-CI (DP:16)      (Netzversorgt)	70 %	
12.1.2022	09:08:58	+0.1549	Ethernet-CI (DP: 6) sender <AV> til DL_Stue      (Netzversorgt)	82 %	
12.1.2022	09:08:59	+0.0903	DL_Stue sender <BEKREFTELSE> til Ethernet-CI (DP:6)      (Netzversorgt)	100 %	
12.1.2022	09:08:59	+0.1236	Ethernet-CI (DP: 8) sender <AV> til DL_Kjokken      (Netzversorgt)	82 %	
12.1.2022	09:08:59	+0.0899	DL_Kjokken sender <BEKREFTELSE> til Ethernet-CI (DP:8)      (Netzversorgt)	86 %	
12.1.2022	09:08:59	+0.1294	Ethernet-CI (DP: 13) sender <AV> til Pendel_Trapp      (Netzversorgt)	82 %	
12.1.2022	09:08:59	+0.0918	Pendel_Trapp sender <BEKREFTELSE> til Ethernet-CI (DP:13)      (Netzversorgt)	96 %	
12.1.2022	09:08:59	+0.1441	DL_Kjokken sender <Status gammel> til Ethernet-CI (DP:8)      (Netzversorgt)	86 %	
12.1.2022	09:09:00	+0.1607	Barskap_Vegg sender <Status gammel> til Ethernet-CI (DP:12)      (Netzversorgt)	90 %	
12.1.2022	09:09:00	+0.1739	Ethernet-CI (DP: 14) sender <AV> til DL_Spisestue      (Netzversorgt)	78 %	
12.1.2022	09:09:00	+0.0900	DL_Spisestue sender <BEKREFTELSE> til Ethernet-CI (DP:14)      (Netzversorgt)	80 %	
12.1.2022	09:09:00	+0.2150	Ethernet-CI (DP: 15) sender <AV> til DL_Bad  ( Ethernet-CI => Pendel_Trapp )     (Netzversorgt)	86 %	
12.1.2022	09:09:00	+0.0900	Pendel_Trapp sender <BEKREFTELSE> til Ethernet-CI      (Netzversorgt)	98 %	
12.1.2022	09:09:00	+0.0902	Ethernet-CI (DP: 15) sender <AV> til DL_Bad  ( Pendel_Trapp => DL_Bad )     (Netzversorgt)	96 %	
12.1.2022	09:09:01	+0.2824	Ethernet-CI (DP: 19) sender <AV> til Salong      (Netzversorgt)	70 %	
12.1.2022	09:09:01	+0.0921	Salong sender <BEKREFTELSE> til Ethernet-CI (DP:19)      (Netzversorgt)	100 %	
12.1.2022	09:09:01	+0.1659	Ethernet-CI (DP: 5) sender <AV> til Lys_Sov_Oppe  ( Ethernet-CI => Speillys )     (Netzversorgt)	78 %	
12.1.2022	09:09:01	+0.0903	Speillys sender <BEKREFTELSE> til Ethernet-CI      (Netzversorgt)	70 %	
12.1.2022	09:09:01	+0.0902	Ethernet-CI (DP: 5) sender <AV> til Lys_Sov_Oppe  ( Speillys => Lys_Sov_Oppe )     (Netzversorgt)	68 %	
12.1.2022	09:09:01	+0.0895	Lys_Sov_Oppe sender <BEKREFTELSE> til Speillys      (Netzversorgt)	56 %	
12.1.2022	09:09:05	+3.4999	Pendler_Spisestue sender Status til Ethernet-CI  ( <Dimmeaktuator> - Var.: 0 Utgang: Av, 0%, Temperatur: 11°C, Last: OK )	84 %	
12.1.2022	09:09:05	+0.0899	Ethernet-CI sender <BEKREFTELSE> til Pendler_Spisestue      (Netzversorgt)	88 %	
12.1.2022	09:09:06	+0.6674	DL_Stue sender Status til Ethernet-CI  ( <Dimmeaktuator> - Var.: 1 Utgang: Av, 0%, Binærinngang: Av/Av, Temperatur: 30°C, Last: OK )	100 %	
12.1.2022	09:09:06	+0.0900	Ethernet-CI sender <BEKREFTELSE> til DL_Stue      (Netzversorgt)	64 %	
12.1.2022	09:09:06	+0.3714	Pendel_Trapp sender Status til Ethernet-CI  ( <Dimmeaktuator> - Var.: 0 Utgang: Av, 0%, Temperatur: 31°C, Last: OK )	94 %	
12.1.2022	09:09:06	+0.0900	Ethernet-CI sender <BEKREFTELSE> til Pendel_Trapp      (Netzversorgt)	80 %	
12.1.2022	09:09:07	+0.5298	DL_Spisestue sender Status til Ethernet-CI  ( <Dimmeaktuator> - Var.: 0 Utgang: Av, 0%, Temperatur: 12°C, Last: OK )	80 %	
12.1.2022	09:09:07	+0.0899	Ethernet-CI sender <BEKREFTELSE> til DL_Spisestue      (Netzversorgt)	74 %	
12.1.2022	09:09:08	+0.7495	Salong sender Status til Ethernet-CI  ( <Dimmeaktuator> - Var.: 0 Utgang: Av, 0%, Temperatur: 11°C, Last: OK )	100 %	
12.1.2022	09:09:08	+0.0900	Ethernet-CI sender <BEKREFTELSE> til Salong      (Netzversorgt)	66 %	
12.1.2022	09:09:08	+0.5075	Stalampe sender Status til Ethernet-CI  ( <Dimmeaktuator> - Var.: 14 Utgang: Av, 0%, Temperatur: 29°C, Effekt: 0.0W, Lastfeil-utgang: OK, Last: OK )	74 %	
12.1.2022	09:09:08	+0.0899	Ethernet-CI sender <BEKREFTELSE> til Stalampe      (Netzversorgt)	62 %	
12.1.2022	09:09:39	+31 Sek.	Ethernet-CI (DP: 19) sender <PROSENTVERDI> til Salong      (Netzversorgt)	72 %	
12.1.2022	09:09:39	+0.0901	Salong sender <BEKREFTELSE> til Ethernet-CI (DP:19)      (Netzversorgt)	100 %	
12.1.2022	09:09:40	+0.7905	Ethernet-CI (DP: 19) sender <AV> til Salong      (Netzversorgt)	72 %	
12.1.2022	09:09:40	+0.0901	Salong sender <BEKREFTELSE> til Ethernet-CI (DP:19)      (Netzversorgt)	100 %	
12.1.2022	09:09:41	+1.2141	Salong sender Status til Ethernet-CI  ( <Dimmeaktuator> - Var.: 0 Utgang: På, 100%, Temperatur: 11°C, Last: OK )	100 %	
12.1.2022	09:09:41	+0.0899	Ethernet-CI sender <BEKREFTELSE> til Salong      (Netzversorgt)	62 %	
12.1.2022	09:09:43	+1.2220	Ethernet-CI (DP: 16) sender <PROSENTVERDI> til Stalampe      (Netzversorgt)	76 %	
12.1.2022	09:09:43	+0.4000	Ethernet-CI (DP: 16) sender <PROSENTVERDI> til Stalampe      (Netzversorgt)	74 %	
12.1.2022	09:09:43	+0.0894	Stalampe sender <BEKREFTELSE> til Ethernet-CI (DP:16)      (Netzversorgt)	74 %	
[... cut ...]
2022/01/12 09:11:40 EPROM: [d162000000000000]
2022/01/12 09:11:40 Status list messages currently ignored: 62
2022/01/12 09:11:40 RX: [c10253000000000000433001]
2022/01/12 09:11:40 Device 7679249 (channel 2-'Inng_Brannsentral (Binærinngang B)') sent message (battery mains-powered, signal good, cyclic true) event 'switchOff'
2022/01/12 09:12:03 handle_events: error: libusb: interrupted [code -10]
2022/01/12 09:12:07 handle_events: error: libusb: interrupted [code -10]
2022/01/12 09:12:20 handle_events: error: libusb: interrupted [code -10]

karloygard, 2022-01-12:
Pushet 0.50 nå. Test og rapporter tilbake.

shoslemo, 2022-01-12:
Ser veldig lovende ut! Har ikke hatt noen problemer foreløpig, tester mer utover kvelden og i morgen!