karloygard/xcomfortd-go #39: Kommunikasjon fra analog converter

hlowe59, 2022-01-22:
Lysstyring av markiser (lux-måling).
Jeg har en CAEE-02/01 som overfører signaler fra lux-sensorer (to separate målere gjennom to kanaler) for lysstyringen av markiser.
Jeg opplever at den bare gir signalstyrken inn i grensesnittet, er det noe sted jeg kan hente ut den aktuelle verdien?
Det fungerer greit inn i Sensio, og dermed via Xcomfort, men jeg klarer ikke å hente ut verdier annet enn signalstyrken.

Sliter litt her, men det er sikkert bare jeg som ikke forstår det hele....

karloygard, 2022-01-23:
Du har rett, dette er ikke implementert ennå. Kan ta en titt.

karloygard, 2022-02-03:
Tentativ fiks i 0.52. Se om endringen gjør noe fornuftig, jeg har ikke en CAEE-02/01 jeg kan teste med.

hlowe59, 2022-02-03:
Heisann!
Nå ser jeg begge kanalene, blir spennende å se om verdiene endrer seg når lyset kommer. Det er jo lysmålere som er koblet til begge kanaler. Det er bek mørkt ute, klokken er 23.
Får en status etterhvert som lyset dukker opp i morgen, det er til og med meldt lettskyet og sol utover dagen!
Gir tilbakemelding da.

hlowe59, 2022-02-04:
Virker fantastisk. Tusen takk!
[image]

karloygard, 2022-02-04:
Får dessverre ikke satt måleenhet, siden konfigurasjonsfilen ikke sier noe om hva det er som måles.

hlowe59, 2022-02-04:
Alt OK. De måler lux - det er satt i MRF.
This is just perfect!!

closed 2022-02-07 by karloygard