karloygard/xcomfortd-go #40: Actuator not turning off

geeewizzz, 2022-02-09:
I updated HA today, not sure if that is related, but after that I have problems with my heating actuators, they seem to not be responding to messages.

2022/02/09 19:23:42 MQTT message; topic: 'xcomfort/16/set/switch', message: 'false'
2022/02/09 19:23:42 TX (seq 1, 1 parallel): [b1100a0010] 
2022/02/09 19:23:43 STATUS: [c3090610000000]
2022/02/09 19:23:43 TX command failed, retrying (1/2): timeout, no ACK received
2022/02/09 19:23:43 TX (seq 2, 1 parallel): [b1100a0020] 
2022/02/09 19:23:45 STATUS: [c3090620000000]
2022/02/09 19:23:45 TX command failed, retrying (2/2): timeout, no ACK received
2022/02/09 19:23:45 TX (seq 3, 1 parallel): [b1100a0030] 
2022/02/09 19:23:46 STATUS: [c3090630000000]
2022/02/09 19:23:46 WARNING: command for datapoint 16 failed, state now unknown: timeout, no ACK received

I've tried and retried several times, and also restarted the addon and HA itself. But the problem persists.

geeewizzz, 2022-02-10:
After testing this a bit more it seems like the messages are actually reaching the actuators and they are turning off/on as expected, but the status message back/correct state is the only problem. Any thoughts on what could cause this? Maybe it is just poor signal to my communication stick?

karloygard, 2022-02-10:
Likely poor signal or wrong routing. Try recalculating routes and see if that fixes the issue. Using the MRF log may also shed some light on the situation.