karloygard/xcomfortd-go #41: Wall-switches not showing any entities

oyvindth, 2022-04-10:
I have added a CTAA-02/04 Wall-button into my Home Assistant setup, but the only entities I get from the button is battery and signal strength. When I press the button, I can see the events "ON" "OFF" "UP Pressed" "UP Released" "Down Pressed" "Down Released" in the comfort logs, but I have no way of using that in my automations. The CTAA-02/04 seems to be correctly identified by the xComfort add-in, and all events are registred by the xComfort addin, but I see now way of using the info. Is there a possibility to att an entity to reflect the "ON" "OFF" "UP Pressed" "UP Released" "Down Pressed" "Down Released" state of the switch.

karloygard, 2022-04-11:
Do you have some logs? This should work, but it's been a while since I tested it.

karloygard, 2022-04-11:
You *should* be able to see something like this:

 [image]

oyvindth, 2022-04-12:
Thank you. This approach is working.

closed 2022-04-12 by oyvindth