karloygard/xcomfortd-go #44: Dimmer does not turn on at last value from HA.

Futrax, 2022-05-22:
When turning on a dimmer from Home assistant, the light turns on at max brightness every time.
The memory function when programming the dimmer to the communication stick.

Knutingeev, 2022-10-08:
Has anything been done about this?
It would have been nice if you could switch on the dimmer in the old value

karloygard, 2022-10-08:
Might be possible. I'm on vacation, but ping me in a week if you don't see any progress, I may have forgotten.

Knutingeev, 2022-10-16:
Something new here?

karloygard, 2022-10-16:
Tested just now and it seems to work out of the box? Am I misunderstanding something?

Knutingeev, 2022-10-17:
If I go to a dimmer and turn it on and then adjust it to 40%.  If I turn it off, wait a few seconds before I turn it on again, it goes to 100%

Knutingeev, 2022-10-17:
https://user-images.githubusercontent.com/96237669/196100764-67b817b8-b980-4049-b7bf-fd9b618500d4.MOV

docstalek, 2022-10-17:
0.57 seems to have fixed this issue.

closed 2022-10-21 by karloygard