karloygard/xcomfortd-go #45: Scenarios

eivindsor, 2022-07-22:
Is it possible to get support for scenarios? 

Where a datapoint is set to call upon a set value from several actuators? Calling upon light scenarios gives a much nicer dim to value simultaneously.

karloygard, 2022-07-22:
If you can figure out how to specify it with MQTT discovery, then maybe. MQTT discovery tends to be a bit limiting at times. Newer xC dimmers support dim to value, so it's technically possible.

Knutingeev, 2022-10-13:
Has anything been tested on this?
When programming in MRF, you can set several dimmers or relays to have the same data point. What should happen is set on the dimmer or relay in the MRF. Have used this at Sensio before. What is useful about this is that everything is activated at the same time. No or little delay between each device. It would be nice to have this option.

oyvindwe, 2025-07-05:
I am also interested in this. It should be possible by publishing a [scene to MQTT](https://www.home-assistant.io/integrations/scene.mqtt/). I don't know if a scene can be automatically created based on the datapoints configuration, or if it requires some extra configuration for this component.