karloygard/xcomfortd-go #46: Multiple simultaneous commands crash addon

ChristopherSkarsethstuen, 2022-10-07:
I am having problems with the addon crashing when multiple simultaneous commands are sent

Example 1:
7 automations trigger to send current temp from RCT to heating actuators
Example 2:
Turn on/off all the lights on the first floor (8Devices) 

Both will often result in the addon crashing with error:  panic: runtime error: slice bounds out of range [1:0]

Log:
[runtime error.txt](https://github.com/karloygard/xcomfortd-go/files/9735893/runtime.error.txt)

karloygard, 2022-10-07:
Not sure what causes that, but I've pushed a change in 0.55 that should at least prevent the crash. Something appears to be going wrong, though. Set logging to verbose and see if you can provoke the error again?

ChristopherSkarsethstuen, 2022-10-08:
[0.55 runtime error.txt](https://github.com/karloygard/xcomfortd-go/files/9738790/0.55.runtime.error.txt)
[0.55 runtime error verbose.txt](https://github.com/karloygard/xcomfortd-go/files/9738791/0.55.runtime.error.verbose.txt)

karloygard, 2022-10-08:
Ah, wrong fix. Issuing new fix shortly.

karloygard, 2022-10-08:
Pushed 0.56.

ChristopherSkarsethstuen, 2022-10-08:
Seems to work perfect. Thank you for the super fast fix Karl Anders 👍

karloygard, 2022-10-08:
Should be logging an error; "ignoring unexpected input", can your find that?

ChristopherSkarsethstuen, 2022-10-08:
Tried sending alot of commands but cannot find that in the logs
Not in regular log og verbose

closed 2023-06-08 by ChristopherSkarsethstuen