karloygard/xcomfortd-go #47: Consumption from relay with energy monitoring

Knutingeev, 2022-10-08:
I use the Switching Actuator New Generation (CSAU-01/01-1xxx) on the hot water tank, but I do not get any consumption. Or is there no support for consumption?

karloygard, 2022-10-08:
This should be supported. Turn on verbose logging and send me the logs?

Knutingeev, 2022-10-08:
Sorry, but then I almost have to ask how to do this. New to this

karloygard, 2022-10-08:
Are you using the Home assistant add-on? Go to configuration and turn on "Verbose". The "Log" tab will show the logs, find a message from the switching actuator, it should show what the actuator is sending.

Knutingeev, 2022-10-08:
Is this what you want to see?
2022/10/08 17:40:34 Device 6840772 (channel 3-'Varmtvann (Effekt)') sent message (battery mains-powered, signal good, cyclic false) event 'value' with value 0

Knutingeev, 2022-10-08:
I see now that it is sending. It comes up with the name power. When I have several with power readings, they are named power 1, power 2 and above

closed 2022-10-13 by Knutingeev