karloygard/xcomfortd-go #48: Restarting HA or editing automations trigger automations pushbuttons are programmed in

ChristopherSkarsethstuen, 2022-10-08:
When restarting HA or editing automations, pushbuttons will trigger automations they are programmed in
If I use MQTT Explorer to delete the retained flag on the datapoints for the pusbuttons this will stop
the automations from triggering (untill next time buttons are pushed)

Would it be possible to make the addon recognice the pushbuttons, and make them not retained to prevent this?

hlowe59, 2023-06-18:
This is a constant annoyance for me as well. 
I have some push-buttons by entrance doors that is "all light off" when you leave the house. These are fired when reboot of HA. and the whole building goes dark.
It would be nice to know if this is possible to handle through this integration, or if I must look to more intricate solutions that are posted other places.

docstalek, 2023-12-18:
I know this is an old issue, but I have been messing around with a solution to this. Basically it will set retain=false if the event comes from a pushbutton.

I have made a branch here: https://github.com/docstalek/xcomfortd-go/tree/not-retained-when-push-button
The code could possibly be made better by someone who actually knows Go :)

ChristopherSkarsethstuen, 2024-01-04:
Great :), Thank you
Maybe you could make a pull request to get this added easily?

docstalek, 2024-01-05:
I can certainly do that. It has been running here for a couple of weeks now without problems, so it might be time :)

closed 2024-01-17 by ChristopherSkarsethstuen