karloygard/xcomfortd-go #50: Problem with ACKs and CKOZ-00/03

koralfx, 2022-10-19:
Hi.
I am using xcomfortd-go in Home Assistant as addon.
I have ACK errors all the time, even though the modules are close to CKOZ-00/03. It does not bother me. Interestingly, messages from modules are coming without any problems. The problem is trying to re-send the message (two TXs). They delay new incoming MQTT messages. Is it possible to disable these attempts?
In your opinion, is it the fault of CKOZ-00/03 or the modules? I was buying everything over 10 years ago.
Will buying CKOZ-00/14 solve the problem?
Regards

koralfx, 2022-10-21:
I programmed it in MRF quite a long time ago, about 2 years ago, but I didn't change anything at home. So far I use the German Symcon and everything is fine here.

I provide logs from sending the `xcomfort/46/set/switch` with payload `true` command.
You can see that the device showed its status - that is, it worked.

```
2022/10/21 14:20:09 MQTT message; topic: 'xcomfort/46/set/switch', message: 'true'
2022/10/21 14:20:09 Device 4782752 (channel 0-'Lampa Salon Szafka') sent message (battery mains-powered, signal good, cyclic false) status switched on
2022/10/21 14:20:19 TX message was silently lost, likely never sent
2022/10/21 14:20:19 TX command failed, retrying (1/2): timeout, no ACK received
2022/10/21 14:20:29 TX message was silently lost, likely never sent
2022/10/21 14:20:29 TX command failed, retrying (2/2): timeout, no ACK received
2022/10/21 14:20:39 TX message was silently lost, likely never sent
2022/10/21 14:20:39 WARNING: command for datapoint 46 failed, state now unknown: timeout, no ACK received
```

E-K-I, 2022-10-21:
I had this issues with some actuators a while back. Recalculated routing in mrf, been good ever since.

koralfx, 2022-10-21:
I just did but nothing has changed.

koralfx, 2022-10-23:
I think the matter is clear. From Symcon pages:
> Only from the RF version 9.2 (approx. July/2008) the "Acknowledge" sent via the protocol level are evaluated. In previous releases, the system can not tell whether a command has been successfully delivered to a device. **_The RF version is a hardware upgrade_**, and should not be confused with the FW version that can be updated via software.

I wonder if this applies to the communication module or all ordinary modules ...

Good is disable ACKs option in addon config page.

For now, in SYMCON (on a separate raspberry) I made something like the XCOMFORT <-> MQTT gate. Not a very smart solution, but it works.

karloygard, 2022-10-24:
The issue is stated clearly in the logs: `This software may not work well with RF Revision < 9.0`. What I've read aligns with Symcon's statement ("*The RF version is a hardware upgrade*"); the device cannot be firmware upgraded to support ACK. You would likely have to purchase the new CKOZ 00/14, but the remaining devices should be fine.

I'll take a look at supporting this, but since we don't know if the devices actually receive messages, you'll have to pay a little extra attention when setting up the xComfort network in MRF and make sure that routing and signal strengths are good.

karloygard, 2022-10-24:
Can you turn on verbose and send me the logs with errors? I need to see how this device is behaving.

karloygard, 2022-10-24:
Okay, try 0.58 and see if the errors are gone.

koralfx, 2022-10-24:
Looks like it's working perfectly right now. Of course I will do more tests and if they are positive I will forget about SYMCON. Thanks for a great job, Karl.

koralfx, 2024-07-03:
Hi Karl,
is it possible to add an option to the Xcomfort configuration in the Home Assistant add-on so that the blinds do not move after restarts? Every time there is a HAOS update, the blinds either go up or down. Most of the time I do it at night so I only hear clicks in the modules. But it varies (and probably depends on the previous states) and I would prefer them not to react at all at startup.

Greetings.

Tomek Sosnowski