karloygard/xcomfortd-go #52: Does not send off confirmation anymore in node red mqtt

robertwtf, 2022-10-20:
**The xComfort addon does send redback numbers for which level the dimmer is set to. But have stopped to do so when turned to "off". 0 is not sent over mqtt.**

example: 
When using debug in node red, I got xcomfort/8/get/dimmer that gives me 1-100 readbacks when i dim inside home assistant or from xComfort buttons. 
But if I turn dimmer to off, it does not send 0 message anymore and my automations in node red will not register that dimmer actually are switched to 0/off.

karloygard, 2022-10-21:
You are correct, this is a consequence of the fix for #44. It's not immediately obvious to me how I can support both scenarios. A workaround for you is to listen to the boolean status on `xcomfort/8/get/switch`, that will still be sent as before. Other suggestions are welcome.

robertwtf, 2022-10-21:
Thank.  That is a solution I will try out. Thank you :)

closed 2022-10-24 by karloygard