karloygard/xcomfortd-go #54: 'Docker run' syntax please ?

oywino, 2022-12-15:
Hi, I cannot figure out how to pass MQTT parameters to  your add-on in my docker container set up.
The suggested command` ./xcomfortd-go -v -e -s tcp://user:password@mqtthost:1883` doesn't work when used with `docker run`
Also, I have to run the container in HOST mode as well as in privileged mode for it to find the CKOZ-00/14. This is my result so far:

```
2022/12/15 20:51:36 Opened USB device 'vid=188a,pid=1101,bus=1,addr=4', packet size 32/32
2022/12/15 20:51:36 Connecting to MQTT broker 'tcp://' with id 'xcomfort'
[ERROR] [client]   dial tcp: missing address
[WARN]  [client]   failed to connect to broker, trying next
2022/12/15 20:51:36 CI HW/RF/FW revision: 0, 13.0, 0
2022/12/15 20:51:36 CI RF/Firmware release: 2.11, 2.05
2022/12/15 20:51:36 CI serial number: 5290556
```

As you can see, it fails to find my MQTT server. Is there any other way to pass this i.e. as an ENVIRONMENT VARIABLE ?

Rgds
Viking

hans99, 2022-12-15:
You can use environment variable:
MQTT_SERVER

karloygard, 2022-12-15:
Take a look at https://github.com/karloygard/hassio-addons/blob/master/xcomfort/rootfs/etc/services.d/xcomfort/run for some inspiration on which parameters to pass to the daemon. Running in HA, you will likely want much of the same setup.

oywino, 2022-12-16:
> You can use environment variable: MQTT_SERVER
Thanks! And how did you know that? Is it documented somewhere? What about passing the other values, such as Username, Password, portnumber, name and location of Datapoint file etc.?  Care to share those variable names with me as well? (Or even better, point to where this might be documented) 🙂

oywino, 2022-12-16:
Many thanks for bearing with me. Do you think it is at all possible to extract / build a DATAPOINT file from the Bridge?
Alternatively, if I use a CKOZ-00/13 along with a CKOZ-00/14 to build a DATAPOINT file for my existing installation, could doing so interfere with or corrupt the existing setup I have with the Bridge as the current master device?
(I'm just asking for your qualified opinion/guesstimate)

oywino, 2022-12-16:
I did manage to get connection to my MQTT server. The resulting log entries are as follows:
```
2022/12/16 20:23:29 Opened USB device 'vid=188a,pid=1101,bus=1,addr=5', packet size 32/32
2022/12/16 20:23:29 Connecting to MQTT broker 'tcp://192.168.X.X:1883' with id 'xcomfort'
[WARN]  [store]    memorystore wiped
2022/12/16 20:23:29 Connected to broker
[WARN]  [store]    memorystore del: message 1 not found
[WARN]  [store]    memorystore del: message 2 not found
[WARN]  [store]    memorystore del: message 3 not found
[WARN]  [store]    memorystore del: message 4 not found
[WARN]  [store]    memorystore del: message 5 not found
2022/12/16 20:23:29 CI HW/RF/FW revision: 0, 13.0, 0
2022/12/16 20:23:29 CI RF/Firmware release: 2.11, 2.05
2022/12/16 20:23:29 CI serial number: XXXXXX
```
(I X'ed out some info just for security reasons)
The above doesn't mean much to me, except that it looks somehow correct (?) 
But no new entities shows up in HA, and I cannot see anything published even if I subscribe to `xcomfort/#`
Do you have any advise?

karloygard, 2022-12-23:
Looks fine. Add the parameters `--eprom --hadiscovery --hadiscoveryprefix homeassistant` and the devices will be read from the USB stick eprom and auto discovered in Home Assistant.

oywino, 2022-12-23:
I'm not sure I understand. It looks to me like these are parameters one would add to a CLI `docker run` command ?
But I'm using a docker compose YAML file. How do I add this to the YAML ;

```
version: '3'

services:
  xcomfort:
    image: karloygard/xcomfortd:latest
    privileged: true
    restart: "no"
    network_mode: host

    environment:
      LANG: nb_NO.UTF-8
      TZ: Europe/Oslo
      MQTT_SERVER: tcp://usr:pw@192.168.X.X:1883

    volumes:
      - /share/Public:/config
```

WinFinn, 2023-05-22:
Did you figure this one out?
I ran into the same issue as you, not having the supervised HA and trying to integrate xcomfort as a standalone docker.  (I guess basically installing a HA addon manually).

I made it work in the end. Though not as I wanted it too.
Using your docker compose above I just added the command option in compose yaml file. Basically replacing the command sent by original image:
```
command:
  xcomfortd -s tcp://usr:pwd@xxx.xxx.xxx.yyy:1883 --eprom --host ["xxx.xxx.xxx.xxx"] --hadiscovery --hadiscoveryprefix homeassistant
``` 
I'm using both USB stick and an ECI, and all devices connected to those two got detected by my HA container.
I know this is not a good way of doing it, writing usr, passwords and IP's directly in the compose file. But it was the only way I could make it work at the moment.
I could pass the MQTT_SERVER environment fine without replacing the start command. But I found no other way to pass the other arguments i needed when starting xcomfortd.