karloygard/xcomfortd-go #55: Impulse Input (CIZE-02/01)

TomEkk33, 2022-12-31:
Hello,

Your integration works perfect in HA. Thanks for your great work!

I have in my system CIZE-02/01 media impulse input. However it does not show PULSES sensor value in HA system. 
There is only "Signal strength" visible.

In a XCOMFORT log there is a message
"Device 5935420 (channel 0-'LE-Klima (Channel A)') sent message (battery mains-powered, signal good, cyclic true) event 'value' with value 19756"

Could you point me where this (PULSES) should be added in your integration?

Thanks,
Tom.

karloygard, 2023-01-08:
Sorry about the delayed response. Pulses are currently not supported. I would have to figure out how to expose the pulse value with MQTT auto-discovery first. I'll let you know if I make any progress.

TomEkk33, 2023-01-09:
Thanks for your reply. I suppose it should be treated similar as energy value of Energy meter CEMx-01/01 - which works perfect. 
The PULSES value should have potential classes in HA as gas, water, energy counters.

karloygard, 2023-01-09:
Sounds sensible. Do you have any suggested as to how I should choose the unit of measurement/device class? The DPL itself doesn't indicate what's being measured.

karloygard, 2023-01-15:
Attempted a simple fix for this in 0.60. I've exposed the the value as a simple number, no unit of measurement or device class at the moment. See if you can make it do anything useful.

TomEkk33, 2023-01-16:
Thanks - this should be enough to start using the sensor.

Unfortunately, MQTT does not show any value except signal strength.
[image]

[image]

karloygard, 2023-01-16:
Try 0.61.

closed 2023-02-25 by TomEkk33