karloygard/xcomfortd-go #56: -dBm unit

TomEkk33, 2023-01-15:
Hello,

The HA requires that signal strength units should be 'dBm'. Currently XComfort/MQTT reports a minus sign prefix.

unit_of_measurement: '-dBm'

Could you please fix this, because of warnings in HA log file related to this for each device. Thank you.

karloygard, 2023-01-15:
Fixed in 0.59.

closed 2023-01-15 by karloygard

TomEkk33, 2023-01-15:
I got the Warning like this in v0.60

"The unit of sensor.signal_strength_34 is changing, got multiple {'dBm', '-dBm'}, generation of long term statistics will be suppressed unless the unit is stable and matches the unit of already compiled statistics (-dBm). Go to https://my.home-assistant.io/redirect/developer_statistics to fix this"

Maybe it is caused because previous  statistics have -dBm, and new one got dBm?