karloygard/xcomfortd-go #57: How to get the DPL file? I have an MRF file.

mgartin, 2023-02-02:
First, thank you for this software!
I'm new to xComfort and have recently tried this software, but realized that I need the file describing the devices in my system.
Thus, I have borrowed a CKOZ-00/13 USB dongle and have mapped out all devices (well, most of them) using the Eaton RF-Syste 2.75 NO software, which resulted in a .MRF file of 1.4 MB.
However, this is not the format that is needed - I need a .DPL file, right?

How do I get that file? My searches this far did not give any obvious way to do it.

In the MRF manual, it is stated that I should right-click on the USB-RF icon to export a DPL file. But I cannot find such icon in the Eaton RF-System program. What am I missing here?

karloygard, 2023-02-02:
Do you have a CKOZ-00/14?

karloygard, 2023-02-02:
Make sure the CKOZ-00/14 is included in the network and right click that icon. Upload the DPL to the stick rather than downloading it.

mgartin, 2023-02-02:
Ah, I see. I did not have the CKOZ-00/14 connected to anything during the scan. I will try this, thanks!

mgartin, 2023-02-02:
So, I now had the CKOZ-00/14 connected to my Raspberry Pi (which runs Home Assistant), and did a rescan of the xcomfort network (using the CKOZ-00/13 connected to a Windows laptop). It showed up and I was able to transfer the datapoints (DPL) to the 00/14 wirelessly. 

I also managed to start the daemon like this:
`./xcomfortd-go --hd -e -s tcp://192.168.1.2:1883 --verbose  
`
And after restarting Home Assistant, I did find one device named "ukjent (S)". 

When I look at the topology in the MRF software, I notice that the "USB Comm.-stick" only has one connected to this "ukjent (S)", which is disconnected from the rest of my network.

I tried to reset all connections to the stick, but that crashed the program, even Windows in fact. I guess I will have to make the 00/14 part of my xcomfort network somehow. I will continue to try and report back. Thanks so far!

mgartin, 2023-02-07:
Ok, so I have come a litte further. I managed to recalculate all connections in the xcomfort software, and the USB device now has both some red and black lines connecting to the other devices. Not sure what that means, I guess all should be black?

 I uploaded the datapoints to the stick and restarted the daemon (my mqtt broker is at 192.168.1.2):
`./xcomfortd-go --hd -e -s tcp://192.168.1.2:1883 --verbose`

It lists alot of `EPROM` and `EXTENDED` numbers, and repeats "`Datapoint 1:`" lines with devices with different serial numbers, and some "`Datapoint 0:`" lines with the same but "(Statusforbindelse)" in the end, (translates to "Status connection" I guess).
Good!

Ok, so I can now subscribe to the topic like this (Linux):
`mosquitto_sub -h 192.168.1.2 -v  -t xcomfort/\#`

And set values:
`mosquitto_pub -h 192.168.1.2 -t xcomfort/1/set/switch -m false -d`
which turns off ALL the lights. The output from the daemon is this:
```
2023/02/07 22:36:09 MQTT message; topic: 'xcomfort/1/set/switch', message: 'false'
2023/02/07 22:36:09 TX (seq 7, 1 parallel): [b1010a0070] 
2023/02/07 22:36:10 STATUS: [c31c0470100000]
2023/02/07 22:36:10 RX: [c1017000000000000022100c]
2023/02/07 22:36:10 Device 6639043 (channel 0-'Sjalteaktuator (Aktuator)') sent message (battery mains-powered, signal good, cyclic false) status switched off
2023/02/07 22:36:10 RX: [c100733900076a5f004a04101f0000004b1007]
2023/02/07 22:36:10 Device 6253063, type CSAU 01/01-16 sent extended status message: status OFF, temp 31C (battery mains-powered, signal normal)
2023/02/07 22:36:12 RX: [c100733900928f62004d000100002c000000361000]
2023/02/07 22:36:12 Device 6459282, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 44C (battery mains-powered, signal good)
2023/02/07 22:36:12 RX: [c100733900978f62004d000100002300000041100f]
2023/02/07 22:36:12 Device 6459287, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 35C (battery mains-powered, signal good)
2023/02/07 22:36:12 RX: [c100733900f5e663004d0001000023000000401007]
2023/02/07 22:36:12 Device 6547189, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 35C (battery mains-powered, signal good)
2023/02/07 22:36:12 RX: [c100733900f5e563004d000100001d000000471000]
2023/02/07 22:36:12 Device 6546933, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 29C (battery mains-powered, signal normal)
2023/02/07 22:36:13 RX: [c100733900568f62004d000100002000000049100f]
2023/02/07 22:36:13 Device 6459222, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 32C (battery mains-powered, signal normal)
2023/02/07 22:36:13 RX: [c1007339005f9262004d0001000025000000321006]
2023/02/07 22:36:13 Device 6459999, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 37C (battery mains-powered, signal good)
2023/02/07 22:36:13 RX: [c1007339005f8f62004d000100001e0000004a1001]
2023/02/07 22:36:13 Device 6459231, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 30C (battery mains-powered, signal normal)
2023/02/07 22:36:13 RX: [c100733900d5f76a004d0001000018000000481006]
2023/02/07 22:36:13 Device 7010261, type CDAU 01/04 sent extended status message: status OFF, value 0, temp 24C (battery mains-powered, signal normal)
```

Likewise, sending `true` gives this output:
```
2023/02/07 22:36:25 MQTT message; topic: 'xcomfort/1/set/switch', message: 'true'
2023/02/07 22:36:25 TX (seq 8, 1 parallel): [b1010a0180] 
2023/02/07 22:36:25 STATUS: [c31c0480100000]
2023/02/07 22:36:25 RX: [c1017000620000000022100e]
2023/02/07 22:36:25 unknown switching actuator status 98
2023/02/07 22:36:25 Device 6639043 (channel 0-'Sjalteaktuator (Aktuator)') sent message (battery mains-powered, signal good, cyclic false) unknown
2023/02/07 22:36:25 Message not handled [c1017000620000000022100e]
2023/02/07 22:36:26 RX: [c100733900076a5f004a04201f0000004c1008]
2023/02/07 22:36:26 Device 6253063, type CSAU 01/01-16 sent extended status message: status ON, temp 31C (battery mains-powered, signal weak)
2023/02/07 22:36:27 RX: [c1007339005f9262004d0002fc0024000000331007]
2023/02/07 22:36:27 Device 6459999, type CDAU 01/04 sent extended status message: status ON, value 252, temp 36C (battery mains-powered, signal good)
2023/02/07 22:36:27 RX: [c100733900928f62004d0002fc002b000000361001]
2023/02/07 22:36:27 Device 6459282, type CDAU 01/04 sent extended status message: status ON, value 252, temp 43C (battery mains-powered, signal good)
2023/02/07 22:36:28 RX: [c100733900568f62004d0002fc001f000000491000]
2023/02/07 22:36:28 Device 6459222, type CDAU 01/04 sent extended status message: status ON, value 252, temp 31C (battery mains-powered, signal normal)
2023/02/07 22:36:28 RX: [c100733900f5e663004d0002fc0023000000451008]
2023/02/07 22:36:28 Device 6547189, type CDAU 01/04 sent extended status message: status ON, value 252, temp 35C (battery mains-powered, signal normal)
2023/02/07 22:36:29 RX: [c100733900f5e563004d000296001c000000491001]
2023/02/07 22:36:29 Device 6546933, type CDAU 01/04 sent extended status message: status ON, value 150, temp 28C (battery mains-powered, signal normal)
2023/02/07 22:36:30 RX: [c100733900978f62004d0002dd0022000000401000]
2023/02/07 22:36:30 Device 6459287, type CDAU 01/04 sent extended status message: status ON, value 221, temp 34C (battery mains-powered, signal good)
2023/02/07 22:36:30 RX: [c1007339005f8f62004d0002fc001d000000481002]
[... cut ...]
```

This is great, but how do I control individual switches/dimmers?
Is there
  - Another MQTT message that I need to send (I tried using the serial number rather than "1", but that didn't work)
  - Have I somehow misconfigured the datapoints file in the Xcomfort program?

Also, do I need to manually configure all switches with MQTT commands in the Home Assistant YAML file, or is there another way to make it discover the MQTT switches/dimmers? Any docs on this?

I appreciate all help I can get!

karloygard, 2023-02-08:
You need to read up a bit about MRF and xComfort. If you want to control individual dimmers/actuators you need to create individual datapoints for those devices. It looks like you have connected all the dimmers to datapoint 1, which will switch all the dimmers as a group. That's useful in its own way, but maybe not what you wanted.

mgartin, 2023-02-08:
Thank you @karloygard!

I scanned briefly through the 250 page PDF, but moved quickly on to trial and error and found that I had to re-draw the connections between the USB icon and the actuators in the MRF program, and define distinct datapoint numbers (thanks for that hint!). 

After a few rounds running around the house and touching all switches during recalculation of connections I right clicked the USB icon and generated the datapoint list and finally got it transferred to the CKOZ-00/14 stick and your program recognized it all.

Now Home Assistant automatically found the dimmers and switches, yes!

Thanks a bunch!

closed 2023-04-16 by mgartin