karloygard/xcomfortd-go #58: Pushbutton only listed as a battery sensor

lidvarko, 2023-02-07:
Devices are imported from file and all switches and dimmers seems to work fine. However, all pushbuttons does not have any state that show if the button was pressed. It only show the battery level for the button. How can I detect if any button is pressed?
[image]

karloygard, 2023-02-07:
The buttons don't have a state, they only fire events. Try making an automation with one, HA should discover the event types.

lidvarko, 2023-02-08:
I have looked for events also (in HA Developer Tools - Events). Subscribed to all events and I see all other events, but none that appear when I push a button. If I try start a Automation from the screen shown above, then there are no option to select an event. I only can select from the state of the battery and signal strength. It does not seem like HA has discover the event or that HA is even receiving the events at all from the pushbuttons.

lidvarko, 2023-02-08:
Never mind, seems to work now after restart of XComfort add-on and reload of MQTT integration.
Thanks for help :-)

closed 2023-02-08 by lidvarko