karloygard/xcomfortd-go #61: Support for Heating Actuator / Room Controller Touch (CHAU-01/01-10E - CPAD-00/198 aka CRCA-00/08)

mgartin, 2023-07-21:
Hi, I have a set of Room Controller Touch and heating actuators for controlling heating in floors (bathrooms, hallways).
Currently only on pair of these are active, but I will get the rest installed if I can get help solving this:

The device is not exposed as a climate device in HA (Home Assistant). Do I need to do anything special in MRF to control the floor temperature?
If I draw a connection between the USB stick and the Heating Actuator in MRF, I can choose between `Energi` (Energ) and `Belastningsfeil` (Loaderror?).  For the Room Controller Touch, I can choose between `Fuktighet` and `Touch-sensor A`. All four are already established connections.

Is something needed in HA for it to be a controllable climate device? All help appreciated - if I can't control these xComfort devices from HA, I will install Z-Wave devices instead I think.

oyvindth, 2023-08-26:
Have a look here. https://www.facebook.com/groups/449526166277051/?ref=share_group_link  
everythinh is already supported.

mgartin, 2023-08-26:
Thanks @oyvindth, I looked at that post (why do people invest their time in Facebook for such things I cannot understand - it is pretty hard to read a thread in there... anyways..).

So I looked in HA and find the two entities:

- ActHeatBathroom2 (Heating Actuator, CHAx-01/xx) which has Energi, Power (unknown) and Signal strength. Something seems a bit off in the naming though - the parenthesis don't match:

[image]

- RoomControlBathroom2 (CRCA-00/xx) which has Battery, Fuktighet, and Signal Strength. Also here are the parenthesis a bit off:

[image]

**So, I don't have any Temperature!**

If I look in `home-assistant.log`, I find that the RoomController has id `8111755`
```

$ cat ~/.homeassistant/home-assistant.log|grep 8111755 |cut -d\  -f3-
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor xcomfort_8111755_battery
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor xcomfort_8111755_rssi
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor xcomfort_8111755_RoomControlBathroom2_Fuktighet_
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: device_automation xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__on
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: device_automation xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__upPressed
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: device_automation xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__upReleased
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: device_automation xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__off
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: device_automation xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__downPressed
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: device_automation xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__downReleased
WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'device': {'identifiers': ['8111755'], 'manufacturer': 'Eaton', 'model': 'Room Controller Touch (CRCA-00/xx)', 'name': 'RoomControlBathroom2 (', 'via_device': 'CI Stick', 'connections': []}, 'device_class': <SensorDeviceClass.HUMIDITY: 'humidity'>, 'name': 'RoomControlBathroom2 (Fuktighet)', 'state_class': <SensorStateClass.MEASUREMENT: 'measurement'>, 'state_topic': 'xcomfort/23/event/+', 'unique_id': '8111755_ch1', 'unit_of_measurement': '%', 'payload_not_available': 'offline', 'availability_mode': 'latest', 'encoding': 'utf-8', 'qos': 0, 'payload_available': 'online', 'force_update': False, 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Fuktighet)'
INFO (MainThread) [homeassistant.components.mqtt.mixins] Device trigger ('device_automation', 'xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__on') has been initialized
INFO (MainThread) [homeassistant.components.mqtt.mixins] Device trigger ('device_automation', 'xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__upPressed') has been initialized
INFO (MainThread) [homeassistant.components.mqtt.mixins] Device trigger ('device_automation', 'xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__upReleased') has been initialized
INFO (MainThread) [homeassistant.components.mqtt.mixins] Device trigger ('device_automation', 'xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__off') has been initialized
INFO (MainThread) [homeassistant.components.mqtt.mixins] Device trigger ('device_automation', 'xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__downPressed') has been initialized
INFO (MainThread) [homeassistant.components.mqtt.mixins] Device trigger ('device_automation', 'xcomfort_8111755_RoomControlBathroom2_Touch_sensor_A__downReleased') has been initialized

```

So, there is something with the naming, but I don't quite understand what.

Similarly, for the Heating actuator, I find id 010125:
```
$ cat ~/.homeassistant/home-assistant.log|grep 010125 |cut -d\  -f3-
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor xcomfort_8010125_power
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor xcomfort_8010125_rssi
INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor xcomfort_8010125_ActHeatBathroom2_Energi_
WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'device': {'identifiers': ['8010125'], 'manufacturer': 'Eaton', 'model': 'Heating actuator (CHAx-01/xx)', 'name': 'ActHeatBathroom2 (', 'via_device': 'CI Stick', 'connections': []}, 'device_class': <SensorDeviceClass.ENERGY: 'energy'>, 'name': 'ActHeatBathroom2 (Energi)', 'state_class': <SensorStateClass.TOTAL_INCREASING: 'total_increasing'>, 'state_topic': 'xcomfort/25/event/value', 'unique_id': '8010125_ch2', 'unit_of_measurement': 'kWh', 'payload_not_available': 'offline', 'availability_mode': 'latest', 'encoding': 'utf-8', 'qos': 0, 'payload_available': 'online', 'force_update': False, 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Energi)'

```

Is this simply a matter of changing my naming scheme in the MRF? What should I change?

From the linked Facebook post, I understand that the way forward is to create a Generic Thermostat, but for that I assume I need the temperature.

mgartin, 2023-09-03:
Ok, so I see there recently was a  [fix](https://github.com/karloygard/xcomfortd-go/commit/571d9e900def34db7539f481b71db6939bc89b95) for the regex (thanks!). I'll give that a try and see if anything changes.

karloygard, 2023-09-04:
Temperature should be supported, but this seems like an MRF configuration issue. I don't have MRF in front of me, but you can check the MRF manual to see how this should be configured.

mgartin, 2023-10-17:
Just want to report here in case others face the same issue:
I had to change firmware on the RCT (room controller) using an USB cable and the MRF software. I changed from 307U to 301L (ie. from universal to legacy). That was a bit of surprise for me. 
I'll close the issue now, since my problem was with the FW of the xComfort device, not this software.

closed 2023-10-17 by mgartin