karloygard/xcomfortd-go #62: MQTT Name changes in 2023:8

FSHelgeland, 2023-08-03:
There have been some changes for naming mqtt-devices in the last update of the HA -core.  Causing for now only warnings, but I belive this will need a fix. Here is an example of a warning from the log:

"2023-08-02 23:27:16.735 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'device': {'identifiers': ['4315348'], 'manufacturer': 'Eaton', 'model': 'Motion Detector (CBMA-02/xx)', 'name': 'Bevegelse Entre (Kanal A)', 'via_device': 'CI Stick', 'connections': []}, 'device_class': <BinarySensorDeviceClass.MOTION: 'motion'>, 'name': 'Bevegelse Entre (Kanal A)', 'payload_off': 'switchOff', 'payload_on': 'switchOn', 'state_topic': 'xcomfort/44/event', 'unique_id': '4315348_ch0', 'qos': 0, 'force_update': False, 'encoding': 'utf-8', 'payload_not_available': 'offline', 'payload_available': 'online', 'enabled_by_default': True, 'availability_mode': 'latest'}, this is not expected. Please correct your configuration. The entity name will be set to `null`"

Hope there will be a fix for this soon. 

Br: Freddy

karloygard, 2023-08-03:
I'll issue a fix when I find some time.

karloygard, 2023-08-27:
I have attempted a fix in 5275306. It's not super obvious to me how naming should work in Home Assistant, but test it and see if it makes sense.

FSHelgeland, 2023-08-28:
I installed as soon as you released it, so far it looks good.

karloygard, 2023-08-28:
Will check tonight.

FSHelgeland, 2023-08-28:
It seems like the switches listen for the topic in Node Red works as before, but the buttons directly integrated in Home Assistant in automations does not do anything. For me that is not a problem as I can trigger it from Node Red, that is meybee the best solution for me. But I do think you should have a look at it.

karloygard, 2023-08-28:
It appears to work just fine here. I didn't touch that particular code, except renaming the discovery topic. The event topic itself was not renamed. Can you dig a bit and see what breaks your setup?

karloygard, 2023-08-28:
Hah, that's no good. I'll extend the regex a bit.

FSHelgeland, 2023-08-28:
Sov 1 Baddør ble til R, nå heter den Sov 1 Bad. Skal sjekke om jeg kan få bryterne til å trigge en automasjon.

FSHelgeland, 2023-08-28:
I dont know what happened, but it works know. Maybe it was on the naming bit? I have removed everything with an .æøå in the names. Restarted the add on, restarted the mqtt broker and now it works fine.

FSHelgeland, 2023-08-28:
I just noticed I have a lot of switches with the letter Ø in and also some with . I havent changed them, så I get a warning that they are missing. I think I wait and see if you manage to fix it before renaming them all, as I have multiple automations with those.

karloygard, 2023-08-28:
Confirm that I see the issue. Fix forthcoming.

FSHelgeland, 2023-08-28:
We do not have bad weather up in north, the summer seems to never end.

FSHelgeland, 2023-08-28:
But I cant find any update though

FSHelgeland, 2023-08-28:
After a restart it popped up, installing now.

ChristopherSkarsethstuen, 2023-08-28:
Names are working now, but I still can't get any pushbuttons to trigger automations

karloygard, 2023-08-28:
Can you try to figure out what has broken your setup?

ChristopherSkarsethstuen, 2023-08-28:
Working now. Deleted the triggers from the automations and re-added them

FSHelgeland, 2023-08-28:
When I tested my switches I made a test automation so the switch was in a new automation. So maybe there should have been a release note on that?

karloygard, 2023-08-29:
I *suspect* that this is an unavoidable side effect of fixing the naming issue, but it's regrettable nevertheless. I'll add a notice of the breaking change.

closed 2023-08-29 by karloygard